Koenkk/zigbee2mqtt #26220: Zigbee Unstable. Need help

arthurperezbessa, 2025-02-06:
### What happened?

Hi! I have a Zigbee network at home that sometimes shows an error message, but the devices work fine.

I installed a similar system in a client of my company and there I have the same messages, but there there are a lot of delays and sometimes commands won't work

I will try to provide the maximum of information that I can

[image]

The biggest issues are with the devices in the Main Bedroom left of this picture

[image]

### Zigbee2MQTT version

2.0.0 commit: unknown

### Adapter firmware version

8.0.2 [GA] 115200_sw_flow

### Adapter

Itead Sonoff Zigbee 3.0 USB Dongle Plus V2 - EmberZNet - EFR32MG21

### Setup

Home Assistant on a Raspberry Pi4

### Debug log

[2025-02-06 16:59:59] error: 	z2m: Publish 'set' 'state' to 'TW Banheiro Esquerdo' failed: 'Error: ZCL command 0xa4c13813ef17e501/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":42862,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":158},"zclSequence":13,"commandIdentifier":11} timed out after 10000ms)'
[2025-02-06 17:00:00] error: 	z2m: Publish 'set' 'state' to 'Closet 2' failed: 'Error: ZCL command 0xa4c1385c8cdece0f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":37343,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":167},"zclSequence":6,"commandIdentifier":11} timed out after 10000ms)'
[2025-02-06 17:05:20] error: 	z2m: Publish 'set' 'brightness' to 'DIMMER 4' failed: 'Error: ZCL command 0xa4c1382d5d12a4c4/1 genLevelCtrl.moveToLevelTuya({"level":1000,"transtime":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":13858,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":11},"zclSequence":42,"commandIdentifier":11} timed out after 10000ms)'
[2025-02-06 17:05:20] error: 	z2m: Publish 'set' 'state' to 'DIMMER 3' failed: 'Error: ZCL command 0xa4c138950625e844/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":10113,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":16},"zclSequence":44,"commandIdentifier":11} timed out after 10000ms)'
[2025-02-06 17:05:20] error: 	z2m: Publish 'set' 'state' to 'MÓDULO 6' failed: 'Error: ZCL command 0xa4c138834593b603/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":65045,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":19},"zclSequence":47,"commandIdentifier":11} timed out after 10000ms)'
[2025-02-06 17:05:20] error: 	z2m: Publish 'set' 'state' to 'MÓDULO 7.1' failed: 'Error: ZCL command 0xa4c1382f53379346/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":36468,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":20},"zclSequence":48,"commandIdentifier":11} timed out after 10000ms)'
[2025-02-06 17:07:47] error: 	zh:ember: Delivery of BROADCAST failed for '65533'.

Total 33
By device type
Router: 26
End devices: 7
By power source
Mains (single phase): 26
Battery: 6
undefined: 1
By vendor
_TZ3000_a8mhmsol: 8
_TZ3000_785olaiq: 6
_TZ3210_ysfo0wla: 4
_TZ3000_wkai4ga5: 3
_TZ3000_lwthnp7j: 2
_TZ3000_xzercucx: 2
_TZE200_fphxkxue: 2
SONOFF: 1
_TZE204_nvxorhcj: 1
_TZ3000_jjdkhueq: 1
Yale: 1
undefined: 1
_TZ3290_8biqfysp: 1
By model
TS0002: 11
TS0003: 6
TS110E: 4
TS0601: 4
TS0044: 3
TS0004: 2
DONGLE-E_R: 1
NO LOCK PRESENT: 1
TS1201: 1

Pyla5-Katy0-Vibe7, 2025-02-10:
Seems like that youve got the same issue as me: https://github.com/Koenkk/zigbee2mqtt/issues/25945

r0m1CH, 2025-02-10:
Seems like that youve got the same issue as me: https://github.com/Koenkk/zigbee2mqtt/issues/25945

Nerivec, 2025-02-19:
Can you run a network scan (both channel usage & existing networks) with the adapter so we can get an idea of what is going on around it?
https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network

arthurperezbessa, 2025-02-20:
Is it possible to run this in Home Assistant OS on Raspberry Pi?

Nerivec, 2025-02-20:
No, not easily, but you can run it from any computer/OS.
https://github.com/Nerivec/ember-zli/wiki#installation

arthurperezbessa, 2025-02-25:
I ran a network scan and here's what I found:

Some devices close to the coordinator are experiencing errors and delays, but the real issues seem to be with the devices that need to route through other routers. Shouldn't Zigbee be able to handle that? I only have around 40 devices, and the routes aren't that complex—just Device → Device Router → Coordinator at most.

```
✔ Menu Scan network
✔ Radio transmit power [-128-127] 5
✔ Type of scan Scan each channel for existing networks
✔ Duration of scan per channel 998 msec
[2025-02-25 11:51:34.942] info:         cli: Found network:
[2025-02-25 11:51:34.943] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.943] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.943] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.943] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.943] info:         cli:   - Node RSSI: -52 dBm | LQI: 236
[2025-02-25 11:51:34.943] info:         cli: Found network:
[2025-02-25 11:51:34.944] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.944] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.944] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.944] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.944] info:         cli:   - Node RSSI: -63 dBm | LQI: 192
[2025-02-25 11:51:34.944] info:         cli: Found network:
[2025-02-25 11:51:34.944] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.945] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.945] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.945] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.945] info:         cli:   - Node RSSI: -74 dBm | LQI: 148
[2025-02-25 11:51:34.945] info:         cli: Found network:
[2025-02-25 11:51:34.945] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.945] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.945] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.946] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.946] info:         cli:   - Node RSSI: -63 dBm | LQI: 192
[2025-02-25 11:51:34.946] info:         cli: Found network:
[2025-02-25 11:51:34.947] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.947] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.947] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.947] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.947] info:         cli:   - Node RSSI: -79 dBm | LQI: 128
[2025-02-25 11:51:34.947] info:         cli: Found network:
[2025-02-25 11:51:34.948] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.948] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.948] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.948] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.948] info:         cli:   - Node RSSI: -74 dBm | LQI: 148
[2025-02-25 11:51:34.948] info:         cli: Found network:
[2025-02-25 11:51:34.949] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.949] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.949] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.949] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.949] info:         cli:   - Node RSSI: -71 dBm | LQI: 160
[2025-02-25 11:51:34.950] info:         cli: Found network:
[2025-02-25 11:51:34.950] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.950] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.950] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.950] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.950] info:         cli:   - Node RSSI: -67 dBm | LQI: 176
[2025-02-25 11:51:34.951] info:         cli: Found network:
[2025-02-25 11:51:34.951] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.951] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.951] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.951] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.951] info:         cli:   - Node RSSI: -73 dBm | LQI: 152
[2025-02-25 11:51:34.951] info:         cli: Found network:
[2025-02-25 11:51:34.951] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.952] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.952] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.952] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.952] info:         cli:   - Node RSSI: -61 dBm | LQI: 200
[2025-02-25 11:51:34.952] info:         cli: Found network:
[... 5 lines cut ...]
[2025-02-25 11:51:34.953] info:         cli: Found network:
[2025-02-25 11:51:34.953] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.953] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.953] info:         cli:   - Channel: 11
[2025-02-25 11:51:34.953] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.954] info:         cli:   - Node RSSI: -70 dBm | LQI: 164
[2025-02-25 11:51:34.954] info:         cli: Found network:
[2025-02-25 11:51:34.954] info:         cli:   - PAN ID: 60427
[2025-02-25 11:51:34.954] info:         cli:   - Ext PAN ID: 196,166,9,227,229,30,188,38
[2025-02-25 11:51:34.954] info:         cli:   - Channel: 11
[... cut ...]
[2025-02-25 11:51:34.956] info:         cli:   - Node RSSI: -88 dBm | LQI: 92
[2025-02-25 11:51:34.956] info:         cli: Found network:
[2025-02-25 11:51:34.956] info:         cli:   - PAN ID: 3480
[2025-02-25 11:51:34.956] info:         cli:   - Ext PAN ID: 33,32,75,244,138,125,241,233
[2025-02-25 11:51:34.956] info:         cli:   - Channel: 20
[2025-02-25 11:51:34.956] info:         cli:   - Allowing join: no
[2025-02-25 11:51:34.956] info:         cli:   - Node RSSI: -84 dBm | LQI: 108
```

arthurperezbessa, 2025-02-25:
> Looks like you only have 1 other network nearby, so, shouldn't be too much of a trouble there. How about the `channels usage` scan, what did that one say?
> 
> Zigbee is able to route to a decent depth (although it does degrade the deeper you go, usually after 3-ish). Tuya is another story, and most of your network is... There are numerous Tuya devices that have really poor routing capabilities, or worse, filtered routing. This can heavily impact nested devices. You can try re-pairing some nested devices to a specific device (priority to the coordinator or the Dongle-E router), but unfortunately, devices may switch back on their own if they deem the link quality to another device better...

Unfortunately, the device options available in my country (Brazil) are limited, with Tuya being one of the few choices. I really wish I could disable the router function on their devices, so I could rely solely on dongles as routers.

Nerivec, 2025-02-25:
You can try adding a couple more Dongle-E routers in strategic places, hopefully they take over most of the traffic.
You can also try to identify if a particular model of Tuya is causing more problems than others, and if so replace it.

arthurperezbessa, 2025-02-25:
> You can try adding a couple more Dongle-E routers in strategic places, hopefully they take over most of the traffic. You can also try to identify if a particular model of Tuya is causing more problems than others, and if so replace it.

Unfortunately, this issue occurs with several Tuya models. I could add more routers, but finding a spot that looks good aesthetically is quite challenging.

closed 2025-05-05 by github-actions[bot]