Koenkk/zigbee2mqtt #26221: Z2M Crash after some time

vilaemail, 2025-02-06:
### What happened?

After couple of hours running Z2M crashes and then fails to reconnect to the adapter (add-on never restarts). My UZG-01 adapter (powered over PoE and connected to via LAN) frontend remains accessible. I need to reboot my coordinator and Z2M to get into partially recovered state (Some devices missing). To fully recover (all devices to appear back online) I need to restart all mains powered devices (which is easy/OK for me at the moment as I am still setting everything up, but would be a huge PITA once I finish the set-up phase).

I believe this goes under [this FAQ question](https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time). Based on FAQ this should be adapter crash, however I am not sure, as the frontend remains accessible. Could you take a look at the logs and let me know what can I do to get more information to get to the bottom of this? Would it be beneficial to get some logs from the adapter and which logs? Thank you!

These are the Z2M logs preceding the crash, crash and subsequent failures to boot.
```
[2025-02-05 20:56:11] error: 	zh:controller:device: Read response to 0xbc026efffee1ad80 failed (ZCL command 0xbc026efffee1ad80/1 genTime.readRsp({"localTime":792098414}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 21:57:08] error: 	zh:controller:device: Read response to 0xbc026efffea7ff3c failed (ZCL command 0xbc026efffea7ff3c/1 genTime.readRsp({"localTime":792098657}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 22:01:50] error: 	zh:controller:device: Read response to 0xbc026efffee19beb failed (ZCL command 0xbc026efffee19beb/1 genTime.readRsp({"localTime":792098658}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 22:05:50] error: 	zh:controller:device: Read response to 0xbc026efffee1bfe0 failed (ZCL command 0xbc026efffee1bfe0/1 genTime.readRsp({"localTime":792098415}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 22:14:09] error: 	zh:controller:device: Read response to 0xbc026efffea7f20f failed (ZCL command 0xbc026efffea7f20f/1 genTime.readRsp({"localTime":792098415}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 22:14:33] error: 	zh:controller:device: Read response to 0xbc026efffee1ad91 failed (ZCL command 0xbc026efffee1ad91/1 genTime.readRsp({"localTime":792098414}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 22:17:45] error: 	zh:controller:device: Read response to 0xbc026efffee1ad75 failed (ZCL command 0xbc026efffee1ad75/1 genTime.readRsp({"localTime":792098658}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 22:29:21] error: 	zh:controller:device: Read response to 0xbc026efffee1d32d failed (ZCL command 0xbc026efffee1d32d/1 genTime.readRsp({"localTime":792098658}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 23:46:31] error: 	zh:controller:device: Read response to 0xb43522fffedf8550 failed (ZCL command 0xb43522fffedf8550/1 genTime.readRsp({"localTime":792098658}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-05 23:57:38] error: 	zh:controller:device: Read response to 0xb43522fffedf7378 failed (ZCL command 0xb43522fffedf7378/1 genTime.readRsp({"localTime":792098658}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-06 00:30:27] error: 	zh:controller:device: Read response to 0xbc026efffee1ad91 failed (ZCL command 0xbc026efffee1ad91/1 genTime.readRsp({"localTime":792098657}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":26,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
Error: ZCL command 0x3410f4fffe208b1b/1 genTime.write({"timeStatus":1,"time":792103396,"timeZone":3600}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:252:45
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:26)
    at ZStackAdapter.dataRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1046:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:509:35)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:334:20)
[00:31:17] INFO: Preparing to start...
[00:31:17] INFO: Socat not enabled
[00:31:17] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-06 00:31:22] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-06 00:31:22] info: 	z2m: Starting Zigbee2MQTT version 2.1.0 (commit #unknown)
[2025-02-06 00:31:22] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-06 00:31:22] info: 	zh:zstack:znp: Opening TCP socket with 10.1.42.3:6638
[2025-02-06 00:31:22] info: 	zh:zstack:znp: Socket connected
[2025-02-06 00:31:22] info: 	zh:zstack:znp: Socket ready
[2025-02-06 00:31:22] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-06 00:31:23] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-06 00:31:41] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-06 00:31:41] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-06 00:31:41] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-06 00:31:41] error: 	z2m: Exiting...
[... cut ...]

### Setup

Add-on on Home Assistant OS on RPi4

### Debug log

I started setting up my network couple of weeks ago (doing it slowly) and have added 50+ devices so far. Recently I've had issue of very slow network and unable to use the add-on so I turned off debug logging and additionally turned off info logs for z2m:mqtt which fixed that issue. If there are any tips how to enable debug logging and keep everything functional (so the logging does not impact the actual result) please let me know.

vilaemail, 2025-02-07:
This is a consistent repro for me, it crashed again after ~4 hours. I will try to set-up debug logging for tonight in hopes that it will crash with the same issue and not some other issue due to performance impact. If someone has any idea based on logs I have already posted, please let me know. Thanks!

HenrikHills, 2025-02-07:
Im getting similar errors, which eventually lead to BUFFER_FULL errors, and only solveable by reseating the Sonoff-dongle (restart doesnt do anything), started to happen after updating Z2M and triggering OTA updates

Zigbee2MQTT version
    [2.1.1]commit: [unknown]
Coordinator type
    zStack3x0
Coordinator revision
    20240710
Frontend version
    0.9.4
zigbee-herdsman-converters version
    21.31.0
zigbee-herdsman version
    3.2.5

Koenkk, 2025-02-07:
@vilaemail this error means z2m cannot reach the adapter anymore, it is likely a network connection issue (LAN)

19jm81, 2025-02-07:
Same errors here, had no issues before

19jm81, 2025-02-07:
I get this message since the latest update:

zh:controller:device: Read response to 0xa4c13885d66a288d failed (ZCL command 0xa4c13885d66a288d/1 genTime.readRsp({"localTime":792285099}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":228,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))

vilaemail, 2025-02-10:
> [@vilaemail](https://github.com/vilaemail) this error means z2m cannot reach the adapter anymore, it is likely a network connection issue (LAN)

Thanks for the reply. I have moved some of the SPM01 to another Zigbee network (as Rene on discord suggested they are putting a strain on the network and I have seen them send a lot of messages in my sniff) and I no longer experience crashes (3+ days). For the LAN network I don't see any packet drops/errors. Given this two information I think that issue lies in the adapter and/or network, I am thinking it is "crashing" somehow.

What additional info could help diagnose issues? Would network sniff at the time when issue happens help? Is there any way to get additional (useful) logs from the adapter?

Is it possible to run 20240710 firmware at 4x baud rate? (I've tried but Z2M fails to connect). I am asking as I've seen network speeds (Tx+Rx) on the LAN interface reach the 110kbaud on my switch, so I thought maybe adapter<->Z2M connection can not keep up.

Koenkk, 2025-02-10:
Could you try with this firmware? 

[CC1352P7_coordinator_99240710_baudrate_460800.hex.zip](https://github.com/user-attachments/files/18740197/CC1352P7_coordinator_99240710_baudrate_460800.hex.zip)

vilaemail, 2025-02-12:
I still have the problem. Here is the update:

1. Changing the firmware didn't help, my **assumption** is that SPM01 routers cannot keep up with their messages.
2. Based on [this](https://github.com/Koenkk/zigbee2mqtt/issues/22236#issuecomment-2325492263) I've bought Tuya gateway, updated SPM01 firmware to 3.1.3 and reconfigured them to update at most once every 10 seconds. I can confirm in Z2M using `last_seen` that devices go from "Just now" to "10/11/12/13/14 seconds ago" and then back to "Just now".

It is no longer crashing, but all devices go offline and after restarting Z2M, it fails to reconnect to the adapter. I've enabled debug logging and started a network sniff, I will post more info as I have.

vilaemail, 2025-02-14:
So I have debug logs and sniff. 

Here are my findings:
- In this case all devices stopped responding in 15 second window. In the sniff that is roughly 4830-4845 seconds. This corresponds roughly to 21:39:15 - 21:39:28 in the logs. (Other times I saw devices slowly going off network, but I don't have sniff for that only logs, I can upload if needed).
- I can see in the sniff that SPM01 device is happily sending messages as usual for ~3 more minutes before starting to ask for route to coordinator.
- Coordinator goes silent on 4842s apart from 5 more messages that i recorded [see here for last 10+ messages](https://github.com/user-attachments/assets/21171b7f-9274-44b2-a5f5-8e9bb8ea5e03), unsure if this is relevant. Messages at 4842s seem normal to what I've been seeing in the sniff before. At the moment of going silent I didn't see anything different in the sniff than before.
- I additionally isolated last messages Z2M tried to send to 2 devices. And i can find those messages in the sniff
- As seen in Z2M logs eventually it tries to ping devices (set availability check for 30 minutes), however I do not see those messages in the sniff.

Here are the logs:

[log_3.log](https://github.com/user-attachments/files/18802496/log_3.log)

Any ideas?

Koenkk, 2025-02-15:
It seems the coordinator simply cannot catch up with the amount of messages, these devices are spamming a lot. In a time span of +- 100 seconds, z2m receives 4500 messages from the coordinator (0.02 sec per message) (`zh:zstack:znp: <--) and sends 826 (0.12 sec per message) (`zh:zstack:znp: -->`). Besides removing these devices from your network there are 2 things that we can try:
1. Try with this fw for your coordinator: https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226
2. Reduce the amount of message send from the coordinator (have to dive a bit deeper if this is possible), let's first try 1

Koenkk, 2025-02-17:
Its not P7 compatible indeed, could you try this one: https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2646590166 ? (CC1352P7 fw works on CC2652P7)

vilaemail, 2025-02-17:
I've flashed it, it is running, let's see what happens. Could you share what is different about the firmware that might affect this issue? Thanks!

Koenkk, 2025-02-17:
@vilaemail this fw has the settings of the 20221226 with the 20240710 SDK (https://github.com/Koenkk/Z-Stack-firmware/issues/518)

vilaemail, 2025-02-18:
It's been running without issues for 22 hours. I will test more in the following days and reply back here with findings.

vilaemail, 2025-02-27:
There is a lot of debug logs and I want to avoid wearing SD card (as I'd need to keep it running for days to have it crash), so I needed to disable them. When I get a chance, I will stand up separate VM/service for logging and send logs there, which would get us debug logs. Then I will reply here with more info.

I will update issue title to be specific it is about this particular device (as right now all devices on this network are the same).

FWIW I've set-up automation to detect when devices are offline or addon crashed and to then restart addon and power cycle adapter in particular sequence so that it comes back online. I have it counting number of resets to track how often it happens. So, I kind of found a workaround + monitoring.

In case anyone is interested below is the code. You will need "network state" entities and apply label on those (I've used "z2m_edge_critical_devices"). To create network state entities [follow this guide](https://community.home-assistant.io/t/add-network-state-online-offline-to-every-z2m-device-in-3-easy-steps/835532) or look at my example below. You will also need a way to power cycle the adapter. I am using Mikrotik PoE switch (CSS610 firmware SwOS Lite version 2.18) so I stood up a node.js server with REST API to allow querying state and commanding the PoE links on/off. Server is listening on port 3000 and nginx is reverse proxying into it and providing TLS + basic auth for HAOS to access. It is less than ideal as SwOS Lite does not provide HTTPS endpoint and I'd like to use mTLS for HAOS->NGINX, but it is what it is. Potentially I could just send Zigbee + ESP32 reset commands to UZG-01 (XZG) frontend, however I wanted to integrate the switch into HAOS anyway, so I chose that route. I know there is Mikrotik integration for HAOS, but I dislike adding so many permissions (or in case of this switch - auth used over HTTP with all permissions) to an account which is used directly from HAOS and allowed to talk to network gear, so I cobbled-up the VM serving the REST API.

[automation.txt](https://github.com/user-attachments/files/19010172/automation.txt)
[mqtt.txt](https://github.com/user-attachments/files/19010219/mqtt.txt)
[rest_switch.txt](https://github.com/user-attachments/files/19010257/rest_switch.txt)
[start.js.txt](https://github.com/user-attachments/files/19010323/start.js.txt)
[package.json](https://github.com/user-attachments/files/19010341/package.json)
[nginx_http.d.txt](https://github.com/user-attachments/files/19010387/nginx_http.d.txt)

vilaemail, 2025-04-29:
This is still happening, in last 4 days my automation did ~20 restarts. I will try to come up with logs and update in the coming weeks.

closed 2025-07-07 by github-actions[bot]