Koenkk/zigbee2mqtt #26226: PTM215ZE  (Zigbee Green Power) only works a few minutes

maltejahn, 2025-02-07:
### What happened?

I bought a Enocean PTM215ZE to try a these devices. After having some trouble to pair them, finally the device appeared. And worked for some minutes.
Afther these few minutes, the device is still visible in Z2M, but doenst react anymore.

When i use a second Stick (Conbee2) with ZShark + Wireshark on a different PC, i still see that data is transmitted when pressing a button
If I just do a tail -f on the current log file(debug enabled, on the raspberry) nothing appears when pressing a button

I used this firmware:
Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip

### How to reproduce it (minimal and precise)

remove it from Z2M (including force remove)
reset the device using the Enocean app (NFC)
Pair again
-> doesnt react to button press after few minutes

### Zigbee2MQTT version

2.1.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle Plus P ZStack

### Setup

Standalone Raspberry 3B

### Debug log

[image]

chris-1243, 2025-02-07:
> Update:

> Found this after a while -> switch works of z2m is in pairing mode. If pairing Mode is off - it stop working:
#14198 (comment)

This happens as you did not strictly follow the pairing procedure. As you are trying to pair a PTM215ZE and as you are using the Enocean app, I would use this app as well for the pairing. It is really much easier.
- Open your app and scan your PTM. Select your network and click `apply`. Once the data is sent to the device, click one time on the energy_bar. This has to be done for the settings to be applied by your device.
- Scan again your device and go to `installation`. Click on `Execute` and nothing else.
- Z2M: choose a router able to [translate ZGP frames](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices). Instead of allowing `permit_join all`, choose your dedicated router `permit_join my_router`. Now permit join is active, press `energy_bar` and if everything was done correctly, your PTM215ZE should appear and the pairing process is done!

Otherwise, strictly follow the pairing instructioj [there](https://www.zigbee2mqtt.io/devices/PTM_215ZE.html).

> Update 2:

> I am lost:

> trying to "directly" pair it with the coordinator doenst help/work
Whas ist meant by "switch to unicast"

`zstack` is not able to pair such devicd directly. There is a must to use a dedicated router. `ember` is able in the other hand. This how firmware and other parameters are implemented by the manufacturer if the chip...
`unicast` is a mean to send a message directly from the device via the router to the coordinator. `broadcast` will do this but your message will be routed by all your router...It may slow down a lot your network. Not all router are able to send ZGP frames as `unicast`. Check the link above.

> Update 3:

> SAm I wright -> Sonoff Dongle P doenst work as a coordinator?

Dongle-P works as coordinator or router with the good firmware.

maltejahn, 2025-02-10:
Hi,

lets recap for myself:
Dongle-P is fine as a Coordinator, but not without a router from this list (https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices) 
Enocean Tool: Channel 25 (as Coordinator), Commissioning(Sequence?): All channels; Direct: active

If its probably the solution to get e ZGP Router, than you can ignore the rest of the reply

I used a second Zigbee Stick: Conbee 3 on another Pi with Phoscon + HA on another PC (inside Proxmox). There i had no Problems to connect using the Enocean app. But i dont like to have another instance for Zigbee stuff. -> So, PTM215ZE is working fine

Because it worked with that stick, I used the Conbee3 with a fresh Raspberry (3B) installation + Z2M (switched Z2m + PTM251ZE back to channel 11 to avoid problems with the existing Zigbee network). This time, it also seems to be connected but i get a different error
`GREEN_POWER INDICATION - RangeError: Offset is outside the bounds of the DataView`

`
[2025-02-10 12:53:20] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt2/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"js8j3-5"}'
[2025-02-10 12:53:20] info:     z2m: Zigbee: allowing new devices to join.
[2025-02-10 12:53:20] debug:    zh:deconz:driver: send write parameter request from queue. seqNr: 32 paramId: 33 param: 254
[2025-02-10 12:53:20] debug:    zh:deconz:frameparser: write parameter response - parameter id: 33 - status: 0
[2025-02-10 12:53:20] debug:    zh:deconz: Permit joining on coordinator for 254 sec.
[2025-02-10 12:53:20] debug:    zh:deconz: zclFrame to all - 242
[2025-02-10 12:53:20] debug:    zh:deconz: sendZclFrameToAll - message send
[2025-02-10 12:53:20] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 33 request id: 6
[2025-02-10 12:53:20] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 34 request id: 7
[2025-02-10 12:53:20] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 6 status: 0
[2025-02-10 12:53:20] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:20] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 7 status: 0
[2025-02-10 12:53:20] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:21] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2025-02-10 12:53:21] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:21] debug:    zh:deconz:driver: query aps data confirm
[2025-02-10 12:53:21] debug:    zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 35
[2025-02-10 12:53:21] debug:    zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 7 confirm status: 0
[2025-02-10 12:53:21] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:21] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"js8j3-5"}'
[2025-02-10 12:53:21] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2025-02-10 12:53:21] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:21] debug:    zh:deconz:driver: query aps data confirm
[2025-02-10 12:53:21] debug:    zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 36
[2025-02-10 12:53:21] debug:    zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 6 confirm status: 0
[2025-02-10 12:53:21] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:22] debug:    zh:deconz:driver: send read device state from queue. seqNr: 37
[2025-02-10 12:53:22] debug:    zh:deconz:frameparser: device state: 100010
[2025-02-10 12:53:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:26] debug:    zh:deconz:frameparser: GP commissioning notification
[2025-02-10 12:53:26] debug:    zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x156b9e8 cmd id: 0xe0 frameCounter: 0xa9c
[2025-02-10 12:53:26] info:     zh:controller:greenpower: Received commissioning from '47592'
[2025-02-10 12:53:26] debug:    zh:controller:greenpower: Payload.Options: 58696 wasBroadcast: true
[2025-02-10 12:53:26] debug:    zh:deconz: zclFrame to all - 242
[2025-02-10 12:53:26] debug:    zh:deconz: sendZclFrameToAll - message send
[2025-02-10 12:53:26] debug:    zh:deconz:frameparser: GP commissioning notification
[2025-02-10 12:53:26] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 38 request id: 8
[2025-02-10 12:53:26] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 8 status: 0
[2025-02-10 12:53:26] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-10 12:53:27] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2025-02-10 12:53:27] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
[2025-02-10 12:53:35] debug:    zh:deconz:frameparser: GREEN_POWER INDICATION - RangeError: Offset is outside the bounds of the DataView
[2025-02-10 12:53:35] debug:    zh:deconz:frameparser: GP commissioning notification
[2025-02-10 12:53:35] debug:    zh:deconz:frameparser: GREEN_POWER INDICATION - RangeError: Offset is outside the bounds of the DataView
[2025-02-10 12:53:35] debug:    zh:deconz:frameparser: GP commissioning notification
[2025-02-10 12:53:35] debug:    zh:deconz:frameparser: GREEN_POWER INDICATION - RangeError: Offset is outside the bounds of the DataView

`

chris-1243, 2025-02-10:
Well, according to the documentation, `deconz` driver is not fully supported in Z2M.

https://www.zigbee2mqtt.io/guide/adapters/deconz.html

There are several threads opened with various errors in regard of ZGP devices. The problem here is not the stick but the implementation in the software.

Definitely, I would use either TI chip (`zstack`) or Silab chip (`ember`).

**zstack**: you must use a router to translate

**ember**: normally, you may pair a ZGP device directly to the coordinator

maltejahn, 2025-02-11:
Hi,

thank you. After reading your replies with the documentation it worked immediately:
In the Enocean Tool i changed: Channel 25 (as Coordinator), Commissioning(Sequence?): All channels; Direct: active
+ pressed one button at the end so the changes are adopted. And checked the settings by rereading the nfc

Still using Dongle-P but got a Ikea E1702 or so from a friend. I paired the Ikea outlet to my zigbee network. After this, I enabled pairing (not to all or to the coordinator -> pairing to the ikea outlet)

Then I used the Enocean tool to start pairing and pressed a button. The PTM215ZE  appeared immediately and still work since a few hours.

Now I have to figure out how to implement this into Home Assistant. HA sees the switch, but only Link Quality and last seen is visible. No button entry shows, I cant even see a Message under /homeassistant topic
Under /zigbee2mqtt topic I have messages like this
{"action":"release_3","last_seen":"2025-02-11T18:34:59+00:00","linkquality":204}
But not under /homeassistant/button or ./switch

Probably i have to use NodeRed.. e.g. i try this as a starting point:
https://github.com/Koenkk/zigbee2mqtt/issues/13361

chris-1243, 2025-02-11:
Would you mind leave this thread open ? I now have some requests and suggestions for you...

Working blueprint for your PTM225Z/ZE (made by myself with the help of the HA community. This is what I use.)

https://github.com/chris-1243/HA---Blueprint

As you were able to sniff your trafic, would it be possible to check if the IKEA E1702 is used in `unicast` mode and what cluster (input/output) you have on endpoint 242 ?

If paired as `unicast`, you should not lose any messages. It might happen you receive `press_energy_bar`. This means you did not press the button correctly (you press the external corner). Or, an other sequence like `press_1_and_3` because your finger may be to close of two buttons...

maltejahn, 2025-02-12:
somehow i closed the "issue" accidentally

chris-1243, 2025-02-12:
I should have been a bit more precise, my apologize.

Could you apply simple filter `zbee_zcl` in Wireshark (you will get more traffic captured). Now, when you press a button, you should see a GreenPower notification

[image]

Click on it and expend the field `Zigbee Cluster Library Frame` and check `Options`

[image]

If you may send a screenshot like above, it would be awesome.

When you pair your device, it is really important to select a specific router. In your case, the IKEA E1702. Select `permit join - address/name of the E1702`. By doing so, you force the `unicast` mode and avoid some troubles in the future.

The cluster might be seen in the device tab in Z2M frontend

maltejahn, 2025-02-12:
Hi,

sadly, there is absolutely no output when appyling the filter "zbee_zcl" and press some button.
I paired the device again to 0x7444 -> ikea outlet. Still "broadcast" appears in the log

As i read your posts I would have expected a Input Cluster for ZGP*

Screenshot 1: Cluster ot the E1703 (was wrong with E1702 *: maybe not close enough)
Screenshot 2: Cluster of the PTM215ZE
Screenshot 3: Commissioning

** Also these outlet have the "same" clusters: Nous A1Z, Aquara SP-EUC01 - just for the case I should try them instead

[image]
[image]

[image]

chris-1243, 2025-02-12:
E1703? This does not exist in Z2M database. Maybe you meant E1603? Anyway, this device should translate ZGP frames as `unicast`... 

In Wireshark, have you added your network key? It is in Edit --> Preferences --> Protocols --> Zigbee --> edit. Without this, you will not be able to decode completely the traffic. 
Check here: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#adding-the-trust-center-link-key

[image]

As it seems you are still in `broadcast`, I would remove the device, restart Z2M and pair it again selecting the IKEA router.

> ** Also these outlet have the "same" clusters: Nous A1Z, Aquara SP-EUC01 - just for the case I should try them instead

If you may check with these outlet, it would be great. My goal is to add more routers in [Zigbee2mqtt.io documentation](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices)

maltejahn, 2025-02-12:
Hi,

never wanted to dive that deep... :-)
Selecting to join one of these devices Nous A1Z, Aquara SP-EUC01 (Ikea 1703 removed from socket) results in a successful pairing, but switch only works during pairing is enabled. But if i put the ikea back in the socket it works fine

Screenshot 1: This time with the cluster library frame: 0x7444 Ikea E1703 (really, atleast its whats printed on the device, see below). 

If its the wrong device, maybe also adding a information that this device doenst work for sure?

[image]

[image]

chris-1243, 2025-02-12:
Ok, there are plenty model number of this device. They all share the same modelID.... 

https://zigbee.blakadder.com/Ikea_E1603.html

According to your last screenshot, your device is paired in `unicast` mode. Great!

In order to test other outlet, you must completely remove your PTM215ZE and pair it directly to the new router and check Wireshark output....

maltejahn, 2025-02-12:
Hi,

i remove them with "force" and do even a reboot for whatever reason. E.g. with the NOUS A1Z 0x073A (selected to pair with), it states that paring was successful. It also works until i disable paring mode. But after that, it doenst work  inside Z2M. Strange (for me) is, that something appears inside Wireshark (Screenshot 1). 

Screenshot 2
So, this bit "Also Unicast: True" is it that indicates Unicast. I thought when the switch src 0x15... -> does a broadcast (Step 1) that is to worry about. Then its forwarded by the ikea device 0x7444 finally to the coordinator (which says unicast)

I I can help out with any testing on these devices?

Thanks for your support

[image]

[image]

chris-1243, 2025-02-12:
Below, what you do not want to see. This is what is happening when you pair such device as `broadcast`. Also, the `Communication mode: Groupcast to pre-commissioned GroupID` is telling you you are in `broadcast`. And then, problem starts (slow down) as your network is saturated of message and according to specification, only 8 broadcast messages may be sent each second.

[image]

> So, this bit "Also Unicast: True" is it that indicates Unicast.

Yes

> I thought when the switch src 0x15... -> does a broadcast (Step 1) that is to worry about

No, you do not need to worry about... 

If you always need your IKEA plug to translate, it might tell us your other plug are not doing the job. Maybe, the firmware implementation is not done correctly. There are some known issues in Philips firmware where messages are always sent as `broadcast`. #22897

You may keep testing  your lights/plugs and if you find some able to translate ZGP frames, just let us know or you may edit the documentation in zigbee2mqtt.io... I am honestly trying to improve the documentation and help people getting these devices working as much as possible.

maltejahn, 2025-02-14:
Hi,

i just tried a Conbee 3 Stick with a Outlet "Nous A1Z" which i received a few days ago. I also installed it on a "new" Raspberry installation and I resetted the PTM215Z (so it is Channel 11)
1) pair the Nous A1Z
2) Pair PTM with Nous only....

It work even when pairing is disabled since 2 hours
Screenshot 1: Output with the setup above
Screenshot 2: Zigbee2Mqtt Info

Next try: using the "old" Nous A1Z on the new Raspberry/Z2M Installation
-> It only works during Pairing Mode (did a reset on the PTM215Z device)
Screenshot 3: The old Nous A1Z Outlet
Screenshot 4: The "new" one

-> New and old one is plugged in -> PTM215 Works without repairing or so
-> Old one only - doenst work
-> new one only  without re pairing -> works

Conclusion.. Nous A1Z can work, but I am not able to distinguish between good and bad

[image]

[image]

[image]

[image]

chris-1243, 2025-02-16:
Well, it is a Tuya bases device and Zigbee standard is not well followed by this brand unfortunately.

Thanks for trying 👍

Nerivec, 2025-04-09:
A lot has changed in GP in April release, so, not exactly sure what still applies, but my guess is the root cause of "works for a few minutes" is related to permit join being on/off: https://github.com/Koenkk/zigbee2mqtt/issues/25620

closed 2025-06-16 by github-actions[bot]