Koenkk/zigbee2mqtt #26227: Hue Dimmer Switch & Smartbutton missing Actions in HA.

ExquisiteMetropolis, 2025-02-07:
### What happened?

For some reason, HA is not listing all possible actions for the Hue Dimmerswitch & Smartbutton.

Increase/decrease and "On_press" are missing. Button doesn't work anymore now. 
Kind of hard to turn on the lights now. 

[image]

[image]

I do see when I press the button it's coming into Z2M, but for some reason it's no longer available in HA. 

Any clue to why this is? And how to fix it? :)
No changes were made to the automation. Must be something underwater. 

### What did you expect to happen?

When I press the button, like I used to, the lights turn on. 

### How to reproduce it (minimal and precise)

Press the button, nothing happens, check automation "on_press" action is missing from the triggers list.

### Zigbee2MQTT version

2.1.1.

### Adapter firmware version

8.0.2

### Adapter

Sonoff ZBDongle-E Zigbee

### Setup

Add-on via Home Assistant.

quadcom, 2025-02-10:
Yup, I just determined this is the reason none of my dimmers are working anymore.

No 'action' exists for the Hue 324131092621, 929002398602 or my IKEA E2001/E2002, E1524/E1810 dimmers.

It shows up in Z2M, but there is nothing equating to 'action' listed in HAOS Under Devices/MQTT, and none of the automation I have built functions anymore.

These kinds of glitches really way down the WAF.

[image]

[image]

bluemoehre, 2025-02-10:
I had the same issues with new devices the past days. 

But if I press all possible combinations on the button, it takes only a few minutes and then the events got listed in HA. Looks like HA is learning by doing. Anyhow, you can simply edit the YAML anytime which the correct values - that's not convenient, but works.

layemac, 2025-02-14:
Same for me with my Philipps dimmer switch and Ikea Rodret dimmer.

ExquisiteMetropolis, 2025-02-16:
> I had the same issues with new devices the past days.
>
> But if I press all possible combinations on the button, it takes only a few minutes and then the events got listed in HA. Looks like HA is [...]

You mention simply edit the YAML, for me that's not so simple. Which YAML do you mean and what should I add there?
(Until this bug is fixed).

Skons, 2025-02-16:
By creating an MQTT sensor, the action of the remote can be made available again
```yaml
- unique_id: 26227_action_workaround
  state_topic: "zigbee2mqtt/[name in z2m]/action"
```

layemac, 2025-02-16:
I remove this line : homeassistant_legacy_triggers: false in the configuration.yaml and it solve my problem.

ExquisiteMetropolis, 2025-02-17:
> I remove this line : homeassistant_legacy_triggers: false in the configuration.yaml and it solve my problem.

thanks, but that's not present in my Z2M configuration.yaml.

> By creating an MQTT sensor, the action of the remote can be made available again
> 
> - unique_id: 26227_action_workaround
>   state_topic: "zigbee2mqtt/[name in z2m]/action"

would I create this in the Z2M configuration.yaml? Or the HA equivalent?

Skons, 2025-02-17:
The fix of @layemac can be found here also

[image]

if you still want the action without that configuration @ExquisiteMetropolis , my configuration is a manual MQTT sensor
https://www.home-assistant.io/integrations/sensor.mqtt/

ExquisiteMetropolis, 2025-02-17:
Thanks, i will see if I can manage that. 
Following a reboot of my hardware, de off_press is now also gone in HA. 
Very strange. 

I do not prefer to use ha legacy options.
Hopefully I can manage to create it manually. Although I find it strange that there are not more complaints about this. I can imagine there are many more dimmer switches out there. What is the reason I am observing this behaviour v.s. others? 

Hopefully a permanent native Z2M (or HA) fix will come. Seems strange to be creating manual stuff for this.

closed 2025-02-18 by ExquisiteMetropolis

ExquisiteMetropolis, 2025-02-19:
@quadcom do you have an example of the blueprint you are using?

layemac, 2025-04-07:
Hello, I ended up changing all my buttons in nodered using the mqtt block (sorry, it's in French):

[image]

And here is the options :

[image]

You can use mqtt explorer to find the topics etc...

I think it's better to do this way because the fix of Skons will be removed at some point...