Koenkk/zigbee2mqtt #26229: Z2M fails to start with "device...must be string" when playing with "Settings/Home Assistent Name"

sbolten, 2025-02-07:
### What happened?

Added a "Home Assistent Name" through "Device --> Settings --> Home Assistent Name"
Removed the name through "Device --> Settings --> Home Assistent Name" (emptied the field)
Restart of Z2M fails with an error 

[09:57:48] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- devices/0x3410f4fffee1354c/homeassistant/name must be string
If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Looking at the configuration.yaml I found the following for this device:

  '0x3410f4fffee1354c':
    friendly_name: switch_3
    description: Studio Bad
    optimistic: true
    retain: true
    homeassistant:
      name: null

The crucial part is obviously the "name:null"

Replacing this with "homeassistant: {}" or some valid string (not null) solves the problem

### What did you expect to happen?

Z2M restarting without errors even though I fiddled around with the Home Assistant Name

### How to reproduce it (minimal and precise)

See "What happened"

Field I changed:
[image]

Config before change:
[image]

Changed the value of the field:
[image]

Config after change:
[image]

Removed string from field again:
[image]

Config after removing the string:
[image]

Restarting and error log of Z2M:
[image]
[image]

Editing the config back to its original situation (removing "name: null") solved the issue:

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on HAOS

closed 2025-04-16 by github-actions[bot]

SixFive7, 2025-08-02:
Still bugged on Zigbee2MQTT version [2.6.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.0).

SixFive7, 2025-08-22:
@Koenkk PR https://github.com/Koenkk/zigbee2mqtt/pull/28364 should fix it.