Koenkk/zigbee2mqtt #26230: Philips Hue dimmer switch (RWL021) OTA not working

Mar1usW3, 2025-02-07:
### What happened?

The switch reports a OTA Update available (Verion: 6.1.1.28573). I start the OTA but after some time it fails. I also get the switch active by pressing some buttons.

### How to reproduce it (minimal and precise)

click on the Update device firmware for the switch

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.4

### Adapter

HA Yellow with ember driver

### Setup

HA Yellow

### Debug log

```
info 2025-02-07 10:33:23z2m: Updating 'Esszimmer Schalter' to latest firmware

info 2025-02-07 10:33:26zhc:ota: [0x00178801105f9808 | RWL021] Getting latest image, meta: {"fileName":"Switch-ATmega_0012.sbl-ota","fileVersion":1124102917,"fileSize":240760,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Hue/Switch-ATmega_0012.sbl-ota","imageType":265,"manufacturerCode":4107,"sha512":"6bec6b6dce7ef9bb47c4467643222871788256d5c3f0aa88ded80be24fc002dbdda525ca2cafa78b996ff7fe0c18d7c9194288e4f6b40f0f37d147bc1724dd4e","otaHeaderString":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","originalUrl":"https://otau.meethue.com/storage/ZGB_100B_0109/3a1c8cf8-3f4c-4464-93f0-24cc7f67f0d7/Switch-ATmega_0012.sbl-ota"}

error 2025-02-07 10:35:56z2m: Update of 'Esszimmer Schalter' failed ([0x00178801105f9808 | RWL021] Timeout. Device did not start/finish firmware download after being notified. ({"target":18185,"apsFrame":{"clusterId":25,"profileId":260,"sequence":0,"sourceEndpoint":1,"destinationEndpoint":2,"groupId":0,"options":0},"commandIdentifier":3} timed out after 150000ms))
```

chenks, 2025-02-07:
came there to report the same

```
[2025-02-07 10:28:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer Switch (Bedroom)/action', payload 'on_press'
[2025-02-07 10:28:19] info: 	z2m: Updating 'Hue Dimmer Switch (Bedroom)' to latest firmware
[2025-02-07 10:28:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer Switch (Bedroom)', payload '{"action_duration":null,"battery":93.5,"linkquality":128,"update":{"installed_version":1107324829,"latest_version":1124102917,"state":"available"}}'
[2025-02-07 10:28:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Hue Dimmer Switch (Bedroom)' failed ([0x0017880102d4ae38 | RWL021] Failed to find an endpoint which supports the OTA cluster)","status":"error"}'
[2025-02-07 10:28:19] error: 	z2m: Update of 'Hue Dimmer Switch (Bedroom)' failed ([0x0017880102d4ae38 | RWL021] Failed to find an endpoint which supports the OTA cluster)
```

rmi1974, 2025-02-07:
Hello folks,

can confirm. The old version of the dimmer switch:

https://www.zigbee2mqtt.io/devices/324131092621.html

and new version (v2) fail.

https://www.zigbee2mqtt.io/devices/929002398602.html

All of my Philips dimmer switches (1x old) and (10 x new v2) refuse to OTA update.

```
[2025-02-07 19:02:57] info: 	z2m: Updating 'Wohnzimmer Dimmerschalter' to latest firmware

[2025-02-07 19:04:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Dimmerschalter', payload '{"action":null,"action_duration":null,"battery":60.5,"device":{"applicationVersion":2,"dateCode":"20190410","friendlyName":"Wohnzimmer Dimmerschalter","hardwareVersion":1,"ieeeAddr":"0x0017880103cbb45d","manufacturerID":4107,"manufacturerName":"Philips","model":"324131092621","networkAddress":29371,"powerSource":"Battery","softwareBuildID":"6.1.1.28573","stackVersion":1,"type":"EndDevice","zclVersion":1},"last_seen":"2025-02-07T17:34:16.274Z","linkquality":80,"update":{"installed_version":1107324829,"latest_version":1124102917,"state":"available"},"update_available":false}'

[2025-02-07 19:04:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Wohnzimmer Dimmerschalter' failed (Device didn't respond to OTA request)","status":"error","transaction":"zd55w-4"}'
```

```
{
    "battery": 100,
    "last_seen": "2025-02-07T19:28:13.413Z",
    "linkquality": 94,
    "update": {
        "installed_version": 33565954,
        "latest_version": 33574183,
        "state": "available"
    },
    "action": null,
    "device": {
        "applicationVersion": 2,
        "dateCode": "20210729",
        "friendlyName": "HG OG1 Wohnzimmer Lichtschalter Nord",
        "hardwareVersion": 1,
        "ieeeAddr": "0x001788010e8ce559",
        "manufacturerID": 4107,
        "manufacturerName": "Signify Netherlands B.V.",
        "model": "929002398602",
        "networkAddress": 23744,
        "powerSource": "Battery",
        "softwareBuildID": "2.45.2_hF4400CA",
        "stackVersion": 1,
        "type": "EndDevice",
        "zclVersion": 2
    }
}
```

Regards

mundschenk-at, 2025-02-07:
@rmi1974 Did you wake up your v2 dimmers by pressing a button? With that, mine all updated. The v1 dimmers did not. I got one to update by making it perma-awake via reporting settings (1/5/0), but hat to re-interview afterwards to get the new FW recognized by Z2M.

jonofe, 2025-02-08:
I also had problems in updating my old v1 dimmers, but finally 2 of them are updated now and one ist still in the process of updating. You need a lot of patience, always restart, do not more than 1 device update in parallel. Keep device alive. Press a button immediately before clicking the Ok button to start the update. It took me the entire day to get 2 out of 3 updated. 

Newer dimmers and also the old indoor motions sensors as well as the newer outdoor motions sensors updated without problems.

mundschenk-at, 2025-02-08:
I beliebe the difference is that the v1 accessories are not ZigBee 3.0 conformant, whereas the newer ones are.

chenks, 2025-02-08:
I have 4 v1 and 1 v2 of the switches. 

The v2 updated fine. 3 of the v1's appear to update but the following morning say they need update again, they will go thru the process of updating and appear to update but again will claim they need updated the following morning

1 of the v1's point blank refuses to update giving the log errors I posted earlier. 

I never had issues updating these the last time there was updates for them.

rmi1974, 2025-02-08:
@mundschenk-at  thanks for the hint. I forgot about the sleep mode on those devices. 

After running around, pressing one of the buttons once/twice, the v2 devices were able to update from:

20210729	[2.45.2_hF4400CA](https://www.philips-hue.com/en-us/support/release-notes/accessories)

20241001	[2.77.39](https://www.philips-hue.com/en-us/support/release-notes/accessories)

Related to this "deep sleep" mode, but off-topic:

This (deep) sleep mode on those philips dimmer switch devices is really annoying. I have various other multi-button zigbee remotes (tuya), that never show this behaviour, reacting instantly on button presses.
Especially when people need to press buttons at least twice because the device needs a "wakeup" to have some action done because the first button press is "lost" (required to wakeup?)
Annoying for guests who expect the switches to operate like normal physical switches.
Can someone tell me if it's possible to disable the sleep mode on those dimmer switches. I'd rather have to drain them batteries faster than the "this switch doesn't work properly" experience.

Regards

chenks, 2025-02-08:
I'm using a sonoff e controller and it's already on the most ember firmware

nephrotranz, 2025-02-09:
I was able to update one of my Hue v1 dimmer switches, even at the first try, funnily the firmware version stayed the same after the update.

Still trying to update the other two, one of them got until 52%, but failed then, the third one didn't even start to update until now.

chenks, 2025-02-09:
It didn't update, it failed. 
Tomorrow morning it'll be back in the list to showing update available

nephrotranz, 2025-02-09:
> It didn't update, it failed. Tomorrow morning it'll be back in the list to showing update available

The update was already three days ago and it definitely went to 100%.

chenks, 2025-02-09:
If the firmware version didn't change then it didn't update

I've been thru this for 3 days running. Each switch claims to have update but the next morning they all appear again saying update available

nephrotranz, 2025-02-09:
> If the firmware version didn't change then it didn't update
> 
I think you could be right, there are no clusters available anymore in the dev console, so perhaps the update broke it, although it is fully working.

nephrotranz, 2025-02-09:
New update, it now fully stopped working in Zigbee2MQTT although I can connect it. 
But when connected there is simply no light flashing when I press any of the buttons.

So I really do not know what happened.

I will connect it to the Hue Bridge again now and will not try to update the other switches.

chenks, 2025-02-09:
To confirm my status, I only have issues with v1 hue switches. 

My v2 switches, hue bulbs and hue motion sensor all updated fine. 

It's just my 4 v1 switches that won't update - 3 appear to update and complete but the following morning they appear for update again and 1 refuses to even start to try and update

jonofe, 2025-02-10:
> If the firmware version didn't change then it didn't update
> 
> I've been thru this for 3 days running. Each switch claims to have update but the next morning they all appear again saying update available

I made the same experience and I tried it 3 times. 

3 times the dimmer v1 was updated up to 100%, but the version did not change.

Finally I found out, if you trigger an new interview after "successfull" (100%) update , the new firmware version is recognized and the dimmer shows the updated FW version in the z2m portal.

chenks, 2025-02-10:
> I made the same experience and I tried it 3 times.
>
> 3 times the dimmer v1 was updated up to 100%, but the version did not change.
>
> Finally I found out, if you trigger an new interview after [...]

what do you mean by that?
you mean you have to sit and watch the live log until you see successfull" (100%) update, and then interview it again?
that is going to be quite time consuming if you have to multiple switches.

jonofe, 2025-02-10:
> what do you mean by that?

simply do an re-interview after the update has finished without errors. 
If the update breaks with errors, this obviously does not help.

chenks, 2025-02-10:
> > what do you mean by that?
> 
> simply do an re-interview after the update has finished without errors. If the update breaks with errors, this obviously does not help.

all 4 of my v1 switches aren't showing a update fail, but a re-interview does nothing.

jonofe, 2025-02-10:
> all 4 of my v1 switches aren't showing an update fail, but a re-interview does nothing.

okay, then it's not a general solution. For my last v1 dimmer at least it worked out.

bart1, 2025-02-10:
I seem to experience the same issue
```
info 2025-02-10 23:32:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Switch Slaapkamer Ouders' failed ([0x0017880102c1b40b | RWL021] Timeout. Device did not start/finish firmware download after being notified. (Timeout - 57453 - 2 - undefined - 25 - 3 after 150000ms))","status":"error","transaction":"nhuqw-2"}'
error 2025-02-10 23:32:53z2m: Update of 'Switch Slaapkamer Ouders' failed ([0x0017880102c1b40b | RWL021] Timeout. Device did not start/finish firmware download after being notified. (Timeout - 57453 - 2 - undefined - 25 - 3 after 150000ms))
```

ejpenney, 2025-02-10:
Adding my voice to the cacophony:

I have 5 V1 switches, I wake one of them up by pressing buttons before and after I tell Z2M to update and the firmware upgrade seems to work, but then the next day updates are available again.  Obviously one device at a time is safest...

I've tried using the "Interview" button in the Z2M web UI to re-interview the devices, alongside the same method of "waking up the device" by pressing buttons for a few seconds before and after... This doesn't seem to be doing anything.  Is there a different way we should be waking up/preparing these V1 dimmers for re-interview?  Perhaps the old paperclip in the back, or holding On/Off for 10 seconds?

Otherwise I'm stuck in the same boat as a few others here, firmware update appears to work but either the firmware doesn't actually apply or I'm not getting the correct firmware version from the switches.  It's hard to tell, either way Z2M doesn't seem to believe anything is amiss.

Sonoff E, running zStack3x0 (revision 20230507), Z2M 2.1.1

chenks, 2025-02-11:
> > what do you mean by that?
> 
> simply do an re-interview after the update has finished without errors. If the update breaks with errors, this obviously does not help.

i tried your suggestion, it didn't work.
i triggered an update on the switch, watched the log to see it complete to 100%.
re-interviewed the switch by pressing the blue "i" button within zigbee2mqtt devices screen. it said it completed.
waited 24 hours and the device re-appeared in the list of devices requiring an update.

flotux2, 2025-02-11:
I had the same issue with the Philips Hue dimmer switch V2. For me it helped to trigger the update in Z2M and then click twice the off button on the dimmer. Then after few seconds the update progress started - first with time N/A and then later with ~ 30 minutes. It worked for my 4 dimmer switches the same way. I did not have to re-add them.

chenks, 2025-02-11:
> I had the same issue with the Philips Hue dimmer switch V2. For me it helped to trigger the update in Z2M and then click twice the off button on the dimmer. Then after few seconds the update [...]

this issue is related to v1 switches, not v2.
v2 switches have no issues updating.

tomtom13, 2025-02-11:
I've been trying to update my V1 switches for week or so. When I try to reconfigure to check whenever firmware update been successful, I get this:

```
2025-02-11 13:31:33z2m: Failed to configure 'Staircase 2nd floor switch 2', attempt 2 (Error: Bind 0x0017880108f50b19/1 genOnOff from '0x00124b0029e7e16a/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23) at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43) at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/reporting.ts:42:9) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/devices/philips.ts:2379:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21))
```

nephrotranz, 2025-02-11:
So my final status is the following now:

**Hue Dimmer Switch v1 No. 1:**
Successfully updated with Z2M after 2 tries
Z2M is still showing the old firmware version and date, but when I connect it to the Hue bridge the Hue app shows the new firmware version
Re-connection to Z2m does not change it, BUT when I search for new firmware via OTA, it says there is no update available

**Hue Dimmer Switch v1 No. 2:**
I had no chance updating it with Z2M, failed 4 or 5 times
After connecting it to the Hue bridge two days ago the old firmware version was shown in the Hue App,  I had to wait until tonight, but in the end the switch was updated to the new firmware version via the Hue bridge.
BUT now comes the big surprise - I re-connected it to Z2M, but here it still showing the **old** firmware date and version although it was updated.
Searching for updates also brings the result that no new firmware is available

So the conclusion is the update for v1 is possible with Z2M, but it can be a pain in the a**
The frontend does wrongly show the old firmware version and firmware date, but when searching for an update it is recodnized that the switches are updated to the newest version.

mundschenk-at, 2025-02-11:
@nephrotanz: Yeah, if Z2M says there are no updates available,  you can just reinterview to show the correct version.

tomtom13, 2025-02-11:
I only have v1 dimmers. It's been a right pain since the new version showed up.
I've took out my old hue hub and connected two dimmers, and they updated OK. Problem is that hub firmware got updated as well and now I can't connect any of my v1 switches with old firmware on it. So I'm stuck with that list in Z2M:

[image]

nephrotranz, 2025-02-11:
> [@nephrotranz](https://github.com/nephrotranz): Yeah, if Z2M says there are no updates available, you can just reinterview to show the correct version.

I deleted and re-connected the switches several times, re-configured them several times - no chance, both switches still showing 6.1.1.28573 (as long as they do work I do not have any problem with it).

tomtom13, 2025-02-11:
> > Problem is that hub firmware got updated as well and now I can't connect any of my v1 switches with old firmware on it. So I'm stuck with that list in Z2M:
> 
> That's strange. My Hue bridge is always up to date as it is still in use, no problem connecting the ones with the old firmware.

I don't know what's going - however after updating those two dimmers, now I can't pair any new switch with this hub. I've even tried changing the channel on the hub. Maybe I'll try to pair a bulb or something to it. It's just puzzling to me as well. (and yes I've jumped through all the burning hoops of dimmer factory reset)

tomtom13, 2025-02-11:
OK, so adding a bulb to the hub allowed me to pair more switches. However two of the switches that I've just paired show up as having a new version - while those were showing the old version on Z2M. After removing those from HUE hub, and paring back to Z2M, I get this error while z2m is doing it's automatic interrogation:

```
[2025-02-11 17:53:49] error: 	z2m: Failed to configure '0x0017880108711046', attempt 1 (Error: Bind 0x0017880108711046/2 genPowerCfg from '0x00124b0029e7e16a/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43)
    at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16)
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30)
    at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/devices/philips.ts:2384:13)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21))
[2025-02-11 17:54:10] error: 	z2m: Failed to configure '0x0017880108711046', attempt 2 (Error: Bind 0x0017880108711046/1 genOnOff from '0x00124b0029e7e16a/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43)
    at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16)
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30)
    at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/reporting.ts:42:9)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/devices/philips.ts:2379:13)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21))
```

tomtom13, 2025-02-11:
@Koenkk - I think this is something for you that automatic interrogation doesn't work.

Koenkk, 2025-02-12:
Does the new version show up when force removing the device through the z2m frontend, restarting z2m and re-pairing the device?

tomtom13, 2025-02-12:
> Does the new version show up when force removing the device through the z2m frontend, restarting z2m and re-pairing the device?

Nope. Seems that:
1. Old information of the device is stuck to it when -> removing -> conecting to different hub (hue hub) to upgrade -> reconnecting (even with HASS reboot in the meantime) 
2. doing the manual "reconfigure" throws an error
3. upgrade fails for some of the devices to engage
4. upgrade seems to proceed up to 100% without an error, and then device is stuck in old version and in the page it still claim that update is possible. What's funny here is that one can try and upgrade one dimmer several times without an error in a row.
5. Some devices alto have an low version number - when hitting "check for update" either throw an error or don't throw an error are are stuck there.

Out of people that reported this issue I think I've got the largest set of those dimmer switches to play with so let me know what shall I test.

Here are my stats from hass z2m page:
Zigbee2MQTT version
[2.1.1] commit: [unknown]
Coordinator type
zStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
REDACTED
Frontend version
0.9.4
zigbee-herdsman-converters version
21.31.0
zigbee-herdsman version
3.2.5

tomtom13, 2025-02-13:
OK, so I've performed the test exactly how you've advised and it seems that new version got registered:
Froce remove -> reboot HASS VM to get Z2M rebooted fully -> repair the dimmer = new firmware version show up.

nephrotranz, 2025-02-13:
> Does the new version show up when force removing the device through the z2m frontend, restarting z2m and re-pairing the device?

Yes, this worked for me with the first try. Thank you.

JensjJ8200, 2025-02-13:
I got this in my Zigbee2mqtt, when trying to update.. Does that help in finding out, where the problem is?

[image]

JensjJ8200, 2025-02-13:
I wanted to add another comment. 
I have two different dongles. One is silicon labs and the other is texas instruments. I use my silicon labs dongle for production, and the texas instrument for an test invorement. I could not update on the production, but when I moved the dimmer to the test coordinator, then it updated straight away. So could the problem be with the ember software? or perhaps its pure luck from my side.

mundschenk-at, 2025-02-13:
>  So could the problem be with the ember software? or perhaps its pure luck from my side.

Pure luck, or maybe less traffic on the test network.

nephrotranz, 2025-02-13:
> I got this in my Zigbee2mqtt, when trying to update.. Does that help in finding out, where the problem is?
> 
> <img alt="Image" width="312" src="https://private-user-images.githubusercontent.com/136071042/413038330-1617b824-52cc-4f76-9e27-4b4d1f3137cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzk0NzgzMDcsIm5iZiI6MTczOTQ3ODAwNywicGF0aCI6Ii8xMzYwNzEwNDIvNDEzMDM4MzMwLTE2MTdiODI0LTUyY2MtNGY3Ni05ZTI3LTRiNGQxZjMxMzdjZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIxM1QyMDIwMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02M2JiMTUxYWQ3NzhhNzEwYTllYzRjZjZlZmYwNGYyZWRhNDQxNTUyY2E5ZmIyMTE2ZDlmNTI1NjJmNDRlZDM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.R7PiHY5bBTdEugTiMycbbqUyxmXPVHlMZLBxJfK-pek">
> and
> 
> <img alt="Image" width="352" src="https://private-user-images.githubusercontent.com/136071042/413041493-c8fca257-9ea5-4408-9bed-14514e5c5f34.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzk0NzgzMDcsIm5iZiI6MTczOTQ3ODAwNywicGF0aCI6Ii8xMzYwNzEwNDIvNDEzMDQxNDkzLWM4ZmNhMjU3LTllYTUtNDQwOC05YmVkLTE0NTE0ZTVjNWYzNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIxM1QyMDIwMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YmUzNjY1YmMwYzNlNjVhNjEyZmQ5ZTc5NmFjYmI2ZTIxMTk1MzZjYmY2ODBhNjIyNzIzMGFkOGQ2MTdlNjZmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.odzDesSp2jsON6ka137UcoiG7cU6vp8PB6_J7JR3ZtM">
> and
> 
>

nephrotranz, 2025-02-13:
@JensjJ8200 Did you keep the switch alive by pressing the buttons after it was said that it's getting the firmware update?

tomtom13, 2025-02-13:
> [@JensjJ8200](https://github.com/JensjJ8200) Did you keep the switch alive by pressing the buttons after it was said that it's getting the firmware update?

Actually what I had to do to trigger any upgrade (which always fail BTW) is to get "start the upgrade" and imediatelly hit any button on the dimmer to get it to wake up.

JensjJ8200, 2025-02-14:
> [@JensjJ8200](https://github.com/JensjJ8200) Did you keep the switch alive by pressing the buttons after it was said that it's getting the firmware update?

actually no. But I did have to remove the device from Z2M, restart Z2M and then readopt it. But after that, then it was updated.

JensjJ8200, 2025-02-14:
> > So could the problem be with the ember software? or perhaps its pure luck from my side.
> 
> Pure luck, or maybe less traffic on the test network.

I would not say that there is a lot of traffic on my production network (only 26 devices in total). Perhaps luck, but then again, perhaps it's worth to find out if people are using Ember or Zstack. What if, it mainly or only was on Ember? I mean it's quite easy to ask, and it does not cost anything to ask. I don't know a lot about trouble shooting in this kind of setting, but if it was on support, when I did IT support, then I would defiently look too see if there is a common denominator.

daemongloom, 2025-02-14:
> What if, it mainly or only was on Ember? I mean it's quite easy to ask, and it does not cost anything to ask. 

Zstack user here - and I have the same issue.

JensjJ8200, 2025-02-14:
> > What if, it mainly or only was on Ember? I mean it's quite easy to ask, and it does not cost anything to ask.
> 
> Zstack user here - and I have the same issue.

Thank you for replying. Then I do see, that it's not a valid point, that i could "just" be an ember problem... :-(

tomtom13, 2025-02-14:
> > > What if, it mainly or only was on Ember? I mean it's quite easy to ask, and it does not cost anything to ask.
> > 
> > 
> > Zstack user here - and I have the same issue.
> 
> Thank you for replying. Then I do see, that it's not a valid point, that i could "just" be an ember problem... :-(

#METO: ```adapter: zstack```

odeBuXTeR, 2025-02-15:
zstack for me too and unable to update the dimmer switch. Z2M 2.2.2, SLZB-06P10 coordinator with 20240716 ZB firmware, 2.7.1 core FW.
All working fine for the update, led flashing RED/GREEN once flashed but stay with older FW

nephrotranz, 2025-02-15:
So my last status update now:
I was able to update one of my three v1 switches with Zigbee2MQTT after two or three tries.
I tried the other ones, but gave it up and connected both to my Hue Bridge where the update was done automatically within a few hours.

After a restart of the whole HA system I reconnected them to Zigbee2MQTT and they were correctly recognized with the new firmware version now.

So done for me in the end.

JensjJ8200, 2025-02-15:
> zstack for me too and unable to update the dimmer switch. Z2M 2.1.1, SLZB-06P10 coordinator with 20240716 ZB firmware, 2.7.1 core FW. All working fine for the update, led flashing RED/GREEN once flashed but stay with older FW

Have you seen, that there is an 8.X.X update for det SLZB?

odeBuXTeR, 2025-02-15:
Yes I saw that, but it's a dev udpate, and I want to stay with stable release to prevent hazard

mbrucher, 2025-02-15:
Also have an issue with Hue. Interestingly enough, Ikea OTA worked whereas it didn;t for a long time. Or maybe it's just that the switch is not used often :D:D:D

jammi, 2025-02-16:
I'm confirming I have one of my two hue dimmer switches like this. The update doesn't start until you press a button on the device to wake it up, but even when it completes, it's just showing up as needing an update again. However, both of mine are [324131092621](https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621), both show
>Firmware build date 20190410
>Firmware version 6.1.1.28573
But only one of them is shown to need an update.

[image]

tomtom13, 2025-02-17:
OK, so since there was not requests from @Koenkk to do further testing. I've upgraded all remaining dimmers. I can report few things (some of the fishy):
1. procedure provided by @Koenkk works (force remove -> reboot z2m -> re-join dimer) and all the data works OK.
2. while removed from z2m attaching to HUE hub (with additional bulb for good measure) upgrade process albat slow was successful and I upgraded most of dimmers that way
3. few of dimmers I attempted to rejoin with "the procedure" and they have shown up again as needing the upgrade in Z2M -> I've upgraded those, but after the upgrade one shown up as having old firmware but was working without a glitch. Other have reported back as having new firmware but caused very trange behaviour in HASS, where actions and triggers were reported in logbook, but those were NOT doing anything. Fun-part is that everything shown up in HASS, but NODE-RED was not showing up the dimmer, actions and triggers at all ?! Rebooting, the hass did not resolve it. I had to perform the rejoin "procedure" to get the dimmer working properly. 
4. Battery reporting is somehow wonky with new firmware, reported battery state is quite high, more than 70% but dimmers stoped working and shown RED light while pressing a button.

dergreg, 2025-02-25:
Has there been any further progress except doing all the workarounds mentioned earlier of which some seem to work and some not?

I'm assuming that since OTA works via original Hue bridge, the issue must be somewhere within z2m?

tomtom13, 2025-02-26:
Yeah, but it seems that there are few more things wrong now with Z2M - v1 dimmers are reporting as having 100% batter, yet when you press the button they light up red, indicating that the bettery might be low.

trashkpi, 2025-03-19:
My v1 switches update either fails or updates to the same version

> info 2025-03-19 17:52:26z2m: Device 'sw_kidsroom' was updated from '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}' to '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}'

sfaria, 2025-03-25:
I just saw this was marked as completed - was there a fix or something in this thread I missed?

tomtom13, 2025-03-25:
I was about to ask the same thing - issue is NOT fixed for me, I had to use philips HUE HUB to do the upgrades, had to jump through hoops of reconnecting the dimmer switched to Z2M ... yeah it unblocked me, but it's far from having a fix in the code.

[35 more replies]