Koenkk/zigbee2mqtt #26238: [New device support]: _TZE284_khah2lkr Floor Heating Thermostat

yergo, 2025-02-07:
### Link

https://smartbudynek.pl/pl/products/termostat-zigbee-sterownik-ogrzewania-tuya-te-1z-przewodowy-do-podlogowki-598.html

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0x583bc2fffe1cab69","nwkAddr":15610,"manufId":4098,"manufName":"_TZE284_khah2lkr","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��6/i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_khah2lkr","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1738934548365}

### Zigbee2MQTT version

2.1.1-1

### Comments

Hello. Added new device, it paired well but is not supported. Im totally new to this things but was able to probe some datapoints and crafted an external definition for some of you hopefully.

Despite they did sent it to me with wrong manual (a manual to wifi device, I own zigbee non-wifi device) but also can share its for setttable data reference.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;

const tuyaDatapoints = [
    [16, 'internal_sensor_temperature', tuya.valueConverter.divideBy10],
    [50, 'target_room_temperature', tuya.valueConverter.divideBy10],

    // data type 1
    [102, 'running_state', tuya.valueConverterBasic.lookup({idle: false, heating: true})],
    [103, 'external_sensor_temperature', tuya.valueConverter.divideBy10],

    // can be changed by setting 112 below 20, data type 1
    [106, 'high_temperature_protection_state', tuya.valueConverterBasic.lookup({off: false, protecting: true})],
    

    // can be changed by setting 113 over 10, data type 1
    [107, 'low_temperature_protection_state', tuya.valueConverterBasic.lookup({off: false, protecting: true})],

    // range -9 to +9, data type 2, affects shown room temperature (even tho sensors detect its 19, you can make it show 21 by setting this to 2)
    [109, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],

    // according to manual settable betwwen 0.5 and 2.5 degree.
    // staring with 5 as 0.5 degree, and 25 as 2.5 degree (data type 2) 
    [110, 'temperature_return_difference', tuya.valueConverter.raw],

    // range 1-9. How far should temperature drop to turn back heating, if hight temp protection kicked in
    [111, 'external_sensor_return_difference', tuya.valueConverter.raw],

    // High temperature protection
    // range 20-70, trying to turn below 20 keeps this datapoint at 20 but turns 106 to 0
    [112, 'high_temperature_protection_setting', tuya.valueConverter.divideBy10],

    // range 1-10, trying to turn over 10 keeps this datapoint at 10, but turns 107 to 0
    [113, 'low_temperature_protection_setting', tuya.valueConverter.divideBy10],
    [114, 'max_settable_temp', tuya.valueConverter.divideBy10],

    // choose_sensor
    // 0: device sensor. Switches "hight_temperature_protection_state" off
    // 1: external sensor / hight temperature protection off
    // 2: internal for room + external for high temperature protection. Switches "hight_temperature_protection_state" on
    [116, 'choose_sensor', tuya.valueConverterBasic.lookup({IN: 0, OU: 1, AL: 2})],

    [119, 'unknown_119', tuya.valueConverter.raw],
    [120, 'unknown_120', tuya.valueConverter.raw],
    [121, 'unknown_121', tuya.valueConverter.raw],
    [122, 'unknown_122', tuya.valueConverter.raw],

    // device state
    // 0: standby mode - displays temperature but will not turn heating. Manual configuration is accessible only in this state
    // 1: fully functional, can turn on heating
    [125, 'device_state', tuya.valueConverterBasic.lookup({OFF: 0, ON: 1})],

    // manual_mode
    // 1: automatically turns heating when protection levels hit
    // 2: you can turn off heating if it heats. You can set up target temperature
    // 3: you can set up target temperature and it will automatically try to maintain it
    [128, 'manual_mode', tuya.valueConverterBasic.lookup({AUTO: 1, MANUAL: 0, MIXED: 3})],

    [129, 'device_locked', tuya.valueConverterBasic.lookup({LOCKED: 1, UNLOCKED: 0})],

    // data type 5. Was [0] but went [16] when high temperature protection went into alert mode
    [130, 'high_temperature_alarm', tuya.valueConverter.raw],
];

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'Tuya Zigbee Thermostat',
    vendor: '_TZE284_khah2lkr',
    description: 'Tuya Floor Heating Thermostat',
    fromZigbee: [
        tuya.fz.datapoints,
    ],
    toZigbee: [
        tuya.tz.datapoints, 
    ],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [],
    meta: {
        tuyaDatapoints: tuyaDatapoints,
    }
};
module.exports = [definition];
```

### What does/doesn't work with the external definition?

So far definition seems working in terms, that some data are starting to appear in "state" tab:

```
{
    "external_sensor_temperature": 20,
    "internal_sensor_temperature": 20,
    "linkquality": 220,
    "manual_mode": "MANUAL",
    "running_state": false,
    "target_temperature": 20
}
```

Im trying to adjust it towards different defices as its in _TZE204_edl8pz1k but I'm afraid I lack too much knowlege

[image]
[image]
[image]

mistastn, 2025-02-08:
Can you post the datapoints from Tuya? I am curious about couple things. I have a device with the same ID but it looks slightly different.

It is this one https://device.report/manual/14903751 cannot find it on any international store.

I am having issues with the way you named the datapoints it does not match the ones z2m expects and makes it hard to create an actual climate entity.

mistastn, 2025-02-08:
Not the best but this lets me control my thermostat in HA

```ts
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {climate, child_lock, away_mode} = require('zigbee-herdsman-converters/lib/modernExtend');

const e = exposes.presets;
const ea = exposes.access;

const tuyaDatapoints = [
    [16, 'local_temperature', tuya.valueConverter.divideBy10],
    [50, 'current_heating_setpoint', tuya.valueConverter.divideBy10],

    // data type 1
    [102, 'running_state', tuya.valueConverterBasic.lookup({heat: tuya.enum(1), idle: tuya.enum(0)})],
    [103, 'external_sensor_temperature', tuya.valueConverter.divideBy10],

    // can be changed by setting 112 below 20, data type 1
    [106, 'high_temperature_protection_state', tuya.valueConverterBasic.lookup({off: false, protecting: true})],
    

    // can be changed by setting 113 over 10, data type 1
    [107, 'low_temperature_protection_state', tuya.valueConverterBasic.lookup({off: false, protecting: true})],

    // range -9 to +9, data type 2, affects shown room temperature (even tho sensors detect its 19, you can make it show 21 by setting this to 2)
    [109, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],

    // according to manual settable betwwen 0.5 and 2.5 degree.
    // staring with 5 as 0.5 degree, and 25 as 2.5 degree (data type 2) 
    [110, 'temperature_return_difference', tuya.valueConverter.raw],

    // range 1-9. How far should temperature drop to turn back heating, if hight temp protection kicked in
    [111, 'external_sensor_return_difference', tuya.valueConverter.raw],

    // High temperature protection
    // range 20-70, trying to turn below 20 keeps this datapoint at 20 but turns 106 to 0
    [112, 'max_temperature', tuya.valueConverter.divideBy10],

    // range 1-10, trying to turn over 10 keeps this datapoint at 10, but turns 107 to 0
    [113, 'low_temperature_protection_setting', tuya.valueConverter.divideBy10],
    [114, 'max_settable_temp', tuya.valueConverter.divideBy10],

    // choose_sensor
    // 0: device sensor. Switches "hight_temperature_protection_state" off
    // 1: external sensor / hight temperature protection off
    // 2: internal for room + external for high temperature protection. Switches "hight_temperature_protection_state" on
    [116, 'choose_sensor', tuya.valueConverterBasic.lookup({IN: 0, OU: 1, AL: 2})],

    [119, 'unknown_119', tuya.valueConverter.raw],
    [120, 'unknown_120', tuya.valueConverter.raw],
    [121, 'unknown_121', tuya.valueConverter.raw],
    [122, 'unknown_122', tuya.valueConverter.raw],

    // device state
    // 0: standby mode - displays temperature but will not turn heating. Manual configuration is accessible only in this state
    // 1: fully functional, can turn on heating
    [125, 'away_mode', tuya.valueConverterBasic.lookup({OFF: false, ON: true})],

    // manual_mode
    // 1: automatically turns heating when protection levels hit
    // 2: you can turn off heating if it heats. You can set up target temperature
    // 3: you can set up target temperature and it will automatically try to maintain it
    [128, 'preset', tuya.valueConverterBasic.lookup({AUTO: 1, MANUAL: 0, MIXED: 3})],

    [129, 'child_lock', tuya.valueConverterBasic.lookup({LOCK: true, UNLOCK: false})],

    // data type 5. Was [0] but went [16] when high temperature protection went into alert mode
    [130, 'high_temperature_alarm', tuya.valueConverter.raw],
];

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'Tuya Zigbee Thermostat',
    vendor: '_TZE284_khah2lkr',
    description: 'Tuya Thermostat',
    fromZigbee: [
        tuya.fz.datapoints,
    ],
    toZigbee: [
        tuya.tz.datapoints, 
    ],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [            
            e.child_lock(),
            e.away_mode(),
            e.climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withPreset(['auto', 'manual', 'mixed'])
                .withLocalTemperatureCalibration(-3, 3, 1, ea.STATE_SET)
        ],
    meta: {
        tuyaDatapoints: tuyaDatapoints,
    }
};
module.exports = [definition];
```

yergo, 2025-02-09:
> Can you post the datapoints from Tuya? I am curious about couple things. I have a device with the same ID but it looks slightly different.
>
> It is this one https://device.report/manual/14903751 [...]

This one looks quite familiar, dont know anything about vacation mode tho. I have developed some definitions so far, will try to post it soon. Indeed I needed to change some atapoints namings - sorry for inconvinience, I am totally new to HA and all smart things. Nevertheless some programming experience and lots of already done definitions, togerher with chat gpt helped a bit. Give me some time to post the code for you to review, so we can process it for others

yergo, 2025-02-09:
Please look into pull request to verify it everything is according to the starndards. What is not done, It is supposed to have cooling mode and I don't have how to test it. I would love to turn "presets" readonly to prevent trying to change them from interface as this is one way ticket. Also I want to fully test it once I will be in location again.

Please note, that I have limited possibilities to provide information. Datapoints are collected from the logs and everything is with a decent help of chat gpt and my experience on other platforms. Therefore I count on a careful check if everything is right as I totally not trust myself here.

closed 2025-02-22 by yergo

yergo, 2025-03-05:
@aknarts thanks! You asked for presets: in my device I can change them remotely from auto/semi-auto to manual, but afterwards cant change them back - it seems in interface it is changed, but on device screen it is still manul nevertheless.. Yet device seems to reset to auto after a while if heating is off