Koenkk/zigbee2mqtt #26241: XIAOMI GZCGQ01LM illuminance values issue since Z2M 2.1.0

ALaDoffe, 2025-02-07:
### What happened?

Hello 

Since Z2M 2.1.0, my illuminance sensor reports bad values during night period.
Indeed, it alternates between 1 lx and 0 lx as you can see in the graph below, that is a problem for my light automations.

[image]

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

Sonoff P

### Setup

add-on on HA OS

ALaDoffe, 2025-02-08:
The battery is less than a month old. 
The issue appears to be just after the Z2M update to 2.1.0

mrmaximas, 2025-02-10:
imho it's no z2m issue. i have two GZCGQ01LM with no this issue. i'm using latest dev every day.

tjkgdd, 2025-02-12:
I have exactly the same issue with all my (3) xiaomi lux devices. I was wondering why my automation stops working for couple od days. All of my devices in total darkness shows 1lx instead 0lx.
For me it start after update to 2.1.1 z2m 2 days ago.

[image]

mrmaximas, 2025-02-12:
It seems not difficult to replace the rule = 0 with <2, but the behaviour is really strange, my two such sensors showing 0 at night with the latest dev.

closed 2025-04-22 by github-actions[bot]