Koenkk/zigbee2mqtt #26242: After upgrading to version 2.1.1-1, 2.1.0-1, 2.0.0-2. the module does not start. HomeAssistent

vde69, 2025-02-07:
### What happened?

After upgrading to version 2.1.1-1, 2.1.0-1, 2.0.0-2. the module does not start.
Version 1.40.2-1 is stable. 
Log
---------------------
[18:31:51] INFO: Starting socat process ...
2025/02/07 18:31:51 socat[46] N PTY is /dev/pts/0
2025/02/07 18:31:51 socat[46] N listening on AF=2 0.0.0.0:8888
[18:31:51] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-07 18:31:55] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-07 18:31:55] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-07 18:31:55] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-07 18:31:55] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-07 18:31:55] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-07 18:31:55] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-07 18:31:55] error: 	z2m: Exiting...
[2025-02-07 18:31:55] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
--------------------------------------------------------------

Config:
--------------------------------------------------------------
data_path: /config/zigbee2mqtt
socat:
  enabled: true
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8888,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: homeassistant
  password: ********
serial:
  port: tcp://192.168.1.53:8888
external_converters:
  - ext_converter_ts0011.js
  - ext_converter_ts0012.js
  - ext_converter_ts0013.js

### What did you expect to happen?

Normal start

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

external TC/IP adapter

### Adapter

ATHOM Zigbee 3,0 Zigbee2MQTT CC2652P ESP32

### Setup

Core 2025.1.2 Supervisor 2025.02.0 Operating System 14.1

chris-1243, 2025-02-07:
> [2025-02-07 18:31:55] error: z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.

From your log.

You need to specify `port` and `adapter` in your configuration.

It should look like:
```
serial:
  port: tcp://192.168.1.53:8888
  adapter: zstack
```

`zstack` as you have a TI chip...

vde69, 2025-02-08:
Thank you, indeed the "adapter: zstack" parameter helped to solve the problem. But in previous versions it wasn't there and everything worked :) 
You can probably close the ticket.

chris-1243, 2025-02-08:
I will let you close your ticket.

This is due to numerous changes in v2.x.x.
#24198 
#24364

closed 2025-04-17 by github-actions[bot]