Koenkk/zigbee2mqtt #26251: [New device support]:

Miedzcu, 2025-02-08:
### Link

https://botland.com.pl/tuya-automatyka-domowa/24810-tuya-czujnik-temperatury-i-wilgotnosci-zigbee-wyswietlacz-lcd-sonda-zewnetrzna-rtx-zths3-5903794123298.html

### Database entry

{"id":34,"type":"EndDevice","ieeeAddr":"0xa4c138c2feed2c2f","nwkAddr":32339,"manufId":4417,"manufName":"_TZE284_qyflbnbj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"*�)/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":77}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1738997979104}

### Zigbee2MQTT version

1.22.1

### Comments

Hello, I'm newbie to this, tried to follow add new device support page, but got stuck. Ineptly, I tried to adapt configurations from already supported devices:
https://www.zigbee2mqtt.io/devices/TS0601_temperature_humidity_sensor_1.html
https://www.zigbee2mqtt.io/devices/TS0601_temperature_humidity_sensor_2.html
https://www.zigbee2mqtt.io/devices/TS0601_temperature_humidity_sensor_3.html

I have noticed that support has already been added in ZHA Device Handlers:
https://github.com/zigpy/zha-device-handlers/issues/3647

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_qyflbnbj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

works:

linkquality

does not work

temperature,
humidity,

danielcherubini, 2025-02-08:
I just got this device a hour ago.. 

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_qyflbnbj',
        },
    ],
    model: 'TS0601',
    vendor: '_TZE284_qyflbnbj',
    description: 'Temperature, Humidity and Battery Sensor (using datapoints)',
    fromZigbee: [legacy.fromZigbee.tuya_temperature_humidity_sensor],
    toZigbee: [],
    exposes: (device, options) => {
            const exps = [e.temperature(), e.humidity(), e.battery()];
            if (!device || device.manufacturerName === '_TZE200_qyflbnbj' || device.manufacturerName === '_TZE204_qyflbnbj') {
                exps.push(e.battery_low());
                exps.push(e.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'));
            }
            return exps;
    },
};

module.exports = definition;
```

This worked for me.

The change needs to happen here

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L979

I'll open a PR tonight

Koenkk, 2025-02-09:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8777

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-09 by Koenkk