Koenkk/zigbee2mqtt #26254: Ubisys shutter control J1 stopped working beginning with 2.1.0

svenso, 2025-02-08:
### What happened?

Starting with 2.1.0 (also 2.1.1), the ubisys J1 do not work reliable anymore. 
I have 10 of them. After the first upgrade to 2.1.0 none of them worked. Today I tried with 2.1.1, only one was working after the update.
All other functions with the J1 seems to work (interview can be completed, position get reported back correctly, ..). Just the operation (Up / Down / Stop) does not work anymore.

### What did you expect to happen?

The shutter should move :)

Instead nothing happened and after few seconds, a timeout appears in the logs:

error 2025-02-08 09:42:05z2m: Publish 'set' 'state' to 'markise_gross' failed: 'Error: ZCL command 0x001fee000000653c/0 closuresWindowCovering.downClose({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (56) Error: waiting for response TIMEOUT)'
error 2025-02-08 09:42:07z2m: Publish 'set' 'state' to 'markise_tur' failed: 'Error: ZCL command 0x001fee0000006555/0 closuresWindowCovering.downClose({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (57) Error: waiting for response TIMEOUT)'

The one that was working responded successfully:
info 2025-02-08 09:42:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buro_j1', payload '{"energy":0.01,"last_seen":"2025-02-08T08:42:57.775Z","linkquality":255,"position":0,"power":0,"state":"CLOSE","tilt":81,"update":{"installed_version":38798407,"latest_version":38798407,"state":"idle"},"update_available":false}'
info 2025-02-08 09:42:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buro_j1', payload '{"energy":0.01,"last_seen":"2025-02-08T08:42:57.777Z","linkquality":255,"position":0,"power":0,"state":"CLOSE","tilt":81,"update":{"installed_version":38798407,"latest_version":38798407,"state":"idle"},"update_available":false}'
info 2025-02-08 09:42:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buro_j1', payload '{"energy":0.01,"last_seen":"2025-02-08T08:42:57.777Z","linkquality":255,"position":0,"power":0,"state":"CLOSE","tilt":71,"update":{"installed_version":38798407,"latest_version":38798407,"state":"idle"},"update_available":false}'
info 2025-02-08 09:42:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/buro_j1', payload '{"energy":0.01,"last_seen":"2025-02-08T08:42:59.412Z","linkquality":255,"position":0,"power":0,"state":"CLOSE","tilt":72,"update":{"installed_version":38798407,"latest_version":38798407,"state":"idle"},"update_available":false}'

### How to reproduce it (minimal and precise)

Update, try any operation on a Ubisys J1

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Add-On on Homeassistant, Intel NUC Platform, as VM in Proxmox

### Debug log

[log.log](https://github.com/user-attachments/files/18717670/log.log)

sjorge, 2025-02-16:
Have you tried powering off the J1 and repowring them after a few seconds, or completely repairing them?

The timeout seems to indicate some sort of issue where the device never gets the message.

svenso, 2025-02-16:
Thanks for your answer.
I did not re-pair them or tried any other deeper repair actions, as they started working again very reliable after the downgrade of z2m.
It has to be smt in the update..

svenso, 2025-03-06:
Rejoining 8 of 10 devices solved the issue.

closed 2025-03-06 by svenso