Koenkk/zigbee2mqtt #26255: No access to USB dongle after update Zigbee2MQTT version 2.1.1

3raider, 2025-02-08:
### What happened?

Hello!After updating to latest version faced with problem (log file below)

[2025-02-08 12:16:44] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-08 12:16:44] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-08 12:16:44] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-08 12:16:44] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"a8337ad52322ec11acf61dc32c86906c","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_a8337ad52322ec11acf61dc32c86906c-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=4
[2025-02-08 12:16:44] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-08 12:16:44] info: 	zh:zstack:znp: Serialport opened
[2025-02-08 12:16:44] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-08 12:16:46] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-08 12:17:04] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-08 12:17:04] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-08 12:17:04] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-08 12:17:04] error: 	z2m: Exiting...
[2025-02-08 12:17:04] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

NA

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Addon on Home Assistant

bubblegum6754, 2025-02-09:
+2

No longer works with Sonoff Zigbee 3.0 USB Dongle Plus

MrFrenzyyy, 2025-02-09:
+3

was running fine but after update by ZBdongle-E ondly works for 10 min after a reflash of the Ember firmware. I have now reverted back to EZSP and it seems to work

Sergocoss, 2025-02-09:
+4

r375ch, 2025-02-09:
+5

WerVa, 2025-02-09:
+6

Koenkk, 2025-02-13:
Could someone provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

bubblegum6754, 2025-02-13:
Here is the log with debug turned on:

[2025-02-13 13:10:24] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-02-07.22-07-39'
[2025-02-13 13:10:24] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-13 13:10:24] debug: 	z2m: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
[2025-02-13 13:10:24] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-13 13:10:24] debug: 	z2m: sd-notify loaded
[2025-02-13 13:10:24] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-13 13:10:24] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":17278,\"extendedPanID\":[167,244,244,10,50,52,233,162],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{},\"adapter\":{\"disableLED\":false}}"'
[2025-02-13 13:10:24] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"6af08de6521fef11af1057d0639e525b","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6af08de6521fef11af1057d0639e525b-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-02-13 13:10:24] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"6af08de6521fef11af1057d0639e525b","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6af08de6521fef11af1057d0639e525b-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=4
[2025-02-13 13:10:24] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":17278,"extendedPanID":[167,244,244,10,50,52,233,162],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-02-13 13:10:24] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-13 13:10:24] info: 	zh:zstack:znp: Serialport opened
[2025-02-13 13:10:24] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-13 13:10:24] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-13 13:10:24] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-13 13:10:24] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2025-02-13 13:10:25] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-13 13:10:25] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-13 13:10:26] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-13 13:10:26] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-13 13:10:26] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-13 13:10:29] debug: 	zh:zstack:unpi:parser: <-- [1,66,161,168,89,40,85,198,166,200,126]
[2025-02-13 13:10:29] debug: 	zh:zstack:unpi:parser: --- parseNext [1,66,161,168,89,40,85,198,166,200,126]
[2025-02-13 13:10:31] debug: 	zh:zstack:unpi:parser: <-- [18,66,161,169,84,42,125,56,176,25,224,1,125,51,126]
[2025-02-13 13:10:31] debug: 	zh:zstack:unpi:parser: --- parseNext [1,66,161,168,89,40,85,198,166,200,126,18,66,161,169,84,42,125,56,176,25,224,1,125,51,126]
[2025-02-13 13:10:32] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-13 13:10:32] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-13 13:10:38] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-13 13:10:38] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-13 13:10:44] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-13 13:10:44] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-13 13:10:44] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-13 13:10:44] error: 	z2m: Exiting...
[2025-02-13 13:10:44] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

Koenkk, 2025-02-14:
It seems it cannot connect to the adapter, does downgrading to 2.1.0 fix this?

bubblegum6754, 2025-02-14:
I'm on home assistant and unfortunately there is no option (that I know off) to roll back.

closed 2025-04-23 by github-actions[bot]