Koenkk/zigbee2mqtt #26265: Climate device set local temperature calibration error: payload error Error: extra keys not allowed @ data['payload_template']

willhaggan, 2025-02-08:
### What happened?

publishing message from Home assistant to zigbee2mqtt/{{ target_device }}/set/local_temperature_calibration returns extra keys not allowed error, i beleive the error only appeared after the recent update to 2.1.1

### What did you expect to happen?

the local temperature calibration value to update

### How to reproduce it (minimal and precise)

Tested with 6x Sonoff TRV2B TRV and 1x Moes BRT100 devices same error on both models.

Executed: 8 February 2025 at 23:38:14
Error: extra keys not allowed @ data['payload_template']
Result:

params:
  domain: mqtt
  service: publish
  service_data:
    topic: zigbee2mqtt/0x0ceff6fffedd94d4/set/local_temperature_calibration
    payload_template: -5.4
  target: {}
running_script: false

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Setup

Add on Home Assistant OS

JJanssen123, 2025-02-10:
I just had the same message after updating HA core to 2025.2 (not using Z2M here). 

Maybe this can help you?

In the automation I changed:
   payload_template: "{\"chsetpoint\": {{states('sensor.smart_thermostart_opentherm_setpoint')}}}"
to:
  payload: '{"chsetpoint": {{states(''sensor.smart_thermostart_opentherm_setpoint'')}}}'

Now it works again. Apparently the escaping of " is no longer allowed? 

(Having more troubles with the mqtt-integration lately... stops working regularly since 2025.1 but not able to pinpoint it yet)

willhaggan, 2025-02-10:
> I just had the same message after updating HA core to 2025.2 (not using Z2M here).
> 
> Maybe this can help you?
> 
> In the automation I changed: payload_template: "{\"chsetpoint\": {{states('sensor.smart_thermostart_opentherm_setpoint')}}}" to: payload: '{"chsetpoint": {{states(''sensor.smart_thermostart_opentherm_setpoint'')}}}'
> 
> Now it works again. Apparently the payload_template option is no longer valid?
> 
> (Having more troubles with the mqtt-integration lately... stops working regularly since 2025.1 but not able to pinpoint it yet)

Hi thanks for the update, this give my the idea to just add the same action again within the automation which beleive it or not worked as you will see the action service data has changed in the way it is formatted so not sure what has changed with regards the mqtt publish action and when this happened.

This works....

action: mqtt.publish
metadata: {}
data:
  qos: 1
  retain: false
  payload: >-
    {{(states(external_temperature)|float(0) -
    state_attr(climate_device,'current_temperature')|float(0) +
    states(adjust)|float(0))|round(1) }}
  topic: zigbee2mqtt/{{ target_device }}/set/local_temperature_calibration

Orginal does not work...

data_template:
  topic: zigbee2mqtt/{{ target_device }}/set/local_temperature_calibration
  payload_template: >-
    {{states(external_temperature)|float(0) | round(1) -
    state_attr(climate_device,'current_temperature')|float(0)| round(1) +
    states(adjust)|float(0)| round(1) }}
action: mqtt.publish

closed 2025-04-19 by github-actions[bot]