Koenkk/zigbee2mqtt #26268: Zemismart ZM90-Z-300 (Tuya TS0601 _TZE284_fzo2pocs) window opener CLOSE and OPEN buttons are inversed

ZeusTerran, 2025-02-09:
### What happened?

When you select a position number, the unit goes to the correct position (e.g. 0 is closed, 100 is fully open), but when you select OPEN or CLOSE from the State options, they do the opposite of what you would expect (OPEN makes the unit go to 0, and CLOSE makes the unit go to 100). Interestingly when the unit finishes moving, it cleans up it's state (e.g. when you PRESS CLOSE when the unit is 10% open, it goes to 100% open and as it's moving it's state says "CLOSE", but when it gets to 100% open the state changes to "OPEN").

Note that I have the unit set to Reverse direction: back. That makes position 0 have the shortest chain length. Changing this option to "forward" makes the CLOSE / OPEN button presses work as expected, but the state is still wrong afterwards. E.g. pressing CLOSE in forward mode makes the chain go all the way in, but once it gets there it reports as position 100 and state: OPEN. Nether forward or back lead to coherent position-state outcomes.

The big problem is that it means none of open/close buttons in dashboards work because when the unit is closed the open button is disabled because it thinks it's already fully open and vice versa.

### What did you expect to happen?

Pressing OPEN would open the unit to 100 (unless STOP is selected while it's going), and pressing CLOSE would move the unit to position 0. When the unit is at position 0, it should remain reporting position CLOSE, and when not at 0 it should have a state of OPEN. 

Even better, it would be great to have states of OPENING/CLOSING/OPEN/CLOSED, but that's just a nice to have. 

### How to reproduce it (minimal and precise)

Simply pair the unit and press OPEN or CLOSE on state, it will do the opposite.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.1.0 build 0

### Adapter

EZSP

### Setup

Add-on to HA in windows Hyper-V, Intel based machine

ZeusTerran, 2025-02-09:
After a bit more tinkering I think it might just be that this line:

`[1, 'state', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2)})],`

should actually be:

`[1, 'state', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(2), STOP: tuya.enum(1), CLOSE: tuya.enum(0)})],`

Hopefully someone smarter than me can confirm this, because the default code groups together a number of devices under "TS0601_cover_8" and I'm not sure that they all suffer from this problem?

lattice0, 2025-03-20:
Same here: https://github.com/Koenkk/zigbee2mqtt/issues/26800

Sorry for the duplicated.

closed 2025-05-28 by github-actions[bot]