Koenkk/zigbee2mqtt #26269: IKEA E1766 cannot pair

digidolphin, 2025-02-09:
### What happened?

I have tried reading through all the info on breaking changes, and the configuration changes needed, but despite trying all sorts of combination, I just cannot get to pair the IKEA E1766 to Z2M anymore. I was able to before Z2M v2.

I have these configuration entries:
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

I have   legacy_action_sensor: true, but also tested with false on this one. Have also teste with api true and payload true. Regardless of setting, I get Error: Interview failed because can not get active endpoints
I have other E1766 devices connected to the network, that work fine, but fear that one day they'll lose connection, and I cannot reconnect them.

### What did you expect to happen?

I expected it to complete pairing once starting the interview

### How to reproduce it (minimal and precise)

Enable pairing. Four presses on the ikea button. Wait.

### Zigbee2MQTT version

2.1.1.1

### Adapter firmware version

7.4.1.0 build 0

### Adapter

Sonoff zigbee 3.0

### Setup

Add-on on HAOS running on RPI5

### Debug log

warning 2025-02-09 07:55:16z2m: Device '0x0c4314fffe109492' left the network
info 2025-02-09 07:55:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe109492","ieee_address":"0x0c4314fffe109492"},"type":"device_leave"}'
info 2025-02-09 07:55:29zh:controller: Interview for '0x0c4314fffe109492' started
info 2025-02-09 07:55:29z2m: Device '0x0c4314fffe109492' joined
info 2025-02-09 07:55:29z2m: Starting interview of '0x0c4314fffe109492'
info 2025-02-09 07:55:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe109492","ieee_address":"0x0c4314fffe109492"},"type":"device_joined"}'
info 2025-02-09 07:55:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe109492","ieee_address":"0x0c4314fffe109492","status":"started"},"type":"device_interview"}'
error 2025-02-09 07:55:50zh:controller: Interview failed for '0x0c4314fffe109492 with error 'Error: Interview failed because can not get active endpoints ('0x0c4314fffe109492')'
error 2025-02-09 07:55:50z2m: Failed to interview '0x0c4314fffe109492', device has not successfully been paired
info 2025-02-09 07:55:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe109492","ieee_address":"0x0c4314fffe109492","status":"failed"},"type":"device_interview"}'

Hessenpower01, 2025-02-11:
The problem is also handled here: https://github.com/Koenkk/zigbee2mqtt/issues/25614

paulz-gh, 2025-02-17:
I'm also having problems with E1766 remotes. 
Cannot get them to pair for most attempts.
Got it paired once and it never showed the battery status.
Not using home assistant here, just plain z2m 2.1.1.

Removing and reinserting the battery got it paired. 
But even though I see open and close actions in the logs, the status remains offline.

closed 2025-04-27 by github-actions[bot]