Koenkk/zigbee2mqtt #26271: BUFFER_FULL with sonoff dongle-P

HenrikHills, 2025-02-09:
### What happened?

Suddenly network stops responding and logs get filled with BUFFER_FULL error messages, see attached debug-log

[9336c2b0_zigbee2mqtt_2025-02-09T07-45-35.752Z.log](https://github.com/user-attachments/files/18723677/9336c2b0_zigbee2mqtt_2025-02-09T07-45-35.752Z.log)

### What did you expect to happen?

to not get buffer-full messages

### How to reproduce it (minimal and precise)

Starts sporadically happening, have to re-seat usb-dongle and restart Z2M addon 

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0 USB Dongle plus

### Setup

HAOS on mini-pc and z2m as HA addon

### Debug log

[9336c2b0_zigbee2mqtt_2025-02-09T07-45-35.752Z.log](https://github.com/user-attachments/files/18723679/9336c2b0_zigbee2mqtt_2025-02-09T07-45-35.752Z.log)

HenrikHills, 2025-02-09:
Might be related, but I also get these a few times at start-up of Z2M, seems to be crashing 1-2 times before it manages to start

`2025-02-07 14:30:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/bridge/state', payload '{"state":"offline"}'
[2025-02-07 14:30:22] info: 	z2m: Disconnecting from MQTT server
[2025-02-07 14:30:22] info: 	z2m: Stopping zigbee-herdsman...
[2025-02-07 14:30:30] error: 	z2m: Failed to read state of 'Kontoret Spotter' after reconnect (ZCL command 0x7cc6b6fffebc42ea/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-07 14:30:35] error: 	z2m: Not connected to MQTT server!
[2025-02-07 14:30:35] error: 	z2m: Cannot send message: topic: 'zigbee2mqtt2/Vibrasjonssensor', payload: '{"action":null,"angle":6,"angle_x":-1,"angle_x_absolute":91,"angle_y":-2,"angle_y_absolute":92,"angle_z":-88,"battery":100,"device_temperature":32,"linkquality":87,"power_outage_count":222,"sensitivity":null,"strength":148,"vibration":false,"voltage":3015,"x_axis":-15,"y_axis":-20,"z_axis":-745}
[2025-02-07 14:30:36] error: 	z2m: Failed to read state of 'Stellerom Taklampe' after reconnect (ZCL command 0x881a14fffefe5f12/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
[2025-02-07 14:30:42] error: 	z2m: Publish 'set' 'state' to 'Frient Plug 1' failed: 'Error: ZCL command 0x0015bc002f00cee8/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
[2025-02-07 14:30:48] error: 	z2m: EventBus error 'OTAUpdate/deviceMessage': CommandResponse 0xf4b3b1fffef951a2/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)`

HenrikHills, 2025-02-10:
Error starter to appear / network stopped responding at around line 59162

[log1.log](https://github.com/user-attachments/files/18730146/log1.log)

Koenkk, 2025-02-10:
Could you try with the following fw instead? https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226

HenrikHills, 2025-02-10:
Will reflash it with that fw, i have moved some devices back to my conbee ii which handled the network with no issue (running two Z2M instances now while im troubleshooting to see if i can lighten the load of the network)

Will come back here in a day or two (or when something happens) with update

Shadoxity, 2025-02-11:
I am having the same issues which start almost daily.
I have reflashed to this old firmware just now, so ill see how it goes over the next couple days

HenrikHills, 2025-02-11:
> Could you try with the following fw instead? https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226

At the moment it seems like this fixed it, but still a little early to say, I will migrate the devices back from the Conbee-stick tomorrow and see if that changes anything

Shadoxity, 2025-02-12:
Im still getting the same errors after changing to 20221226. It seems to be almost every 12 hours that I notice it coming up wit h the error.  (At least it was at just after 11pm and then 12pm today
However I just realised I have not restarted home assistant after i flashed it, so am going to do that now and see if its still happening.

 'Error: Command 1 genOnOff.off({}) failed (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":179,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,165,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-02-12 12:38:11z2m: Publish 'set' 'state' to 'hallway' failed: 'Error: Command 1 genOnOff.off({}) failed (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":180,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,166,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-02-12 12:38:11z2m: Publish 'set' 'state' to 'hallway' failed: 'Error: Command 1 genOnOff.off({}) failed (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":181,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,167,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-02-12 12:38:11z2m: Publish 'set' 'state' to 'hallway' failed: 'Error: Command 1 genOnOff.off({}) failed (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":182,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,168,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-02-12 12:38:11z2m: Publish 'set' 'state' to 'hallway' failed: 'Error: Command 1 genOnOff.off({}) failed (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":183,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,169,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
error 2025-02-12 12:38:11z2m: Publish 'set' 'state' to 'hallway' failed: 'Error: Command 1 genOnOff.off({}) failed (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":184,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,170,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
warning 2025-02-12 12:38:11z2m: Failed to ping 'Washer\Dryer' (attempt 1/2, ZCL command 0xa4c1389bca702a6b/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
warning 2025-02-12 12:38:11z2m: Failed to ping 'Washer\Dryer' (attempt 2/2, ZCL command 0xa4c1389bca702a6b/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 49622 - 1 - 180 - 0 - 1 after 10000ms))
error 2025-02-12 12:38:11z2m: Failed to read state of 'Washer\Dryer' after reconnect (ZCL command 0xa4c1389bca702a6b/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 49622 - 1 - 206 - 6 - 1 after 10000ms))
warning 2025-02-12 12:38:11z2m: Failed to ping 'hall1' (attempt 1/2, ZCL command 0x001788010e58ea66/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 63718 - 11 - 224 - 0 - 1 after 10000ms))
warning 2025-02-12 12:38:11z2m: Failed to ping 'kitchen7' (attempt 1/2, ZCL command 0x001788010e58ee6a/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
warning 2025-02-12 12:38:11z2m: Failed to ping 'hall4' (attempt 1/2, ZCL command 0x001788010e5b7275/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 31295 - 11 - 38 - 0 - 1 after 10000ms))

HenrikHills, 2025-02-14:
> At the moment it seems like this fixed it, but still a little early to say, I will migrate the devices back from the Conbee-stick tomorrow and see if that changes anything

I have been running all of my devices on the Dongle-P that vaused buffer_full errors and it seems to be running stable on this FW, will run a few more days before i close this issue, thanks a lot!!

HenrikHills, 2025-02-17:
I dont get BUFFER_FULL errors anymore, but now I have the issue of network being overloaded and everything stops working, so I'll close this isseu 

```
2025-02-17 10:46:11z2m: Failed to ping 'Spisebord Spot 4' (attempt 1/1, ZCL command 0x001788010d4c1e89/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
```

closed 2025-02-17 by HenrikHills