Koenkk/zigbee2mqtt #26273: Z2M doesn't start after updating from 2.0.0 to 2.1.0 (or 2.1.1)

savaloi, 2025-02-09:
### What happened?

I run HA and Z2M in docker containers on my Ubuntu server running on an older mini PC. Z2M has been rock solid since I've had it.
I updated from Z2M 1.x to Z2M 2.0 when it was released earlier in the year and had no issues updating my configuration as listed in [Discussion #24364 ](https://github.com/Koenkk/zigbee2mqtt/discussions/24364).
Had to add this to my Z2M configuration.yaml:
serial:
  adapter: ember

I also had to update my Zigbee ZBDongle-E firmware to v7.4.5 when updating Z2M to v2.0.0 as per:
https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee

Again, this was working on Z2M 2.0.0 happily for a month or so.
Now that I've updated Z2M to v2.1.0 the container doesn't start and I receive various errors such as "ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT" error.

I tried:
Reverting back to Z2M v2.0.0, but this produced the same error.
Updated to Z2M v2.1.1, but this produced the same error.
I found [Issue #23861](https://github.com/Koenkk/zigbee2mqtt/issues/23861) tonight and updated my ZBDongle-E firmware to 8.0.2 which still produced the same issue.
I even ran up a fresh Z2M docker container all with roughly the same issue.

### What did you expect to happen?

Z2M would start correctly with no errors as it has for years

### How to reproduce it (minimal and precise)

Restart the container

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

8.0.2

### Adapter

Sonoff ZBDongle E

### Setup

Running HA, MQTT and Z2M via docker containers on a Ubuntu Server host

### Debug log

[log.log](https://github.com/user-attachments/files/18724214/log.log)

Nerivec, 2025-02-11:
Since you are seeing these right on serial opening, I'd say you are having interference issues with the USB.
Although, it does start after a few tries, and is not the issue that's causing the container to stop.
This is:
`MQTT failed to connect, exiting... (getaddrinfo EAI_AGAIN mosquitto)`

Aresitoo, 2025-02-14:
Having since latest update similar problems.

Zigbee 2.1.1-1

I'm using raspberry pi 4 running HAOS

Devices started to go randomly off line, reset of HA fixed sometimes the issue but now z2m isn't even starting up.

Also no changes have been made to the system nothing added all started after the update of z2mqtt

[11:21:28] INFO: Preparing to start...
[11:21:28] INFO: Socat not enabled
[11:21:28] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-14 11:21:33] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-14 11:21:33] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-14 11:21:33] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-14 11:21:34] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_226e5fd7cb93eb11bcbe1b4f3d98b6d1-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-14 11:21:34] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-14 11:21:34] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-14 11:21:34] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-14 11:21:34] error: 	z2m: Exiting...

Nerivec, 2025-02-14:
@Aresitoo This is not the same problem. Please open a new issue and include a `debug` log so we can take a closer look.

Aresitoo, 2025-02-14:
> [@Aresitoo](https://github.com/Aresitoo) This is not the same problem. Please open a new issue and include a `debug` log so we can take a closer look.

Will do. Thanks

savaloi, 2025-02-14:
> Since you are seeing these right on serial opening, I'd say you are having interference issues with the USB. Although, it does start after a few tries, and is not the issue that's causing the container to stop. This is: `MQTT failed to connect, exiting... (getaddrinfo EAI_AGAIN mosquitto)`

I’m guessing that appears to be an issue with mosquito then? Or is this saying that it can’t find the mosquito service?

Only work I’ve done on the device is updating the firmware. It’s on 8.0.2 - should I try a different version?

Nerivec, 2025-02-14:
The problem appears to be with mosquitto, nothing to do with the adapter.
EAI_AGAIN indicates a network problem (cannot resolve the address for mosquitto).

closed 2025-04-23 by github-actions[bot]