Koenkk/zigbee2mqtt #26274: [Wrong device]: Tuya ZTH05

ryanrio95, 2025-02-09:
### Link

https://www.zigbee2mqtt.io/devices/ZTH05.html

### Model

ZTH05

### Description

Temperature and humidity sensor

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZTH05.png

### Database entry

{"id":29,"type":"EndDevice","ieeeAddr":"0xa4c1389c54ce578a","nwkAddr":34436,"manufId":4098,"manufName":"_TZ3000_v1w2k9dd","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1026,1029],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_v1w2k9dd","dateCode":"","zclVersion":3,"appVersion":90,"powerSource":3}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2140}},"msRelativeHumidity":{"attributes":{"measuredValue":4200}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0030ca99e0","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":0,"maxRepIntval":1,"repChange":0}],"meta":{}}},"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1739100242095}

### Notes

Wrongly recognized as ZY-ZTH02

deurk, 2025-03-01:
Same here. Have 20+ different ones but the new batch I received is wrongly recognized, the older ones are not.

deurk, 2025-03-17:
With z2mqtt version 2.1.3 they are now correctly recognized.

Nipolf, 2025-05-12:
Same here with Z2M 2.3.0-1, ZTH05 recognized as ZY-ZTH02

A more recently bought is shown as WDS500A. (vendor claims iit is TH03, but this can't be found on Z2M device listand it looks like a ZTH05)