Koenkk/zigbee2mqtt #26278: [New device support]: MOES tuya ZHT-SR-GA-WH-MS

FabriceC01, 2025-02-09:
### Link

https://moeshouse.com/products/zigbee-star-ring-smart-thermostat-temperature-controller-water-boiler-electric-heating

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0xa4c138625ee32078","nwkAddr":62340,"manufId":4417,"manufName":"_TZE204_lpedvtvr","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�F;/i�F;/i\u0004G;/i\u0014G;/i$G;/i4G;/i�F;/i�F;/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_lpedvtvr","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1739098808680}

### Zigbee2MQTT version

2.1.1 

### Comments

Can you help to create a converter ? the device is not supported ...

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_lpedvtvr',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I don't know how to complete the definition...

FabriceC01, 2025-02-11:
Hello,
I'm far from being an expert, but if there is a tutorial somewhere on how to create an external definition, I would give it a try.

SNPN, 2025-02-11:
I guess it's about starting from here: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
But even the pairing does not seem to work out of the box.
Did you manage to detect the device in the Devices list?

FabriceC01, 2025-02-11:
The pairing works for me. The device is detected as TS0601 from _TZE204_lpedvtvr manufacturer. 
[image]

I've learned the "new devices support" but how to complete the configure section.

I've even asked chatGPT some help. It explains plenty of things and give a js file too but Z2M refuse to start because the config isn't correct.

SNPN, 2025-02-11:
Can you please explain the pairing process?
Which dongle are you using?

FabriceC01, 2025-02-11:
I'm using EZSP (Silicon Labs) dongle.
On the ZHT you shoud go to advanced settings then network pairing as described in manual.
On Z2M permit join as usual...

SNPN, 2025-02-14:
Ok I managed to get it sync with the dongle indeed.
I'm at the same point as you are.
Any help would also be appreciated since I understood that Tuya devices are not the easiest device to implement.

SNPN, 2025-02-15:
Good news, I managed to get all the DataPoints + description, directly from MOES support.
Now it's about connecting the dots.
If any expert can help, would be appreciated :)

```
DP ID | DP Name | DP Code | Data transfer type | Data Type | Data Attributes | Remarks
1 | Switch | switch | Issue and report | bool |   |  
2 | Work Mode | mode | Issue and report | enum | Enumerated   values：Manual,Temporary_Manual,Programming,Energy_Saving |  
16 | Current temperature | temp_current | Only report | value | Values range：0-900，Pitch：5，Unit：℃ |  
18 | The lower limit of temperature | lower_temp | Issue and report | value | Values range：50-150，Pitch：10，Unit：℃ |  
28 | Factory data reset | factory_reset | Issue and report | bool |   |  
32 | Sensor selection | sensor_choose | Issue and report | enum | Enumerated values：In,Al,Ou |  
34 | Set temperature ceiling | upper_temp | Issue and report | value | Values range：350-450，Pitch：10，Unit：℃ |  
39 | Child lock | child_lock | Issue and report | bool |   |  
47 | State of the valve | valve_state | Only report | enum | Enumerated values：open,close |  
48 | Backlight brightness | backlight | Issue and report | value | Values range：0-100，Pitch：1，Unit：% |  
50 | Set temperature | temp_set | Issue and report | value | Values range：50-450，Pitch：5，Unit：℃ |  
101 | Temp Calibration | temp_calibration | Issue and report | value | Values range：-10-10，Pitch：1，Unit：℃ |  
102 | Week Program 13 1 | week_program_13_1 | Issue and report | raw |   |  
103 | Week Program 13 2 | week_program_13_2 | Issue and report | raw |   |  
104 | Week Program 13 3 | week_program_13_3 | Issue and report | raw |   |  
105 | Week Program 13 4 | week_program_13_4 | Issue and report | raw |   |  
106 | Week Program 13 5 | week_program_13_5 | Issue and report | raw |   |  
107 | Week Program 13 6 | week_program_13_6 | Issue and report | raw |   |  
108 | Week Program 13 7 | week_program_13_7 | Issue and report | raw |   |  
109 | Floor temp. | floortemp | Only report | value | Values range：0-900，Pitch：5，Unit：°C |  
110 | Dead zone temp. | deadzonetemp | Issue and report | value | Values range：5-50，Pitch：5，Unit：°C |  
111 | High protect temp. | highprotecttemp | Issue and report | value | Values range：100-700，Pitch：10，Unit：°C |  
112 | Low protection temp. | lowprotecttemp | Issue and report | value | Values range：0-100，Pitch：10，Unit：°C |  
113 | Eco cool temp. | eco_temp | Issue and report | value | Values range：100-300，Pitch：10，Unit：°C |  
114 | Screen Time Set | screen_time_set | Issue and report | enum | Enumerated values：10S,20S,30S,40S,50S,60S |  
115 | Rgblight | rgblight | Issue and report | bool |   |  
```

SNPN, 2025-02-15:
See https://github.com/Koenkk/zigbee-herdsman-converters/pull/8803

acasajus, 2025-02-15:
Merged yesterday :)

acasajus, 2025-02-15:
Some quirks still to be ironed out though but the main functionality is there.

SNPN, 2025-02-15:
> Merged yesterday :)

Not sure how it's supposed to work. Will it be pushed in the next update or manual tweak to do?
Should I delete/add again the devices?

FabriceC01, 2025-02-15:
Thanks too for your work.
I'd like to test. can you explain how to manual import the config ?

acasajus, 2025-02-15:
It will probably be in the next release. You don't need to do anything. When you update it will appear supported automatically. Otherwise you can run by hand the development branch. But if you are not comfortable tweaking things I would suggest to just wait. Typically there's a release each month.

FabriceC01, 2025-02-15:
Thanks for your answer. I'm not an expert but I can try to tweak things... with just some advices for how to.
and i've got a backup of my VM... ;-)

besbar, 2025-04-17:
Hi,

Is there any update on the progress for the Ring Thermostat integration?

I’ve summarized the remaining work needed to fully support all its functionalities in this issue: [#25038](https://github.com/Koenkk/zigbee2mqtt/issues/25038). While the latest Zigbee2MQTT update added control for the RGB ring (on/off), features like mode switching and program scheduling still aren’t usable.

Let me know if there’s any way to help move this forward.

Thanks!

closed 2025-06-25 by github-actions[bot]