Koenkk/zigbee2mqtt #26281: IKEA E1524/E1810 sending double payloads to MQTT until z2m restart

adorobis, 2025-02-09:
### What happened?

Since already a few z2m versions, somewhere around end of 2024, after some time Ikea remote E1524/E1810 start sending double mqtt payloads for any button. This causes undesired behavior of my automations in HA as I have programmed actions based on double clicks. 
After restarting z2m everything comes back to normal for some time after which again those duplicated messages start to appear. Not sure how long it takes as I don't use those buttons vey frequently.

### What did you expect to happen?

Just a single entry in MQTT for a single click

### How to reproduce it (minimal and precise)

Click any button on the Ikea remote and double payloads are observed in MQTT

### Zigbee2MQTT version

 2.1.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

Docker install of z2m

### Debug log

[debug_log.log](https://github.com/user-attachments/files/18725401/debug_log.log)
Attached is full debug log.
Here are the duplicated entries from MQTT for 2 cases captured also in the log:
```
{"action":"toggle","battery":100,"identify":null,"linkquality":255,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}
{"action":"toggle","battery":100,"identify":null,"linkquality":255,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}
{"action":"toggle","battery":100,"identify":null,"linkquality":255,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}
{"action":"toggle","battery":100,"identify":null,"linkquality":255,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}
```

stortoaranci, 2025-02-11:
HI. same problem for me. after the upgrade, every time I click a button I see two payloads published twice instantly.

[image]

this is true for all buttons (Toggle, right_click, left_click,...)

i can temporary fix the issue by restarting the service after the boot.

i'm using linux red hat with mosquitto and home assistant.

Koenkk, 2025-02-12:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-12 by Koenkk

adorobis, 2025-02-13:
Thanks for quick fix! Let me test it for a while now :)

jdepypere, 2025-03-02:
I had the same issues with an IKEA RODRET E2201. Updated z2m to 2.1.2 and for now no issues, will keep monitoring it.