Koenkk/zigbee2mqtt #26284: Error: SRSP - SYS - ping after 6000ms occasionally starting Ubuntu virtual machine ( Proxmox )

kitopopo, 2025-02-09:
### What happened?

Dear @Koenkk ,

It’s been a while since we last communicated, as I didn’t have any issues and everything was running smoothly. However, I’ve recently migrated from a Raspberry Pi 5 to a mini PC running Proxmox, with two virtual machines—one running Home Assistant OS (HAOS) and the other running Ubuntu with Zigbee2MQTT.

I’m facing an occasional issue during the startup of the Ubuntu virtual machine. Usually, Zigbee2MQTT starts up fine, but occasionally, it takes 4 to 5 minutes to start, displaying the error "Error: SRSP - SYS - ping after 6000ms." After 4 or 5 minutes, Zigbee2MQTT eventually starts, but this delay is inconvenient. I can reproduce the problem by restarting the virtual machine multiple times, and in one of several restarts, this issue occurs.

It’s important to note that I have followed all the troubleshooting steps mentioned on your website for this error, including switching the USB extension cable to a shorter one. I’ve also tested with the HAOS virtual machine powered off to avoid any potential port conflicts, and I’ve updated the firmware of the coordinator.

While Zigbee2MQTT makes several unsuccessful startup attempts, I have verified that the adapter and port are correctly detected in Ubuntu using "ls -l /dev/serial/by-id/", where the output shows: usb-HIDDEN_USB_Serial-ifxx-portx -> ../../ttyUSB0 . I don’t understand why, if the device is correctly detected, it takes several minutes to start.

I’ve also reviewed the logs (attached below) and noticed a difference: In the log where Zigbee2MQTT fails to start, I see the line:
" zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload" and the error 6000 follows shortly after.  

In contrast, the log where Zigbee2MQTT starts correctly shows: "parsenext" and then detects  "zh:zstack: Detected znp version 'zStack3x0'"
and "zh:adapter:zstack:manager: beginning znp startup."

In both provided logs, the Zigbee adapter is correctly detected at the beginning of the Zigbee2MQTT startup process

I’ve spent about a month testing various solutions, but I still can’t figure out what’s going on. I hope you can provide some guidance or suggestions on how to proceed, as I’m running out of ideas. This zzh CC2652R adapter, which you recommended to me, has worked perfectly until now, and I would like to resolve this issue.

Thank you in advance for your support!

Best regards,

### What did you expect to happen?

I expected Zigbee2MQTT to start correctly right away without delays, as it does in most startups, and that the 'SRSP - SYS - ping after 6000ms' error would not occur.

### How to reproduce it (minimal and precise)

By restarting the Ubuntu virtual machine between 7 and 15 times, the issue can be reproduced.

### Zigbee2MQTT version

 2.1.1 commit

### Adapter firmware version

20240710

### Adapter

zzh CC2652R

### Setup

docker container inside of ubuntu VM

### Debug log

In both provided logs, the Zigbee adapter is correctly detected at the beginning of the Zigbee2MQTT startup process. In each case, after starting the service, the Zigbee adapter is properly loaded, and communication with it is established.

Log 1: It shows that the Zigbee adapter is detected on the /dev/ttyUSB0 port and set as the main port for the Zigbee coordinator, indicating it is ready for communication. The initial configuration is done without errors, and the Zigbee network starts functioning.

Log 2: Similarly, the /dev/ttyUSB0 port is recognized as the Zigbee adapter. The system continues to load devices and establish communication without issues, meaning the adapter is working correctly from the start. However, towards the end of the log, an error appears: the system fails to start zigbee-herdsman due to a connection issue with the adapter. This results in a failure to connect to the adapter after several attempts, as indicated by the error message.

Log1 - In this log zigbee starts correctly:

```
[2025-02-09 19:51:22] debug: 	z2m: Removing old log directory '/app/data/log/2025-02-09.19-30-58'
[2025-02-09 19:51:22] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-09 19:51:22] debug: 	z2m: Loaded state from file /app/data/state.json
[2025-02-09 19:51:22] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #1accb8b66b95b99658c50e4c0209edcb946a5fe2)
[2025-02-09 19:51:22] debug: 	z2m: sd-notify loaded
[2025-02-09 19:51:22] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-09 19:51:22] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6756,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-02-09 19:51:22] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS10"},{"path":"/dev/ttyS11"},{"path":"/dev/ttyS12"},{"path":"/dev/ttyS13"},{"path":"/dev/ttyS14"},{"path":"/dev/ttyS15"},{"path":"/dev/ttyS16"},{"path":"/dev/ttyS17"},{"path":"/dev/ttyS18"},{"path":"/dev/ttyS19"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS20"},{"path":"/dev/ttyS21"},{"path":"/dev/ttyS22"},{"path":"/dev/ttyS23"},{"path":"/dev/ttyS24"},{"path":"/dev/ttyS25"},{"path":"/dev/ttyS26"},{"path":"/dev/ttyS27"},{"path":"/dev/ttyS28"},{"path":"/dev/ttyS29"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyS30"},{"path":"/dev/ttyS31"},{"path":"/dev/ttyS4"},{"path":"/dev/ttyS5"},{"path":"/dev/ttyS6"},{"path":"/dev/ttyS7"},{"path":"/dev/ttyS8"},{"path":"/dev/ttyS9"},{"path":"/dev/ttyUSB0","manufacturer":"HIDDEN","pnpId":"HIDDEN","vendorId":"HIDDEN","productId":"HIDDEN"}]
[2025-02-09 19:51:22] debug: 	zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/ttyUSB0
[... 62 lines cut ...]
[2025-02-09 19:51:19] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-02-09 19:51:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-02-09 19:51:19] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,10,0,105,241,1,1,0,185,0,123,109,216,0,0,5,16,165,0,0,0,154,4,28,11]
[2025-02-09 19:51:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,10,0,105,241,1,1,0,185,0,123,109,216,0,0,5,16,165,0,0,0,154,4,28,11]
[2025-02-09 19:51:19] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,105,241,1,1,0,185,0,123,109,216,0,0,5,16,165,0,0,0,154,4,28] - 11
[2025-02-09 19:51:19] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":61801,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":185,"securityuse":0,"timestamp":14183803,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,165,0,0,0]}}
[2025-02-09 19:51:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-02-09 19:51:21] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-09 19:51:21] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-09 19:51:21] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-09 19:51:21] error: 	z2m: Exiting...
[2025-02-09 19:51:21] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.js:80:19)
    at Zigbee2Mqtt.start (/app/zigbee2mqtt.js:101:19)
    at Object.<anonymous> (/app/zigbee2mqtt.js:142:9)
    at Module._compile (node:internal/modules/cjs/loader:1213:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1283:10)
    at Module.load (node:internal/modules/cjs/loader:1081:32)
    at Function.Module._load (node:internal/modules/cjs/loader:923:12)
    at Module.require (node:internal/modules/cjs/loader:1061:19)
[2025-02-09 19:51:21] debug: 	z2m: Exiting...

```

kitopopo, 2025-02-13:
Dear friend @Koenkk ,

Do you have any suggestions on how to resolve this issue? Every day I reproduce the same problem in each Ubuntu backup, as the machine stops. I don't know how to fix it anymore. What I have discovered is that if I stop the Zigbee2MQTT container and restart it, Zigbee always starts without issues, and the failure doesn't occur. It seems the problem happens during the virtual machine startup. Could it be an issue with the CC2652R or its firmware? I don't see anything unusual in the virtual machine. When I manage to reproduce the problem and Docker doesn't detect the coordinator, I can run a test, and Ubuntu always detects it correctly in ttyUSB0.

Here are some additional details about the device:

```
ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Feb 13 2025 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0

ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Feb 13 2025 /dev/ttyUSB0

lsmod | grep ch341
ch341 24576 1
usbserial 57344 3 ch341
```
Many thanks for your work and dedication. I look forward to receiving a prompt response.

Kind regards.

Koenkk, 2025-02-14:
I know little about Proxmox, bit you can find a tip from another user here: ZHT-SR

kitopopo, 2025-02-15:
> Sé poco sobre Proxmox, pero puedes encontrar un consejo de otro usuario aquí: https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#error-regular-crashes-with-timeout-errors-or-failure-to-start-after-the-serial-port-is-opened

Thank you for your response dear @Koenkk . I’ve already tried it; in fact, I’ve tested it with three different cables, and the issue persists. I’ve also tried all the solutions mentioned in this article: https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html. Do you think the issue could be with Proxmox? Because the adapter is detected correctly in Ubuntu at the time of the failure with all possible commands, but communication is not established. If you check the logs, the message 'zh:zstack: Detected znp version 'zStack3x0'' never appears. I’ll keep investigating. For now, an alternative solution has been to install Zigbee2MQTT in an LXC container on Proxmox. Interestingly, there’s no documentation for this installation method on Zigbee’s official page, but I managed to get it working anyway. I’ll continue testing both methods, although I prefer Docker over npm. Thank you for your help.

closed 2025-04-25 by github-actions[bot]