Koenkk/zigbee2mqtt #26286: Problem with AC Frequency on _TZE204_81yrt3lo - PJ1203A

ViscountHermit, 2025-02-10:
### What happened?

Since the update to version 2.1.0 my PJ1203A electricity meter no longer updates the frequency, it remains stuck at 49hz, before it would vary a bit between 49.01 to 49.99.

I guess there was a change in the definition, I had been running on 1.36.1-1.

### What did you expect to happen?

I expected to see variance in frequency over time. 

### How to reproduce it (minimal and precise)

Run version 2.1.0 of Z2m. 

### Zigbee2MQTT version

2.1.9

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

Standard Z2m install on a home assistant x86 virtual machine install. 

### Debug log

Z2M@192.168.0.135

Show only
all
Filter by text
zigbee2mqtt/ZSM-001/Understairs
Log level
info
Logging level
Logs limit
500
 
info 2025-02-10 00:50:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.112,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:46.369Z","linkquality":116,"power_a":198.4,"power_ab":198.4,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:37.195Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":236.7}'
info 2025-02-10 00:50:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.112,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:46.433Z","linkquality":120,"power_a":198.4,"power_ab":198.4,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:37.195Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":236.7}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.112,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:47.752Z","linkquality":116,"power_a":198.4,"power_ab":198.4,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:37.195Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":236.7}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.112,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:47.809Z","linkquality":116,"power_a":198.4,"power_ab":198.4,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:37.195Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.112,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:47.865Z","linkquality":116,"power_a":198.4,"power_ab":198.4,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:37.195Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.112,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:47.921Z","linkquality":116,"power_a":198.4,"power_ab":198.4,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:37.195Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:47.976Z","linkquality":120,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.030Z","linkquality":116,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.108Z","linkquality":116,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.167Z","linkquality":120,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:37.537Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.221Z","linkquality":108,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:48.221Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.275Z","linkquality":112,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:48.275Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.329Z","linkquality":116,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:48.275Z","update_frequency":10,"voltage":237.1}'
info 2025-02-10 00:50:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:50:48.385Z","linkquality":116,"power_a":203.2,"power_ab":203.2,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:50:47.922Z","timestamp_b":"2025-02-10T00:50:48.275Z","update_frequency": [... cut ...]
info 2025-02-10 00:51:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZSM-001/Understairs Pantry', payload '{"ac_frequency":49,"current_a":1.126,"current_b":0,"energy_a":3067,"energy_b":0,"energy_flow_a":"consuming","energy_flow_b":"consuming","energy_produced_a":61.58,"energy_produced_b":0,"last_seen":"2025-02-10T00:51:09.889Z","linkquality":116,"power_a":202.3,"power_ab":202.3,"power_b":0,"power_factor_a":75,"power_factor_b":100,"timestamp_a":"2025-02-10T00:51:09.392Z","timestamp_b":"2025-02-10T00:51:09.724Z","update_frequency":10,"voltage":236.9}'

Koenkk, 2025-02-10:
fixed in 2.1.1

closed 2025-02-10 by Koenkk

ViscountHermit, 2025-02-11:
Updated to latest release, still no change with the frequency. It's almost as if it's just truncating anything after the decimal point.

----

Zigbee2MQTT version
2.1.1 commit: unknown
Coordinator type
EZSP v12
Coordinator revision
7.3.1.0 build 176
Coordinator IEEE Address
0x003cxxxxxxxxxxxx
Frontend version
0.9.4
zigbee-herdsman-converters version
21.31.0
zigbee-herdsman version
3.2.5