Koenkk/zigbee2mqtt #26288: Innr SP 240  seMetering.read UNSUPPORTED_ATTRIBUTE

buschbaby, 2025-02-10:
### What happened?

When configurating an Innr SP240 (Firmware 20240705,1.7.23) on Z2M homeassistant i am getting the following errors:

[2025-02-10 07:46:16] error: 	z2m: Failed to configure 'Waschmaschine', attempt 3 (Error: ZCL command 0xa4c138abc11896a0/1 seMetering.read(["divisor","multiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29)
    at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:1944:37)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

z2m: Failed to configure 'Waschmaschine', attempt 6 (Error: ZCL command 0xa4c138abc11896a0/1 seMetering.read(["divisor","multiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:1944:37) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

### What did you expect to happen?

Configuration success.

### How to reproduce it (minimal and precise)

Add an Innr SP240 or reconfigure one plug. Also a restart triggers that event.

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

20240710

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Addon on Homeassistant Raspberry PI 4

Koenkk, 2025-02-10:
@GerryInnr is it expected that the SP 240 doesn't have the divisor attribute with this firmware?

buschbaby, 2025-02-11:
My fault then, I did not check for an update after installing my new plugs. With the new firmware it's working correct. Thanks for your support!