Koenkk/zigbee2mqtt #26290: Replace `color_temp` to `color_temp_kelvin` for _TZ3000_riwp3k79

musapinar, 2025-02-10:
### What happened?

Error in the system logs : 
`Got `color_temp` argument in `turn_on` service, which is deprecated and will break in Home Assistant 2026.1, please use `color_temp_kelvin` argument`

Zigbee Model : TS0505A
Zigbee Manufacturer : _TZ3000_riwp3k79
Description : Livarno Home RGB+CCT LED light strip 2m
Manufacturer : [Lidl](https://www.zigbee2mqtt.io/supported-devices/#v=Lidl)
Model : [HG06104A](https://www.zigbee2mqtt.io/devices/HG06104A.html#lidl-hg06104a)

### What did you expect to happen?

Can we please replace `color_temp` to `color_temp_kelvin` for _TZ3000_riwp3k79 ? (aka Lidl LED strip)
Thank you

Currently (example) :
`
[...]
actions:
  - entity_id: light.led_strip
    action: light.turn_on
    data:
      brightness_pct: 50
      color_temp: 500
`

Expected : 
`
[...]
actions:
  - entity_id: light.led_strip
    action: light.turn_on
    data:
      brightness_pct: 50
      color_temp_kelvin: 500
`

### How to reproduce it (minimal and precise)

- pair an led strip of "_TZ3000_riwp3k79"
- code an automation that controls color temp

What happens : You need to use "color_temp" to make it work
Expected behavior : being able to use "color_temp_kelvin" instead of "color_temp". Currently "color_temp_kelvin" is not recognized.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

haos addon, running on oracle vm

### Debug log

n\a

Koenkk, 2025-02-10:
You need to update your automations to set color_temp_kelvin, this is not something that can be fixed from z2m

musapinar, 2025-02-10:
@Koenkk The automation does not work with `color_temp_kelvin`. `color_temp_kelvin` is not recognized as a valid command. It does not raise any error in the system logs. It's just not interpreted as a replacement/alternative for `color_temp`.

Koenkk, 2025-02-10:
>  color_temp_kelvin is not recognized as a valid command.

Could you provide the z2m debug log when sending a `color_temp_kelvin` command from HA?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

musapinar, 2025-02-10:
@Koenkk 

with `color_temp_kelvin` :
```
[2025-02-10 21:57:47] error: 	z2m: Publish 'set' 'color_temp' to 'led_strip' failed: 'TypeError: Cannot read properties of undefined (reading 'x')'
[2025-02-10 21:57:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/led_strip', payload '{"brightness":127,"color":{"h":25,"hue":25,"s":95,"saturation":95,"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":500,"do_not_disturb":false,"effect":null,"linkquality":127,"state":"ON"}'
```
having automation : 
`alias: Turn on led strip
description: Turn on led strip
triggers:
  - entity_id: binary_sensor.capteur_presence_humaine_atelier_presence
    to: "on"
    trigger: state
  - entity_id: sensor.capteur_presence_humaine_atelier_state
    to: move
    trigger: state
conditions:
  - condition: state
    entity_id: light.led_strip
    state: "off"
  - condition: state
    entity_id: person.musa
    state: home
  - condition: state
    entity_id: cover.atelier
    state: closed
actions:
  - target:
      entity_id: light.led_strip
    data:
      brightness_pct: 50
      color_temp_kelvin: 500
    action: light.turn_on
mode: queued
`

with `color_temp` : 

`[2025-02-10 22:00:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/led_strip', payload '{"brightness":127,"color":{"h":25,"hue":25,"s":95,"saturation":95,"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":500,"do_not_disturb":false,"effect":null,"linkquality":116,"state":"ON"}'`
having automation : 
`alias: Turn on led strip
description: Turn on led strip
triggers:
  - entity_id: binary_sensor.capteur_presence_humaine_atelier_presence
    to: "on"
    trigger: state
  - entity_id: sensor.capteur_presence_humaine_atelier_state
    to: move
    trigger: state
conditions:
  - condition: state
    entity_id: light.led_strip
    state: "off"
  - condition: state
    entity_id: person.musa
    state: home
  - condition: state
    entity_id: cover.atelier
    state: closed
actions:
  - target:
      entity_id: light.led_strip
    data:
      brightness_pct: 50
      color_temp: 500
    action: light.turn_on
mode: queued
`

[... 16 lines cut ...]
[2025-02-10 22:18:33] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":102}
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,201,74,0,1]
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,201,74,0,1]
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [201,74,0] - 1
[2025-02-10 22:18:33] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,0,3,201,74,1,1,0,134,0,228,193,99,0,0,5,8,20,11,10,0,201,74,29,26]
[2025-02-10 22:18:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,0,3,201,74,1,1,0,134,0,228,193,99,0,0,5,8,20,11,10,0,201,74,29,26]
[2025-02-10 22:18:34] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,201,74,1,1,0,134,0,228,193,99,0,0,5,8,20,11,10,0,201,74,29] - 26
[2025-02-10 22:18:34] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":19145,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":134,"securityuse":0,"timestamp":6537700,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,20,11,10,0]}}
[2025-02-10 22:18:34] debug: 	zh:controller: Received payload: clusterID=768, address=19145, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=134, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":20,"commandIdentifier":11},"payload":{"cmdId":10,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-02-10 22:18:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-02-10 22:18:34] debug: 	zhc:light: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0xe0798dfffe53ae07!
[2025-02-10 22:18:34] error: 	z2m: Publish 'set' 'color_temp' to 'led_strip' failed: 'TypeError: Cannot read properties of undefined (reading 'x')'
[2025-02-10 22:18:34] debug: 	z2m: TypeError: Cannot read properties of undefined (reading 'x')
    at Function.fromObject (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/color.ts:222:16)
    at Function.fromMireds (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/color.ts:235:24)
    at Object.syncColorState (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/color.ts:779:36)
    at Object.convertSet (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/converters/toZigbee.ts:182:33)
    at Object.convertSet (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/converters/toZigbee.ts:1251:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:250:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)
[2025-02-10 22:18:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/led_strip', payload '{"brightness":127,"color":{"h":12,"hue":12,"s":100,"saturation":100,"x":0.621,"y":0.3705},"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":810,"do_not_disturb":false,"effect":null,"linkquality":134,"state":"ON"}'
[2025-02-10 22:18:34] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,110,226,0,14]`

alexsaas, 2025-02-11:
```
actions:
  - target:
      entity_id: light.led_strip
    data:
      brightness_pct: 50
      color_temp_kelvin: 500
    action: light.turn_on
```

Are you sure that the target light supports 500 Kelvin? That seems too low to me. Try values ​​like 2500 Kelvin or 5000 Kelvin in your automation and see if the error still exists.

closed 2025-02-11 by musapinar