Koenkk/zigbee2mqtt #26292: E1743 Firmware version label does not update

vogtmh, 2025-02-10:
### What happened?

In the last days, I finally updated all my Ikea Dimmers (E1743) via OTA to the latest version. 

Before the update: 24.4.5 / 604241925
After the update: 24.4.6 / 604241926

However, some of my Dimmers still show version 24.4.5 in the About tab, while correctly showing 604241926 in the state tab.

[image]
[image]

### What did you expect to happen?

The label should should 24.4.6 in the "Firmware version" 

### How to reproduce it (minimal and precise)

Update from 24.4.5 to 24.4.6

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Addon in Homeassistant on Home server

weljajoh, 2025-02-10:
Forcing an Interview with the blue 'i' button on the bottom most likely corrects this.

vogtmh, 2025-02-10:
Is it safe to force an interview on a working device? I had some bad experiences with this where the device stopped working afterwards.

vogtmh, 2025-02-10:
I just tried it on one of my dimmers and it still shows up as 24.4.5

weljajoh, 2025-02-10:
> I just tried it on one of my dimmers and it still shows up as 24.4.5

Did you wake the device by pushing a button prior to forcing the interview?

vogtmh, 2025-02-10:
Of course I did. After a few seconds, the interview was confirmed as succesful. The version is still 24.4.5.

closed 2025-04-19 by github-actions[bot]