Koenkk/zigbee2mqtt #26293: INNR SP 240 showing way too high energy consumption

cranxi, 2025-02-10:
### What happened?

Since I updated Z2M to 2.1.1 the INNR smart plugs are registering way too high energy consumption. For example one laundry cycle is 14 kWh. This is not possible. The smart plugs energy consumption meter jumped exactly 1000 kWh up on the day I updated Z2M to 2.1.1. All the meters did this on that day.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20250109

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on HassOS

cranxi, 2025-02-10:
Ok, so restored backup so that Z2M is back to 2.1.0, problem still excist.

Around the same time a new firmware for the SP 240 came out, version 1.9.27.

Could be the problem with the new firmware?

buschbaby, 2025-02-10:
Same configuration, same problem. It even seems that the value is also not really updating constantly, only sometimes. I also opened up an error when recofigurating the device. Does this happen on your setup too?

cranxi, 2025-02-10:
@buschbaby No it does not. I tried to unpair and pair again with Z2M, so the smart plug is factory resetted. Still same problem. 4.2 kWh in half an hour for the washingmachine.

cranxi, 2025-02-10:
This is one cycle for the washingmachine:

Seems to update very frequently. Before is was only updating once a day.

[image]

closed 2025-02-10 by cranxi

shilga, 2025-02-20:
@cranxi I noticed the same issue after updating the firmware. Did the meter jump back down to the old value after re-configuring?

thomasvnl, 2025-05-14:
Even after reconfigure, I have a SP 240 that is pulling around 80W that is reporting in Z2M 2.3.x that it has used 80WH (0.08kWh) in less than 10 minutes. Others having issues? I updated to Z2M 2.x because of a fix for these devices.