Koenkk/zigbee2mqtt #26299: Nieobsługiwany czujnik temperatury i wilgotności z zewnętrzną sondą TS0601

Mikmen23, 2025-02-10:
### Link

https://led-hurt.pl/pl/p/Czujnik-temperatury-i-wilgotnosci-zewnetrzny-z-wodoodporna-sonda-ZigBee-TUYA-LCD/5008?gad_source=1&gclid=CjwKCAiA5Ka9BhB5EiwA1ZVtvG7OJfAeSEprUbBKHl4teOxgMHvdvdxZ_mGzqtuKSmqmSoQ_ah8fwxoCPX0QAvD_BwE

### Database entry

Nie wiem gdzie szukać

### Zigbee2MQTT version

2.1.1-1

### Comments

W ZHA jest już chyba kompatybilny czujnik
https://github.com/zigpy/zha-device-handlers/issues/3647

Na Zigbee2Mqtt wyszukuje mi go ale nie pobiera żadnych wartości, zgodnie z poradnikiem Z2M przechodzę do “konsoli” urządzenia TS0601 i pojawia mi się takie coś:

const definition = {
zigbeeModel: ,
model: ‘TS0601’,
vendor: ‘_TZE284_qyflbnbj’,
description: ‘Automatically generated definition’,
extend: ,
meta: {},
};

jakąkolwiek zmianę chce wprowadzić to niestety nie zapisuje mi się.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_qyflbnbj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Czujnik jest wyszukiwany w urządzeniach Z2M z informacją:
Producent: Niewspierane
Model: TS0601
LQI: 98
Zasilanie: ?

[image]

tomtom13, 2025-02-10:
Nie chce byc dupkiem, ale watpie ze ktokolwiek ci pomoze jak begreport bedzie po polsku. Ale tak na przyszlosc, na HASS mozna miec ZHA i Z2M, tylko potrzebne sa dwie bramki - radzilbym jest ustawic przynajmniej z 5 kanalami odstepu. Wtedy mozna sprawdzic czy cos na zha dziala czy nie.

mrmaximas, 2025-02-10:
https://github.com/Koenkk/zigbee2mqtt/issues/26251
it's works in dev

Mikmen23, 2025-02-10:
Nie bardzo rozumiem "działa w środowisku deweloperskim" tzn. jest już dodany czujnik tylko narazie nie jest dostępny dla użytkownika końcowego?

Mikmen23, 2025-02-10:
ok, dzięki za wyjaśnienie.

mrmaximas, 2025-02-10:
Od teraz staraj się pisać po angielsku. Pozdrowienia z Białorusi.

Mikmen23, 2025-02-10:
Thanks for your help, I didn't know that only English applications were considered. I will know for the future.

mrmaximas, 2025-02-10:
@Mikmen23 people from hundreds of countries are here to help you write better English. Have a good night!

closed 2025-04-19 by github-actions[bot]