Koenkk/zigbee2mqtt #26307: Can't get Zigbee2MQTT 2.1.1-1 with SMLight SLZB-06 running?

mandltho, 2025-02-10:
### What happened?

Dear All, 

Please can you help me. I am trying to install Zigbee2MQTT v. 2.1.1-1 on a freshly installed HAOS / Rapberry PI4. The Zigbee Coordinator to use is a SMLight SLZB-06 with frmware 2.7.1 and connected via Ethernet into the same network. The Raspberry is down in the cellar an I have difficulties on our house to reach the attic.

I can't get the add-on started, it runs into an timeout and watchdog is re-starting every 6000ms. 

I hace looked at all fixes from here https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html but no fix is pointing me into the right direction.

If you miss a Zigbee channel - as the addon does not start, no Zigbee channel ahs been used on this installation.

I had another HAOS installation until recently using a Sonoff P - USB Dongle where Zigbee2MQTT (but not 2.1.1-1) was running properly.

To get rid of a lot of test integrations, I decided to set up a fresk HAOS system and start fresh with Zigbee2MQTT. For the Zigbee equipment I want to use in future, ZHA is only the secnd best solution.

Any help is more than appeciated - I case I oversaw some settings to be made!

Many thanks in advance
Thomas

Syslog:
[22:55:23] INFO: Preparing to start...
[22:55:23] INFO: Socat not enabled
[22:55:23] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-10 22:55:28] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-10 22:55:28] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-10 22:55:28] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-10 22:55:28] info: 	zh:zstack:znp: Opening TCP socket with 192.168.178.119:6638
[2025-02-10 22:55:28] info: 	zh:zstack:znp: Socket connected
[2025-02-10 22:55:28] info: 	zh:zstack:znp: Socket ready
[2025-02-10 22:55:28] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-10 22:55:29] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-10 22:55:47] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-10 22:55:47] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-10 22:55:47] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-10 22:55:47] error: 	z2m: Exiting...
[2025-02-10 22:55:48] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5) 

Zigbee2MQTT add-on config.yaml:
version: 4
homeassistant:
  enabled: true
advanced:
  network_key:
    - 205
    - 29
    - 102
    - 154
    - 212
    - 127
    - 182
    - 196
    - 12
    - 49
    - 133
    - 120
    - 13
    - 119
    - 187
    - 65
  pan_id: 9915
  ext_pan_id:
    - 168
    - 69
    - 24
    - 206
    - 72
    - 7
    - 16
    - 38
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: *pwd*
serial:
  port: tcp://192.168.178.119:6638
  adapter: zstack
  baudrate: 115200
frontend:
  enabled: true
  port: 8099

SLZB06:
> Radiomodule mode:	Zigbee coordinator
> Connection mode:	Ethernet mode
> Ethernet connected:	Yes
> Wi-Fi client turned On:	No
> Wi-Fi AP turned On:	No
> Socket connected:	No
> Device uptime:	0d 01h 21m
> Socket uptime:	Not connected
> MDNS address	SLZB-06-1.local
> Internet access	Yes
> Automatic Zigbee Update	Off
> VPN Enabled	No
> VPN Connection Active	No

> Ethernet enabled	Yes
> Connected	Yes
> DHCP	On
> IP Address:	192.168.178.119
> Subnet mask:	255.255.255.0
> Default gateway:	192.168.178.1
> Connection speed:	100Mbps
> MAC address:	20:43:A8:64:4C:EB
> DNS1	fd98:d13d:508e:0:3e37:12ff:fec2:bec6
> DNS2	2001:9e8:9692:9d00:3e37:12ff:fec2:bec6

> Radiomodule mode  [CC2652P]
> 
>  Zigbee Coordinator
>  
> Socket Port:
> 6638
> Serial Speed:
> 115200

HA Version:

> Core   2025.2.1
> Supervisor   2025.02.0
> Operating System   14.2
> Frontend   20250205.0

### What did you expect to happen?

> __

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

2.7.1

### Adapter

SMLight SLZB-06

### Setup

Add-on on HAOS installed on Raspberry PI4

### Debug log

[2025-02-10 22:38:17] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-10 22:38:18] error: 	z2m: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevicesIterator')
    at Zigbee.devicesIterator (/app/lib/zigbee.ts:353:44)
    at devicesIterator.next (<anonymous>)
    at OnEvent.stop (/app/lib/extension/onEvent.ts:36:20)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))
[2025-02-10 22:38:18] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-10 22:38:18] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-10 22:38:18] info: 	zh:zstack:znp: Opening TCP socket with 192.168.178.119:6638
[2025-02-10 22:38:18] info: 	zh:zstack:znp: Socket connected
[2025-02-10 22:38:18] info: 	zh:zstack:znp: Socket ready
[2025-02-10 22:38:18] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-10 22:38:19] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-10 22:38:37] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-10 22:38:37] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-10 22:38:37] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-10 22:38:37] error: 	z2m: Exiting...
[2025-02-10 22:38:37] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

churchofnoise, 2025-02-12:
Same issue here. Was running fine until a few hours ago, all of a sudden Z2m is not running anymore.
Trying to start yields the same issues as described by the topic starter here.
SLZB-06 adapter web interface works, rebooting the adapter does not solve the issue.

piposant, 2025-02-12:
The same problem

churchofnoise, 2025-02-12:
Managed to get z2m running by rebooting the zigbee chip (using the Web interface) and then restarting z2m within home assistant. Still running so far.

marcinjahn, 2025-02-12:
Similar issue on my end. I tried two connection methods:
- Ethernet
- USB

On Ethernet it runs, sometimes for a few hours, and then eventually the connection gets halted (with various errors).
On USB connection, it never starts properly, and fails with the same error as OP has.

mandltho, 2025-02-12:
Update:

Looking around i was directed to this url: https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#error-srsp-sys-ping-after-6000ms-for-zstack-or-host-fatal-error-for-emberznet 

Although not linked to the SLZB-06 it pointed me towards the Zigbee chip of the device.
I flashed the zigbee firmware from the SLZB-06 control panel (settings and tools - firmware update) to this dev version:

Dev firmware
Revision: 20240710
Release notes
Public latest Coordinator release for CC2652P chips [10-Jul-2024]:

The SLZB-06 has been re-booted and I startet zigbee2mqtt.

It works, so I will close this issue on the weekend, to allow to add some further comments

It it worth to note that if you update the configuration.yaml of the zigbee2mqtt then do this in yaml, not in the GUI.

Many thanks to you, Koen, for this marvellous project!

Take care
Thomas

churchofnoise, 2025-02-13:
@mandltho Just to let you know, I already had the 20240716 dev firmware on my device when the issue occurred, I would dare assume that the solution for you might be linked to what helped me, i.e. the reboot of the Zigbee chip (in your case after the update of the firmware).
Something to consider...

fh-TurbaAI, 2025-02-13:
Had the same issue on my `SLZB-06P7` yesterday. Simple reboot of the adapter didn't work, but power-cycling it did, I suppose this is the same solution as described by others.

kacperziolo, 2025-02-15:
I had the same issue with my SLZB-06

Zigbee2MQTT version: 2.1.1-1
Current core firmware version: v0.9.9 (Mar 3 2023)

I have to add 
adapter: zstack
in serial section of my config.yaml in zigbee2mqtt (not in zigbee2mqtt GUI !)

After restarting Zigbee throught SLZB-06 web UI and starting Z2M straight after that it worked. 

Thanks @mandltho and @telmahjohn

TheCarpetMerchant, 2025-02-18:
I don't get the exact same error with my SLZB-06 on Z2M 2.1.1-1 :

```
[17:54:20] INFO: Preparing to start...
[17:54:20] INFO: Socat not enabled
[17:54:21] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-18 17:54:26] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-18 17:54:26] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-18 17:54:26] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-18 17:54:26] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"52946cd27f84ed11b083cd9f9d1cc348","pnpId":"usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_52946cd27f84ed11b083cd9f9d1cc348-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 2
[2025-02-18 17:54:26] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-18 17:54:26] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-18 17:54:26] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-18 17:54:26] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-18 17:54:26] error: 	z2m: Exiting...
[2025-02-18 17:54:26] error: 	z2m: Error: Error Resource temporarily unavailable Cannot lock port
```

This is what occurs whether I add "adapter: zstack" or "port: /dev/ttyUSB0" or both or none. I supposed they're the default. I tried both GUI and configuration.yaml.
I'm only using the USB connection, no ethernet.

The lack of clear logs is the problem here ; why has herdsman failed to start ? As a new user, this is very discouraging. Not only do the defaults & following the docs fail, but the error messages themselves don't provide a clear indication of what's wrong :/

TheCarpetMerchant, 2025-02-18:
> I had this with my new SLZB-06 (CC2652P chips).
> 
> Zigbee2MQTT version: 2.1.1-1 Current core firmware version: v2.7.1 Zigbee Coordinator Revision: 20221226
> 
> First I tried to update to Zigbee Coordinator Revision: 20240710
> 
> No dice...
> 
> I read **very** carefully the above entry from [@mandltho](https://github.com/mandltho) especially: "_It it worth to note that if you update the configuration.yaml of the zigbee2mqtt then do this in yaml, not in the GUI._"
> 
> This **solved it** for me. Add the Zigbee2MQTT config generated from the SLZB-06 Control Panel to the zigbee2mqtt/configuration.yaml file (watch out for the conflict if it has the _advanced_ transmit_power: 20 entry)
> 
> Kudos to [@mandltho](https://github.com/mandltho)

How can I make the /config/zigbee2mqtt/configuration.yaml be even taken into account ? I've noticed that changes I make there are ignored, but the ones made in the HA UI are.

closed 2025-04-24 by mandltho