Koenkk/zigbee2mqtt #26309: Does not recognize Sonoff USB dongle?

sharpcall, 2025-02-11:
### What happened?

I see the following errors when starting Z2MQTT. Below is my simple config picture:

[image]

I have also tried with the following settings added and it still did not work:

[image]

Here is my error log:
`[21:14:43] INFO: Preparing to start...
[21:14:43] INFO: Socat not enabled
[21:14:44] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-10 21:14:45] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-10 21:14:45] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-10 21:14:45] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-10 21:14:45] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-10 21:14:45] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-10 21:14:45] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-10 21:14:45] error: 	z2m: Exiting...
[2025-02-10 21:14:45] error: 	z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[2025-02-10 21:14:45] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364`

### What did you expect to happen?

ZMQTT to run without errors and detect the Sonoff USB coordinator.

### How to reproduce it (minimal and precise)

Recreate my environment and same will happen

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

Not sure / Where to see this in HaOS?

### Adapter

Sonoff Zigbee USB dongle

### Setup

HaOS

azalin1975, 2025-02-12:
> [@azalin1975](https://github.com/azalin1975) usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0 = zstack

Right. Then adding `adapter: zstack` as you mentioned should work. Like you mentioned in you reply to the OP.

mrmaximas, 2025-02-12:
@azalin1975 usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_хххххх = ember

closed 2025-04-21 by github-actions[bot]