Koenkk/zigbee2mqtt #26320: Routers/coordinator down with message BUFFER_FULL

m1r4x, 2025-02-11:
### What happened?

Today, for the first time, my zigbee network went down (all routers and coordinator seem offline). After installing last HASS update 2025.2. (Maybe a coincidence?). I tried to revert to the previous version but same issue.
In the log I only get error messages about buffer full. Cannot add new devices.
Please help me 🙏

### How to reproduce it (minimal and precise)

After the ha update.
Tried several time to restart ha and the add on.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle-P

### Setup

Add-on on HA OS on x86-64

### Debug log

error: 	zh:controller:device: Handling of poll check-in from 0x881a14fffee9d8a2 failed (ZCL command 0x881a14fffee9d8a2/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":63668,"destendpoint":1,"srcendpoint":1,"clusterid":32,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,8,0,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))

m1r4x, 2025-02-12:
I solved (at the moment) unplugging and plugging again the coordinator adapter...this fixed the network.
I suspect a tuya sensor (pir and radar) was the root cause. From some days it ever showed presence and movement also if not true...I thought was because the low battery but I started to get buffer full messages...
Could be this the problem of such issue?

KDC666, 2025-02-13:
I had the same (or at least very similar) issue. Did update HA to 2025.2.1 and zigbee2mqtt stopped updating values (noticed very hot room next day because temperature sensor stuck at 20°C and AC should heat up to 21°C). Tried different things (updated Zigbee2mqtt to latest version, restarted device/service seral times, played around with the configuration, ...) and today did some more googeling and found this issue here. I've got the same Sonoff firmware and unplugging the stick did the job, at least partially. I had to remove and rebind most of my zigbee devices (I've got only 5 or 6) and now they are working again.

closed 2025-04-22 by github-actions[bot]