Koenkk/zigbee2mqtt #26321: missing Illuminance - Aqara RTCGQ11LM Motion sensor

CaptMilkbeard, 2025-02-12:
### What happened?

Hello everyone,

Until recently I used Illuminance for certain automations. Since the update, the entity is no longer transferred to HA. 

### What did you expect to happen?

I would like to have the value of the Illuminance as an entity to use it in automations as a trigger etc.

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

20250205.0

### Adapter

SONOFF ZBDongle-P

### Setup

HA Addon on a mini PC

mrmaximas, 2025-02-12:
https://github.com/Koenkk/zigbee2mqtt/discussions/24198

CaptMilkbeard, 2025-02-12:
> [#24198](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

Thanks! 
I had used the search but apparently wasn't looking for the right thing.

The post itself didn't help me, but a few more links led me to the hint that it is now possible again. A bit cumbersome and unfortunately not quite as described but through a few other user comments I was able to find the deactivated entity.

But without the nudge in the right direction I probably wouldn't have found it. Many thanks for that