Koenkk/zigbee2mqtt #26325: Homeassistant reports: Invalid color mode 'xy'

runningman84, 2025-02-12:
### What happened?

I get this error in home assistant:
2025-02-11 18:32:10.248 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'xy' received for entity light.bathroom_light_1

The device looks like this in zigbee2mqtt:
```
{
    "brightness": 254,
    "color_mode": "xy",
    "color_options": null,
    "color_temp": 454,
    "effect": null,
    "identify": null,
    "last_seen": "2025-02-12T06:21:51.707Z",
    "linkquality": 54,
    "power_on_behavior": null,
    "state": "ON",
    "update": {
        "installed_version": 587757105,
        "latest_version": 587757105,
        "state": "idle"
    },
    "update_available": false
}
```

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

zstack

### Setup

home assistant and zigbee2mqtt in k8s

jamesonuk, 2025-03-03:
Just started seeing this after upgrade to 2.3.1 too.   (It is my bathroom lights too...)
Mine are Ikea [LED1650R5](https://www.zigbee2mqtt.io/devices/LED1650R5.html#ikea-led1650r5)
These lights don't support colours at all and what is odd is that I have a few others of these bulbs but they don't show a `color_mode` at all.

These may have been one of the first lights I installed so wondering whether this is just that they had `colour_mode: xy` set way back and this has just persisted through retained messages???

jamesonuk, 2025-03-03:
The only way I was able to get `color_mode` to go was to delete the devices from Z2M and re-pair them.   It then gives me my z2m topic payload of
```json
{
    "brightness": 254,
    "effect": null,
    "identify": null,
    "last_seen": "2025-03-03T14:50:05.808Z",
    "level_config": {
        "current_level_startup": "previous",
        "execute_if_off": true
    },
    "linkquality": 255,
    "power_on_behavior": "previous",
    "state": "ON",
    "update": {
        "installed_version": 587810353,
        "latest_version": 587810353,
        "state": "idle"
    }
}
```

RedNo7, 2025-03-11:
I now also this now using
```yaml
tap_action:
  action: light.toggle
```
(my lights are via Z2M).

I upgraded to 2025/3/1 and it still happens.

runningman84, 2025-03-12:
@Koenkk if a delete and join fix this issue, can't you provide a fix for existing devices as well?

alexsaas, 2025-03-13:
I've also been getting this error message since z2m 2.3.1. But only with lights where you can only adjust the brightness, meaning you can't adjust the color temperature or color at all.

RedNo7, 2025-03-13:
Will give this a shot later - thanks for sharing!

alexsaas, 2025-03-15:
Okay? That's strange. After a full system restart, the entries are back.

@jamesonuk Is there perhaps another MQTT topic that needs to be deleted?

jamesonuk, 2025-03-15:
> Okay? That's strange. After the third full system restart a few days later, the entries are back.
> 
> [@jamesonuk](https://github.com/jamesonuk) Is there perhaps another MQTT topic that needs to be deleted?

It depends what you are actually seeing.   The issue here was that at some point in the past devices / groups got a colour mode which then got stored in `states.json` so were being passed to HA when there should have been nothing.
Check what is in both the state (`zigbee2mqtt/<device friendly name>` and discovery (`homeassistant/light/0x<deviceid>/light/config`) messages in MQTT and also in `states.json`
and delete / edit as appropriate (I would just remove the colour mode entry in the states file rather than deleting the whole thing though

alexsaas, 2025-03-15:
@jamesonuk Thank you very much for your quick response!

I think I've found the root cause: Zigbee2MQTT scenes of the lights that were saved with the entry in the status a long time ago.

These scenes were triggered last night for the first time since the change to `state.json`. I haven't opened HA since then. I have an automatic restart at night, so I saw the messages in the log this morning.

So I have now deleted these scenes and reworked the points I listed in the above post. Then I reset the scenes in z2m and tested them, performed restarts and triggered the automations that activates the scenes.

It looks good so far, and I hope it stays that way. Otherwise, I'll have to dig deeper and post an update here.

Thanks again for your advice and quick support.

closed 2025-05-23 by github-actions[bot]