Koenkk/zigbee2mqtt #26329: Schneider Electric A9MEM1570 throws Error messages

SupperCraft5000, 2025-02-12:
### What happened?

Added new Device GreenPower Schneider Electric A9MEM1570 then i got lots of the following Error message 

`z2m: Exception while calling fromZigbee converter: CommandResponse 0x00000000e20412c5/242 greenPower.response({"options":0,"tempMaster":56631,"tempMasterTx":0,"srcID":3791917765,"gpdCmd":254,"gpdPayload":{"commandID":254,"buffer":{"type":"Buffer","data":[0]}}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":242,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL BROADCAST destination=65533] Failed to send with status=BUSY.)}`

### What did you expect to happen?

No Errors in Log

### How to reproduce it (minimal and precise)

Nothing, every few Minutes the Error appears 

### Zigbee2MQTT version

1.42.0 commit: bc6eccf

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Debian on a VM in Proxmox 

### Debug log

`z2m: Exception while calling fromZigbee converter: CommandResponse 0x00000000e20412c5/242 greenPower.response({"options":0,"tempMaster":56631,"tempMasterTx":0,"srcID":3791917765,"gpdCmd":254,"gpdPayload":{"commandID":254,"buffer":{"type":"Buffer","data":[0]}}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":242,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL BROADCAST destination=65533] Failed to send with status=BUSY.)}`

[state.json.zip](https://github.com/user-attachments/files/18771082/state.json.zip)

[image]

paulpalle, 2025-03-16:
I have same problem GreenPower Schneider Electric A9MEM1570

### Zigbee2MQTT version
2.1.3 commit: unknown

### Adapter firmware version
7.4.5 [GA]

### Adapter
EmberZNet

### Setup
Home Assistant Yellow (zigbee2mqtt on ZBDongle-E)

SupperCraft5000, 2025-05-15:
The Error message still exists even after Updating

Zigbee2MQTT version
[2.3.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.3.0) commit: [9d1427c1](https://github.com/Koenkk/zigbee2mqtt/commit/9d1427c1)

Adapter firmware version

7.4.4 [GA]
Adapter

EmberZNet
Setup

Debian on a VM in Proxmox

paulpalle, 2025-07-16:
After i updated my ZBDongl-E with the SONOFF Dongle flasher the error message went away