Koenkk/zigbee2mqtt #26338: Zemismart SPM01 (SPM01-D2TZ) power meter not reporting energy

WazoAkaRapace, 2025-02-13:
### What happened?

Hello,
I recently installed a Zemismart SPM01 and it's reporting the voltage correctly, but not the energy.

[image]

### What did you expect to happen?

I expect it to return the energy consumption.

### How to reproduce it (minimal and precise)

Try one of those device (I think...)

Device type: Router
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE284_iwn0gpzz
Description: Smart energy monitor for 1P+N system
Support status: Supported
Manufacturer: [Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Model: [SPM01V2.5](https://www.zigbee2mqtt.io/devices/SPM01V2.5.html#tuya-spm01v2.5)

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

Skyconnect 

### Setup

Add-on on HomeAssistant OS (Pi 4)

cervcz, 2025-03-04:
I have the same problem. Voltage reports correctly, but power, energy, current, and produced_power remain 0.

**Device**: Zemismart SPM01V2.5
**Zigbee Model:** TS0601
**Manufacturer:** _TZE284_iwn0gpzz (Tuya)
### Issue:
Voltage reports correctly, but power, energy, current, and produced_power remain 0.
MQTT messages confirm no data is sent for these attributes.

### **Using** 
Zigbee2MQTT 2.1.3, SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P (zStack3x0 20240710), latest HAOS in VM

cervcz, 2025-03-05:
Issue Resolved (for me) – No Problem with Zigbee2MQTT

The issue was caused by an incorrect installation of the sensor. Only the Live (L) cable should pass through the sensor—if both Live and Neutral are passed through, the readings cancel out, resulting in 0W for power, energy, and current.

After correcting the installation, the device reports values correctly in Zigbee2MQTT. No changes to the converter are needed.

Thanks for the support! 👍

PiotrRaszkowski, 2025-05-08:
Maybe related? https://github.com/Koenkk/zigbee2mqtt/issues/26905