Koenkk/zigbee2mqtt #26342: Zigbee2Mqtt 0x11: BUFFER_FULL

Capinss, 2025-02-13:
### What happened?

I have two instances of zigbee2mqtt running at the same time - They are on the same network, but physically the coordinators are like 300m away from each other - two separate buildings.
One (the first instance configured) is showing error and not allowing to connect new devices, here is the error below:

[image]
The second coordinator:
[image]

This problem is with the other coordinator (the first one set up) here is the info on that:

[image]

And attaching a log also.
Could it be that the two coordinators are somehow conflicting and thus the problem with the  "BUFFER-FULL" error?

[1st configured 45df7312_zigbee2mqtt_2025-02-13T07-11-59.928Z.log](https://github.com/user-attachments/files/18783056/1st.configured.45df7312_zigbee2mqtt_2025-02-13T07-11-59.928Z.log)
[2nt configured 9336c2b0_zigbee2mqtt_2025-02-13T07-05-07.969Z.log](https://github.com/user-attachments/files/18783057/2nt.configured.9336c2b0_zigbee2mqtt_2025-02-13T07-05-07.969Z.log)

I have reached out to the customer support of SLZB-06, they suggested to rollback to the firmware version 20221226 of the SLZB-06, which I did and that resulted in both coordinators showing the same error when permitting new devices to join. 
After that I was asked to create a ticket here.

### Zigbee2MQTT version

2.1.1.

### Adapter firmware version

20221226

### Adapter

Coordinator type zStack3x0 Coordinator revision 20221226. SLZB-06

### Setup

Home Assistant on VMware Workstation PRO 17

### Debug log

info 2025-02-13 13:35:52z2m: Zigbee: allowing new devices to join.
error 2025-02-13 13:35:52z2m: Request 'zigbee2mqtt_c11/bridge/request/permit_join' failed with error: '--> 'SREQ: ZDO - PERMIT_JOINING_REQUEST - 0ffcfffe01' failed with status '(0xc2: NWK_INVALID_REQUEST)''
info 2025-02-13 13:35:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_c11/bridge/response/permit_join', payload '{"data":{},"error":"--> 'SREQ: ZDO - PERMIT_JOINING_REQUEST - 0ffcfffe01' failed with status '(0xc2: NWK_INVALID_REQUEST)'","status":"error","transaction":"q6l7m-1"}'

info 2025-02-13 13:36:31z2m: Zigbee: allowing new devices to join.
error 2025-02-13 13:36:31z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: '--> 'SREQ: ZDO - PERMIT_JOINING_REQUEST - 0ffcfffe01' failed with status '(0xc2: NWK_INVALID_REQUEST)''
info 2025-02-13 13:36:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"--> 'SREQ: ZDO - PERMIT_JOINING_REQUEST - 0ffcfffe01' failed with status '(0xc2: NWK_INVALID_REQUEST)'","status":"error","transaction":"wqhum-1"}'

HenrikHills, 2025-02-14:
I was advised to downgrade FW to an earlier version, seemed to help with BUFFER_FULL error but its starting to act strange again with timeout-errors, that might not be related though 

https://github.com/Koenkk/zigbee2mqtt/issues/26271#issuecomment-2649135959

TheNutHouse84, 2025-04-03:
I'm getting the same error in a single coordinator system with Sonoff USB Dongle P. Newer Z2M version and newer firmware on dongle. 

Running on bare metal Linux Mint 22.1 
node --version  v20.19.0
npm --version 10.8.2

Zigbee2MQTT version
[2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3) commit: [ba337bd](https://github.com/Koenkk/zigbee2mqtt/commit/ba337bd)
Coordinator type
zStack3x0
Coordinator revision
20240710
Coordinator IEEE Address
0x00124b0024c27437
Frontend version
0.9.4
zigbee-herdsman-converters version
23.2.0
zigbee-herdsman version
3.2.7

TheNutHouse84, 2025-04-05:
I did a fresh install of Linux Mint OS, NodeJS, Z2M (2.0), NodeRed and flashed the dongle with firmware 20240710 on Jan 4 2025. Paired my devices. 

I had no errors at all. It wasn't until I updated to one of the 2.1 releases that I encountered the buffer_full errors.

In my case I don't believe the buffer_full error it related to the firmware, unless newer versions Z2M communicates with the dongle differently.

closed 2025-06-12 by github-actions[bot]