Koenkk/zigbee2mqtt #26346: IKEA LED1923R5/LED2110R3 OTA and/or model identification issue

HNKNTA, 2025-02-13:
### What happened?

I have 4 IKEA bulbs which are identified as IKEA LED1923R5. 2 of them are pretty old, and 2 I bought recently. 
New ones have this identifier on the box: LED2110R3. Here is what I see in the Web UI OTA:

[image]

The fresh bulbs have 1.0.38 on board and the system shows OTA available. The older ones show no OTA, and they're on 1.0.021, which seems older and with a weird SemVer patch number with leading zero. 

So something is not OK ether with OTA or bulb identification (maybe that's different bulbs and should have own ID in Z2M). 

Another difference is that older bulbs cannot do effects, whether new ones can do it.

### What did you expect to happen?

Different model IDs in Z2M. Or fresher firmware for those on 1.0.021

### How to reproduce it (minimal and precise)

Plug and play

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

Rpi4 with docker container

HNKNTA, 2025-02-13:
Probably mentioned https://github.com/Koenkk/zigbee2mqtt/issues/22186

TheFou, 2025-03-24:
Hi,
I also have a LED2110R3 GU10 bulb detected as LED1923R5.
It works fine, but I wonder if the identification error could lead to some issue, like firmware detection / upgrade for instance ?
Can we help with something to correct this ?

HNKNTA, 2025-05-28:
Still actual on 2.3.0

closed 2025-08-04 by github-actions[bot]