Koenkk/zigbee2mqtt #26347: External converters can not be loaded anymore

han-alink, 2025-02-13:
### What happened?

Updating from 2.0.0-2 to a newer version creates the following error. I have looked through the release data, but can not find something that has changed to result in the following error:

[2025-02-13 20:58:05] error: 	z2m: Failed to load external converter 'ha_devices.js'
[2025-02-13 20:58:08] error: 	z2m: Check the code for syntax error and make sure it is up to date with the current Zigbee2MQTT version.
[2025-02-13 20:58:08] error: 	z2m: External converters are not meant for long term usage, but for local testing after which a pull request should be created to add out-of-the-box support for the device
[2025-02-13 20:58:14] error: 	z2m: Failed to call 'ExternalConverters' 'start' (TypeError: Cannot read properties of undefined (reading 'options')
    at prepareDefinition (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:367:23)
    at addDefinition (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:426:18)
    at ExternalConverters.loadJS (/app/lib/extension/externalConverters.ts:49:30)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:24)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))

### What did you expect to happen?

Not this error ....

### How to reproduce it (minimal and precise)

Tried updating 2 times. Same result. Back to 2.0.0-2 everything is working again.

### Zigbee2MQTT version

2.1.0-1 and/or 2.1.1-1

### Adapter firmware version

?

### Adapter

Slaesh's

### Setup

RPI-4 Add on

chris-1243, 2025-02-14:
You have a syntax error. If you share your converter, we may see what is wrong with it.

han-alink, 2025-02-14:
Hello Chris,
Here is the file with external converters.
Thanks for your help,
Han

han-alink, 2025-02-15:
Did you take a look?
If the directory external_converters also contains some other files (not with a .js extension), could that be the cause of the erorors?

chris-1243, 2025-02-16:
It is difficult to help as you did not provide any code unfortunately

han-alink, 2025-02-16:
Sorry,
.js extention not allowed. Renamed to .txt.

[ha_devices.txt](https://github.com/user-attachments/files/18814816/ha_devices.txt)

Thanks for your help,
Han

chris-1243, 2025-02-17:
I try to have a look but it seems your file has a lot of errors.

You may try this file corrected. To be honest, it will be too time consuming to understand your converter for me and find the issues.

[ha_device.txt](https://github.com/user-attachments/files/18822323/ha_device.txt)

han-alink, 2025-02-18:
I pinned the problem that withheld the zigbee2mqtt from starting  (or at least generated an error), to the following device:
When I add the illuminance "fromZigbee, exposes, endpoint and reporting" the error is created.  

    {
        //combined sensor devices
        //Temperature Sensor: Id: 0x0302
        //Humidity Sensor: Id: 0x0308
        //Pressure Sensor: Id: 0x0305
        //Light Sensor: Id: 0x0106
        zigbeeModel: ['MultiSensTHPL'],
        model: 'TempHumPresLightDevice',
        vendor: 'Alink',
        description: 'Temperature Humidity, Pressure and Light device',
        fromZigbee: [fz.temperature, fz.humidity, fz.battery, fz.pressure], // fz.illuminance, 
        toZigbee: [],
        exposes: [e.temperature(), e.humidity(), e.pressure(), e.battery(), e.battery_voltage()], //, e.illuminance()
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const firstEndpoint = device.getEndpoint(1);
            await reporting.bind(firstEndpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            const overrides1 = {min: 300, max: 60000, change: 10};
            await reporting.temperature(firstEndpoint, overrides1);
            await reporting.batteryVoltage(firstEndpoint);
            const secondEndpoint = device.getEndpoint(2);
            await reporting.bind(secondEndpoint, coordinatorEndpoint, ['msRelativeHumidity']);
            const overrides2 = {min: 300, max: 60000, change: 10};
            await reporting.humidity(secondEndpoint, overrides2);
            const thirdEndpoint = device.getEndpoint(3);
            await reporting.bind(thirdEndpoint, coordinatorEndpoint, ['msPressureMeasurement']);
            const overrides3 = {min: 300, max: 60000, change: 10};
            await reporting.pressure(thirdEndpoint, overrides3);
//            const fourthEndpoint = device.getEndpoint(4);
//            await reporting.bind(fourthEndpoint, coordinatorEndpoint, ['msIlluminanceMeasurement']);
//            const overrides4 = {min: 300, max: 60000, change: 50};
//            await reporting.illuminance(fourthEndpoint, overrides4);
        },
    },

Nerivec, 2025-03-20:
`fz.illuminance` no longer exists, it was replaced in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8592
You can use the modern extend `illuminance` (which is already imported in your js file).
```js
extend: [illuminance()],
```

See https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/modernExtend.ts#L652-L690 for more details on current implementation of the modern extend.

closed 2025-03-20 by Nerivec