Koenkk/zigbee2mqtt #26364: SPM01 does not expose configuration option

vilaemail, 2025-02-14:
### What happened?

My SPM01 device does not show configuration option for how often to report measurements.

### What did you expect to happen?

Ability to change configuration option.

### How to reproduce it (minimal and precise)

This is doable from Tuya application. I have captured a sniff where I set value to 10 and then to 100 (see below). I can see 0a and 64 hex values (10 and 100) in the message.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

UZG-01

### Setup

Add-on on Home Assistant OS on RPi4

### Debug log

Setting to 10:
```
No.     Time           Source                Destination           Protocol Length Sequence Number Cluster    Info
  26623 986.242679     0x0000                0x9462                ZigBee HA 78     15              0xef00     Unknown Command: 0x00, Seq: 73

Frame 26623: 78 bytes on wire (624 bits), 78 bytes captured (624 bits)
IEEE 802.15.4 Data, Src: 0x0000, Dst: 0x9462
ZigBee Network Layer Data, Dst: 0x9462, Src: 0x0000
    Frame Control Field: 0x0208, Frame Type: Data, Discover Route: Suppress, Security Data
    Destination: 0x9462
    Source: 0x0000
    Radius: 30
    Sequence Number: 15
    [Extended Source: SiliconLabor_ff:fe:35:2d:52 (84:ba:20:ff:fe:35:2d:52)]
    [Origin: 138]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 184
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 73
    Command: Unknown (0x00)
Data (30 bytes)

0000  05 c3 12 00 00 18 01 01 00 3c 03 01 00 fd 04 00   .........<......
0010  00 b4 07 01 00 00 08 01 00 0a 09 00 00 00         ..............
    Data: 05c3120000180101003c030100fd040000b4070100000801000a09000000
    [Length: 30]

No.     Time           Source                Destination           Protocol Length Sequence Number Cluster    Info
  26625 986.247863     0x9462                0x0000                ZigBee   45     28              0xef00     APS: Ack, Dst Endpt: 1, Src Endpt: 1

Frame 26625: 45 bytes on wire (360 bits), 45 bytes captured (360 bits)
IEEE 802.15.4 Data, Src: 0x9462, Dst: 0x0000
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x9462
[... 169 lines cut ...]
IEEE 802.15.4 Data, Src: 0x0000, Dst: 0x9462
ZigBee Network Layer Data, Dst: 0x9462, Src: 0x0000
    Frame Control Field: 0x0208, Frame Type: Data, Discover Route: Suppress, Security Data
    Destination: 0x9462
    Source: 0x0000
    Radius: 30
    Sequence Number: 235
    [Extended Source: SiliconLabor_ff:fe:35:2d:52 (84:ba:20:ff:fe:35:2d:52)]
    [Origin: 138]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 35
ZigBee Cluster Library Frame, Command: Default Response, Seq: 110
    Frame Control Field: Profile-wide (0x00)
    Sequence Number: 110
    Command: Default Response (0x0b)
    Response to Command: 0x01
    Status: Success (0x00)

```

vilaemail, 2025-02-14:
I am willing to make the change myself, but need pointers how best to support this for Tuya device. I have read the docs on adding Tuya devices and datapoints, but am not sure if this fits the definition.

Koenkk, 2025-02-14:
@vilaemail this goes via datapoints, see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html how to find them.

vilaemail, 2025-02-17:
Is it possible to get them from the sniffed data above? Also, this is more of a "configuration packet sent to the device" than a "device reporting some value". Reason for asking is that device is not easily accessible to me, and I'd need to repair it to Tuya gateway to use the guide...

Koenkk, 2025-02-17:
Could you share the sniff file + the network key?

vilaemail, 2025-02-17:
I have quoted in original submission relevant messages from the sniff (setting value to 10 and then 100). If you need additional messages from the sniff I can upload it.

vilaemail, 2025-02-17:
I have also tried using [this](https://www.elektroda.com/rtvforum/topic3970199.html#20528459) based on [this thread](https://www.elektroda.com/rtvforum/topic4052633.html), by (rather crudely) pasting the data bytes from the sniff and it's not working. It might be that the Tuya UART protocol is additionally packed in the Zigbee data packet, but I am just shooting in the dark here.

Koenkk, 2025-02-17:
I'm not sure if it is the good nwk key, messages stay encrypted:

[image]

vilaemail, 2025-02-17:
Yes, but the relevant messages (26623, 26625, 26627, 26629 and 29075->29200) from device 0x9462 are not encrypted. Can you try filter `_ws.col.def_src == "0x9462" || _ws.col.def_dst == "0x9462"`)? I am not sure what happened there, could it be that devices exchanged some type of "session" key that I didn't capture.

Koenkk, 2025-02-18:
It seems to be datapoint 18 with data type being 0 (raw), something like this should work:

```js
[
    18,
    'report_config',
    {
        to: (v: number) => {
            return Buffer.from([
                // Unknown what these bytes mean
                0x01, 0x01, 0x00, 0x3c, 0x03, 0x01, 0x00, 0xfd, 0x04, 0x00, 0x00, 0xb4, 0x07, 0x01, 0x00, 0x00, 0x08, 0x01, 0x00,
                // This seems to be the report interval byte
                0x64,
                // Unknown what these bytes mean
                0x09, 0x00, 0x00, 0x00,
            ]);
        },
    },
],
```

vilaemail, 2025-02-19:
Thank you! I will play around with this, validate it works on multiple devices, update it if necessary and send a PR.

Koenkk, 2025-02-19:
Try returning an array instead of buffer

vilaemail, 2025-02-19:
It works with an array, I will send PR like that, let me know if I should revert it back to buffer.