Koenkk/zigbee2mqtt #26365: [New device support]:  Hue Centura White and Color (with bluetooth) new model

puttemanp, 2025-02-14:
### Link

https://www.philips-hue.com/nl-be/p/hue-white-and-color-ambiance-centura-inbouwspot/8720169319035

### Database entry

{"id":124,"type":"Router","ieeeAddr":"0x001788010e5c879a","nwkAddr":49780,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003809601","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003809601","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20240902","swBuildId":"1.122.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20240902","swBuildId":"1.122.2","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1739558826441}

### Zigbee2MQTT version

2.11.01

### Comments

Device paired correctly, but as an 'unsupported device'.
It's identical to the Hue Centura White and Color spot, but exposes 'Signify Netherlands B.V.' as vendor instead of Philips

### External definition

```shell
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003809601'],
    model: '929003809601',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.philipsLight({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Device seems fully functional

Koenkk, 2025-02-15:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `929003809601.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2025-02-15 by Koenkk

puttemanp, 2025-02-16:
Picture added.

NickJWStone, 2025-03-04:
It seems Signify made a few different model numbers for effectively the same device. I've bought [this set](https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-3-pack-centura-inbouwspot/8720169318977), which paires as unsupported devices with different modelnames per lamp in the pack: 929003809501_01, 929003809501_02 and 929003809501_03. @Koenkk Let me know if you want me to open a new issue to add these!

```
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003809501_01'],
    model: '929003809501_01',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;

```
and

const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003809501_02'],
    model: '929003809501_02',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;

as well as

const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003809501_03'],
    model: '929003809501_03',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;

Koenkk, 2025-03-04:
@NickJWStone added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `929003809501.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

NickJWStone, 2025-03-05:
@Koenkk picture submitted, thanks!