Koenkk/zigbee2mqtt #26370: Installation guide instructs to run `npm install -g pnpm` without sudo

hmeine, 2025-02-14:
### What happened?

On a pristine Raspbian installation, I have been following the installation instructions at https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing and found it surprising that I should run `npm install -g pnpm` without `sudo`. When doing so, I get the expected error from npm that the target is not writeable for the current user.  

### What did you expect to happen?

Since the surrounding commands are run through `sudo`, I think the guide should just be adapted. Running `sudo npm install -g pnpm` installed pnpm properly, as expected.

PS: Maybe there should be an issue template for documentation as well; many of the below fields do not really apply in my case.

### Zigbee2MQTT version

2.1.1 commit: 1accb8b6

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Manual installation on Raspbian (Pi version B)

schauveau, 2025-03-03:
Did you manage to finish the installation? I can install pnpm with  `sudo npm install -g pnpm` but later I get the following error when running `pnpm i --frozen-lockfile`

``` 
ERROR  Failed to switch pnpm to v10.4.1. Looks like pnpm CLI is missing at "/home/chauveau/.local/share/pnpm/.tools/pnpm/10.4.1/bin" or is incorrect
spawnSync /home/chauveau/.local/share/pnpm/.tools/pnpm/10.4.1/bin/pnpm ENOENT
```

schauveau, 2025-03-03:
I eventually figured out the problem. npm is installing a newer version 10.5 while pnpm version 10.4.1 is hardcoded in package.json 

https://github.com/Koenkk/zigbee2mqtt/blob/ba337bd329aeb4ca17735c0cf09b31293c8cff06/package.json#L6  

So the proper command should be `sudo npm install -g pnpm@10.4.1`

closed 2025-05-10 by github-actions[bot]

hmeine, 2025-08-04:
@schauveau But do you agree that the documentation is missing a `sudo` for the pnpm installation?