Koenkk/zigbee2mqtt #26371: [New device support]:

uncle-wong, 2025-02-15:
### Link

null

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0x0c2a6ffffeda5798","nwkAddr":46943,"manufId":4098,"manufName":"_TZE284_ibnz7orz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":" xC/i xC/i\tyC/i\u001dyC/i1yC/iEyC/iXyC/ilyC/i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_ibnz7orz","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1401274594},"lastSeen":1739607186421}

### Zigbee2MQTT version

2.1.1-1

### Comments

dpid {"5":"场景1","6":"场景2","7":"场景3","8":"场景4","17":"标准场景","24":"开关1","25":"开关2","26":"开关3","27":"开关4","36":"背光灯开关","38":"上电状态设置","101":"背光颜色","106":"感应延时","109":"开关1继电器绑定","110":"开关2继电器绑定","111":"开关3继电器绑定","112":"开关4继电器绑定","117":"预留透传通道","126":"开关1名称","127":"开关2名称","128":"开关3名称 ","129":"开关4名称 ","130":"场景1名称","131":"场景2名称","132":"场景3名称","133":"场景4名称","138":"场景ID","144":"设备类型","152":"L1继电器控制","153":"L2继电器控制","154":"L3继电器控制","155":"L4继电器控制","157":"背光灯开关设置","158":"开关模式切换"}

{
  "result": {
    "properties": [
      {
        "code": "scene_5",
        "custom_name": "",
        "dp_id": 5,
        "time": 1739456098663,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_6",
        "custom_name": "",
        "dp_id": 6,
        "time": 1739456098663,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_7",
        "custom_name": "",
        "dp_id": 7,
        "time": 1739456098663,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_8",
        "custom_name": "",
        "dp_id": 8,
        "time": 1739456098663,
        "type": "enum",
        "value": "scene"
      },
      {
        "code": "scene_id_group_id",
        "custom_name": "",
        "dp_id": 17,
        "time": 1739456098663,
        "type": "raw"
      },
      {
        "code": "switch_1",
        "custom_name": "客厅",
        "dp_id": 24,
        "time": 1739458714162,
        "type": "bool",
        "value": true
      },
      {
        "code": "switch_2",
        "custom_name": "",
        "dp_id": 25,
[... 303 lines cut ...]
        ],
    },

};

module.exports = definition;

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_ibnz7orz',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I have a 4-key switch for doodling. Double-clicking any key can switch to a 4-key scene. The panel also allows setting names, which looks like base64, but the decoded result doesn't match. The functions I want to achieve are controllable switches, controllable scenes, and customizable names. Please help me.

uncle-wong, 2025-02-16:
can anyone help me？

devkaiwang, 2025-03-02:
I wonder if scene is supported by this project?
https://github.com/Koenkk/zigbee2mqtt/issues/18905
https://github.com/Koenkk/zigbee2mqtt/issues/22176
https://github.com/Koenkk/zigbee2mqtt/issues/17827
https://github.com/Koenkk/zigbee2mqtt/issues/26296

I guess there is more...

warrenarnold, 2025-03-17:
I was also able to download the 'Things Data Model' from Tuya for the switch

```
{
  "result": {
    "model": {
      "modelId": "e1krju5w",
      "services": [
        {
          "actions": [],
          "code": "",
          "description": "",
          "events": [],
          "name": "",
          "properties": [
            {
              "abilityId": 1,
              "accessMode": "rw",
              "code": "switch_1",
              "description": "",
              "extensions": {
                "iconName": "icon-power",
                "attribute": "1045"
              },
              "name": "开关 1",
              "typeSpec": {
                "type": "bool"
              }
            },
            {
              "abilityId": 2,
              "accessMode": "rw",
              "code": "switch_2",
              "description": "",
              "extensions": {
                "iconName": "icon-power",
                "attribute": "1045"
              },
              "name": "开关 2",
              "typeSpec": {
                "type": "bool"
              }
            },
            {
              "abilityId": 3,
              "accessMode": "rw",
              "code": "switch_3",
              "description": "",
              "extensions": {
                "iconName": "icon-power",
                "attribute": "1045"
              },
              "name": "开关 3",
              "typeSpec": {
                "type": "bool"
              }
            },
            {
              "abilityId": 4,
              "accessMode": "rw",
              "code": "switch_4",
              "description": "",
              "extensions": {
                "iconName": "icon-power",
                "attribute": "1157"
              },
              "name": "开关4",
              "typeSpec": {
                "type": "bool"
              }
[... 435 lines cut ...]
              "abilityId": 210,
              "accessMode": "rw",
              "code": "random_timing",
              "description": "",
              "extensions": {
                "iconName": "icon-a_nav_mode",
                "attribute": "1248"
              },
              "name": "随机定时",
              "typeSpec": {
                "type": "raw",
                "maxlen": 255
              }
            }
          ]
        }
      ]
    }
  },
  "success": true,
  "t": 1742171833429,
  "tid": "f82d63ea02c711f08d3cfab3a7102e50"
}

```

uncle-wong, 2025-03-25:
> I was also able to download the 'Things Data Model' from Tuya for the switch
>
> ```
> {
>   "result": {
>     "model": {
>       "modelId": "e1krju5w",
>       "services": [
>         {
>           "actions": [...]

I used cursor to write one based on the data points and your examples. Now, the scene and switch toggle buttons work, but the backlight switch, switch status feedback, and scene switch/switch/scene name settings not work。英文真费劲。可以用中文

uncle-wong, 2025-05-06:
没有人可以帮助我吗？

closed 2025-07-13 by github-actions[bot]