Koenkk/zigbee2mqtt #26374: Duplicated zigbee/mqtt messages when pressing any button

xammmue, 2025-02-15:
### What happened?

I noticed that for some reason my devices seems to be reporting duplicated updates, when for example pressing a button. At first I only noticed this for my Hue Smart Button, but then also Hue dimmer switches as well as Hue Wall switches. I was also able to reproduce the problem using an Aqara Temperature Sensor when pressing that button.

According to the logs, it seems like z2m is receiving duplicated zigbee messages, but I don't really understand how that could happen.

I only noticed the issue in the past weeks. I am not sure whether it happened immediatly after I switched my RaspberryPi installation some months ago.
The stick has been working for a few years now.

The double messages seem to only occurre some time after restarting.

### What did you expect to happen?

Only a single message being sent for every button press

### How to reproduce it (minimal and precise)

Not sure what to say, I can reproduce it reliably by pressing any button on any of my devices, but I am pretty sure thats some specific problem with my installation

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Setup

Docker container on Rasberry Pi 5 running OpenWRT

### Debug log

[log.log](https://github.com/user-attachments/files/18810036/log.log)

Some hightlights from the logs combined with the mqtt message that have been sent:

Hue wall switch Flip 1
```
[2025-02-15 11:44:04] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=143, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":9,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-15 11:44:04] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=143, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":9,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-15 11:44:04] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=143, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":10,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":2,"unknown2":0,"time":1},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-15 11:44:04] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=143, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":10,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":2,"unknown2":0,"time":1},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}

{"action":"right_press","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:04+01:00","linkquality":143}
{"action":"right_press","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:04+01:00","linkquality":143}
{"action":"right_press_release","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:04+01:00","linkquality":143}
{"action":"right_press_release","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:04+01:00","linkquality":143}
```

Hue Wall Switch Flip 2
```
[2025-02-15 11:44:09] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=143, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":11,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-15 11:44:09] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=143, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":11,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-15 11:44:09] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=135, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":12,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":2,"unknown2":0,"time":1},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-15 11:44:09] debug:    zh:controller: Received payload: clusterID=64512, address=19099, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=135, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":12,"commandIdentifier":0},"payload":{"button":2,"unknown1":3145728,"type":2,"unknown2":0,"time":1},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}

{"action":"right_press","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:09+01:00","linkquality":143}
{"action":"right_press","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:09+01:00","linkquality":143}
{"action":"right_press_release","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:09+01:00","linkquality":135}
{"action":"right_press_release","battery":100,"device_mode":"dual_rocker","last_seen":"2025-02-15T11:44:09+01:00","linkquality":135}
```

Hue Dimmer Switch Press 1
```
[2025-02-15 11:46:19] debug:    zh:controller: Received payload: clusterID=64512, address=26863, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=127, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":253,"commandIdentifi [... cut ...]
{"battery":100,"humidity":33.44,"last_seen":"2025-02-15T11:54:00+01:00","linkquality":143,"power_outage_count":13536,"pressure":953.8,"temperature":25.85,"voltage":3015}
{"battery":100,"humidity":33.44,"last_seen":"2025-02-15T11:54:00+01:00","linkquality":143,"power_outage_count":13536,"pressure":953.8,"temperature":25.85,"voltage":3015}
{"battery":100,"humidity":33.44,"last_seen":"2025-02-15T11:54:00+01:00","linkquality":143,"power_outage_count":13536,"pressure":953.8,"temperature":25.85,"voltage":3015}
```

Koenkk, 2025-02-17:
fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

xammmue, 2025-02-17:
Thanks a lot!

I updated to the latest dev version and will look into the behavior now. Can probably only tell after a few hours though

xammmue, 2025-02-19:
I've looked at the behavior for some more time now but it looks to me like there are still some small issues.

The duplicated action messages are gone for my dimmer and wall switches now, but I can still obserbe that behavior for my Hue Smart Button.

Logs for what's happening when pressing the smart button once:
```
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: query aps data indication
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 214
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 214 srcAddr: 0xd8f3 destAddr: 0x0 profile id: 0x104 cluster id: 0xfc00 lqi: 87
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: response payload: 1d0b10a2000100003000210000
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-19 12:10:06] debug: 	zh:controller: Received payload: clusterID=64512, address=55539, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=87, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":162,"commandIdentifier":0},"payload":{"button":1,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 214 srcAddr: 0xd8f3 destAddr: 0x0 profile id: 0x104 cluster id: 0xfc00 lqi: 87
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: response payload: 1d0b10a2000100003000210000
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-19 12:10:06] debug: 	zh:controller: Received payload: clusterID=64512, address=55539, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=87, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":162,"commandIdentifier":0},"payload":{"button":1,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2025-02-19 12:10:06] debug: 	z2m: Received Zigbee message from 'Hue Smart Button', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
[2025-02-19 12:10:06] debug: 	z2m: Received Zigbee message from 'Hue Smart Button', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
[2025-02-19 12:10:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Smart Button', payload '{"action":"press","battery":100,"last_seen":"2025-02-19T12:10:06+01:00","linkquality":87,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"},"update_available":false}'
[2025-02-19 12:10:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Smart Button', payload '{"action":"press","battery":100,"last_seen":"2025-02-19T12:10:06+01:00","linkquality":87,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"},"update_available":false}'
[2025-02-19 12:10:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Smart Button/action', payload 'press'
[2025-02-19 12:10:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Smart Button/action', payload 'press'
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: query aps data indication
[2025-02-19 12:10:06] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 215
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 215 srcAddr: 0xd8f3 destAddr: 0x0 profile id: 0x104 cluster id: 0xfc00 lqi: 79
[2025-02-19 12:10:06] debug: 	zh:deconz:frameparser: response payload: 1d0b10a3000100003002210100
[... cut ...]
{"battery":100,"last_seen":"2025-02-19T12:04:18+01:00","linkquality":103,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"},"update_available":false}

{"battery":100,"last_seen":"2025-02-19T12:04:19+01:00","linkquality":111,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"},"update_available":false}

{"battery":100,"last_seen":"2025-02-19T12:04:19+01:00","linkquality":111,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"},"update_available":false}
```
Not sure whether this is intentional or not.

xammmue, 2025-02-26:
Based on the logs my assumption is that this is not necessarily caused by the z2m implementation but some quirk either in my zigbee network or the host I am using.

I ordered a ZigStar UZG-01 now to see whether the problem changes / gets resolved. The possibility of an ethernet connection should also yield some more independence to run z2m on a host without USB connections

Will report back when I've put the adapter into use

xammmue, 2025-04-08:
I faced this issue again some days ago with a new installation, new host and new adapter.
It started after connecting a NOUS smart plug, or actually the 3rd identical NOUS smart plug. It also stopped again after I removed that smart plug from the network and restarted.

So it seems to me like this behavior is caused by the spamming of those NOUS plugs in some way.
While still seeming like a bug, I'd be fine with that information now and will just refrain from adding any more of those plugs in the future.

I'll close this issue if nobody has another opinion on this

closed 2025-04-14 by xammmue