Koenkk/zigbee2mqtt #26375: No paired device respond anymore - Error: ZCL command genOnOff.on

1mfaasj, 2025-02-15:
### What happened?

Every paired zigbee device in zigbee2mqtt does not respond anymore when I try to perform an action like enable a light. 
the error that keeps repeating for every device:

`error 2025-02-15 13:16:07z2m: Publish 'set' 'state' to 'HalBeganegrondLamp' failed: 'Error: ZCL command 0x00178801064e1e73/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 31996 - 11 - 31 - 6 - 11 after 10000ms)'`

the strange thing is that the devices still report data to z2m. what I see in the develop console:
`{
    "brightness": 254,
    "color": {
        "h": 32,
        "hue": 32,
        "s": 81,
        "saturation": 81,
        "x": 0.4574,
        "y": 0.41
    },
    "color_mode": "color_temp",
    "color_options": {
        "execute_if_off": false
    },
    "color_temp": 366,
    "color_temp_startup": 366,
    "last_seen": "2025-02-15T13:18:32+01:00",
    "level_config": {
        "current_level_startup": 254
    },
    "linkquality": 47,
    "power_on_behavior": "on",
    "state": "OFF",
    "update": {
        "installed_version": 16787714,
        "latest_version": 16787714,
        "state": "idle"
    },
    "effect": null
}`

What I already tried:

- restart zigbee2mqtt and SLZB-06 (software but also restart in hardware)
- change back to koenkk/zigbee2mqtt:2.1.0
- all zigbee coordinator firmware updates like 20240710, 20240710, 20240315
- change to wifi connection instead of ethernet.

### What did you expect to happen?

That the device (a light for instance) goes on or off when I do this in z2m or home assistant.

### How to reproduce it (minimal and precise)

When I set this to on:

[image]

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

SLZB-06 (CC2652P)

### Setup

docker

serot23, 2025-02-16:
Same/similar problem here. Z2M not stable anymore. I
didn't have any problems besides one entity changing after the big
breaking change update. I got if i remember correctly 2 updates
after that and both went fine, but now of out nowhere everything is
unstable. It went fine for years and even with the welIl known human detectors which flood the network.i now have removed the flooding sensors, deleted the devices which had problems and reconnected them already couple times now, i reconnected them near the coordinator, it is mostly router devices. The first dlevice was an ikea tradfiri 30w led power supply for the kitchen cabinet lights. l even bought a new one because ithought it was defective. The new one doesn't disconnect anymore so maybe that helped s bit, but now powerplugs and switches become inresponsive. I deleted them and reconnected them and stiill after couple hours they disconnect. I get a time out and lqi errors. I don't get any other errors in the log and i am planning to go on a vacation in couple weeks and my z2m is also part of my alarm system so I don't know what to do now besides starting all over again which i barely have time for at the moment. I tried updating my co rdinator with zigstar i have a sonofif p but this didn't help either. I have also another problem which is not z2m related but it began around the same time and it is one of mly wifi camera's randomly disconnecting and rebooting. I didn't change anything in my wifi or network setup.I am on the latest HAOS, everything is up to date.

[image]

brianveltman, 2025-02-16:
Same issue here. Don't have a solution for it yet

1mfaasj, 2025-02-17:
> I don't think there is a solution for it yet. Does someone here knows a stable version of Zigbee2Mqtt and Coordinator firmware that work fine? I want to install some older version because I always read Z2M is a very solid system, but for me is always unstable

either stick to this topic or create your own ticket, don't go offtopic with your own question if you don't have this problem.

JeanMamelle, 2025-02-17:
Same here ! 
Updated to latest version last week.

brianveltman, 2025-02-17:
What zigbee stick are you all using? Maybe it relates to a specific zigbee controller?

arthurperezbessa, 2025-02-17:
> What zigbee stick are you all using? Maybe it relates to a specific zigbee controller?

I use a ZB Dongle Sonoff with 8.0.2 firmware from Nerivec recommended by Koenkk.

My z2m was restarting all the time and this was solved, but I still get GenOnOff errors all the time.

Sometimes devices won't work or take long time to work

serot23, 2025-02-17:
İ have resetted a lot of my routers today and reconnected them again for the so manied time now 5 to 10cm from the coordinator direct to the coordinator. I have a sonoff p dongle 
Zigbee2MQTT version
2.1.1 commit: unknown
Coordinator type
zStack3x0
Coordinator revision
20240710

brianveltman, 2025-02-17:
I have Sonoff too.. Maybe it has something to do with the sonoff firmware.

darexa, 2025-02-18:
Same thing started happening to me yesterday. It's currently working again by restoring earlier zigbee2mqtt backup and after that upgrading again to latest.
Will see if it lasts.

serot23, 2025-02-19:
I had to reset a device for the 3th time yesterday

Zortris, 2025-02-19:
Same problem here. i use the SLZB device. Latest updates installed. i try downgrading now.

ssunny1081, 2025-02-21:
Similar Issue for me all Devices Stop responding, after restart and removal of the Dongle it worked from minutes to a few hours.

Zigbee2MQTT-Version 2.1.1
Coordinator-Typ
zStack3x0
Coordinator-Version
20240710 Sonoff P Dongle

z2m: Publish 'set' 'state' to '0x00xxx' failed: 'Error: ZCL command 0x00xxx genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)

Just downgraded to 2.1.0, hopefully helps

serot23, 2025-02-21:
> Similar Issue for me all Devices Stop responding, after restart and removal of the Dongle it worked from minutes to a few hours.
>
> Zigbee2MQTT-Version 2.1.1 Coordinator-Typ zStack3x0 [...]

Yeah the problem solves for a couple hours but still, reoccurr after that. Unfortunately since the new update i forgot to turn on the backup, so I don't have a backup of the 2.1.0 version

ssunny1081, 2025-02-22:
After ~16h issue occured again. It was running fine for many month before, ever since latest zstack firmware was released middle of last year it was rock solid .

Going down to 2.0.0 now.

ssunny1081, 2025-02-23:
Still going down after some hours. Swaped the USB Cable even though it was running fine for several month but still happens.

Do the Sonoff Dongle P tend to fail / die early? Got mine Nov. 2022 so not very old.

brianveltman, 2025-02-23:
Could be a broken Sonoff stick indeed.
Have seen enough topics with sonoff issues. And now this one.

I'm considering moving to another zigbee stick as well.

serot23, 2025-02-23:
> Could be a broken Sonoff stick indeed. Have seen enough topics with sonoff issues. And now this one.
> 
> I'm considering moving to another zigbee stick as well.

How can it be broken, when some devices work and some don't. A lot of my devices just work without problems, but the other part is just failing. If the stick would be broken, nothing would work right? I have a new stick laying around could test that one but i have a hard head in believing that is the issue.

brianveltman, 2025-02-23:
> How can it be broken, when some devices work and some don't. A lot of my devices just work without problems, but the other part is just failing. If the stick would be broken, nothing would work [...]

I'm not saying this particular Github issue is related to a broken stick.

All I'm saying is that a broken stick could cause unexpected behavior and could be sensitive to interference caused by other electronic devices, as I've seen multiple posts about that.

I don't know a lot about the setup and exact issues @ssunny1081 is having.

For me personally, yet another issue with Sonoff, either firmware or hardware, I've been fighting Sonoff issues since summer last year.
I'm considering swapping it with another manufacturer.

serot23, 2025-02-23:
> I'm not saying this particular Github issue is related to a broken stick.
>
> All I'm saying is that a broken stick could cause unexpected behavior and could be sensitive to interference caused by [...]

I see, do you have a stick in mind which would be better?

brianveltman, 2025-02-23:
Not sure yet. Might try to flash my current sonoff stick to newer firmware first, as mine came pre-flashed.

Will keep you posted :-)

ssunny1081, 2025-02-24:
Going down to 2.0.0 plus swapped USB Cable did not resolve, recovered an old Backup from 1.4 upgraded again to 2.0.0. 
Ordered a new Dongle if this does not fix it it.

Setup: Proxmox 8.3, LXC Bookwork, Docker CE, Z2M, Sonff DongleP (FW 20240710) on USB extention Channel 25, no Wifi in Higher Channels, only 21 Devices, Hue, Sonnoff, Ikea. Was running fine for many month. 

Stopped working suddenly during the Day last week. I think i've upgrade from 2.0.0 to 2.1.1 the Day before, plus normal Systempatches.

1mfaasj, 2025-02-24:
Ok an update from my side. I'm still on the latest z2m docker version but switched to zigbee coördinator version 20221226. 
Then started again from stretch. I needed to rebuild the whole zignee network by repairing everything again. Now I'm scared to update anymore... The error still occurs sometimes but not EVERYtime anymore

serot23, 2025-02-24:
> Ok an update from my side. I'm still on the latest z2m docker version but switched to zigbee coördinator version 20221226. Then started again from stretch. I needed to rebuild the whole zignee network by repairing everything again. Now I'm scared to update the things... The error still occurs sometimes but not EVERYtime anymore

How long did you test it with the new setup, because sometimes it takes a day before a devices gives error at mine

ssunny1081, 2025-02-25:
Also downgrade to 1.4 + Update did not fix it. Disabled usb autosuspend now, never had this issue before but who knows..

ssunny1081, 2025-02-26:
Again, down after around 18h. Have to wait for the new HW now.

darexa, 2025-02-27:
> Same thing started happening to me yesterday. It's currently working again by restoring earlier zigbee2mqtt backup and after that upgrading again to latest. Will see if it lasts.

Things are still running well for me.

ssunny1081, 2025-02-28:
Reflashed the Dongle with the same Firmware, since than it is running fine, almost two Days without any issue. Still on 2.0.0.

serot23, 2025-03-01:
> Reflashed the Dongle with the same Firmware, since than it is running fine, almost two Days without any issue. Still on 2.0.0.

Which firmware version and do you have a link or something to it?

serot23, 2025-03-01:
> I flashed it again roughly 1.5 days ago with the firmware listed at
> 
> https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
> 
> Specifically for Sonoff Plus, it's the CC1352P2_CC2652P_launchpad_*.zip file as described in the Readme.
> 
> I did had to re-pair all my devices, and zigbee2mqtt didn't start out of the box.
> 
> Had to update the zigbee2mqtt config in HA as well, to generate a new pan_id. Which was easy to do, if you have a file editor available on your HA instance such as VS Code Server.
> 
> In the zigbee2mqtt folder there's a configuration.yml, open that file and ensure you have set the pan_id to GENERATE. Reboot zigbee2mqtt and it will generate a new pan_id.
> 
> More info: https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html
> 
> I'm running 2.1.1-1 now without issues for almost 2 days.

Thanks for the explanation, i will try it in the coming days

ssunny1081, 2025-03-01:
I've also re-flashed CC1352P2_CC2652P_launchpad_coordinator_20240710.zip
First start was empty device list, copied the old config and DB, than Z2M did not start because same Network ID was already detected after around 20-30s.
I've unplugged all non Battery devices and removed the antenna. Than start of the coordinator was fine, all devices still paired and working.

d-d, 2025-03-03:
Similar issue here after docker upgrade. one IKEA [LED1732G11] (https://www.zigbee2mqtt.io/devices/LED1732G11.html#ikea-led1732g11) just stopped responding to commands. However it reports status?

I am using the docker version on a raspberry 5 with a conbee II stick.

```
`z2m: Publish 'set' 'state' to 'livingroom_uplight' failed: 'Error: ZCL command 0xbc33acfffe826587/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":20}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (183) Error: waiting for response TIMEOUT)'`
```

The light still seems to report state:
```
{
    "brightness": 254,
    "color_mode": "color_temp",
    "color_temp": 454,
    "linkquality": 255,
    "state": "ON",
    "update": {
        "installed_version": 587814449,
        "latest_version": 587814449,
        "state": "idle"
    },
    "update_available": false
}
```

serot23, 2025-03-03:
I updated to the last version, and some devices stopped working again. Reconnected some, so i will look if they will stop again after couple hours/days

brianveltman, 2025-03-05:
Still going strong with my zigbee network for about a week now.

I wrote down the steps I took to fix those issues at https://brianveltman.com/blog/solving-zigbee2mqtt-and-sonoff-issues/ maybe it will help some of you to fix their zigbee issues as well.

serot23, 2025-03-09:
It looks like everything goes well so far, my only complain is my keypad from Xfinity UEHK2AZ0 responding very late to the arming and disarming sequence, there is a delay of 20 seconds. I might try repairing it again maybe that will help

serot23, 2025-03-19:
So a update from me, we are 2 weeks further and i have zeeo problems anymore. I have deleted the keypad and reconnected it and that also is fixed now. The rest i have explained earlier i am on the latest z2m and haos but started from scratch by deleting everything and also my configurations. Keep in mind that your automations also break and you need to save them again even when using the same names.