Koenkk/zigbee2mqtt #26378: (BLE-YL01) no update data with version Z2m 2.1.1

ThePhantom79, 2025-02-15:
### What happened?

I got a BLE-YL01 - Pairing was successfull and I get one time the correct data from the device. Sometimes after a while it will be renewed once or twice, but then : nothing. I dont get any new data  or any update from device. If I delete it from config and pair it again, then I get again once the data.

### What did you expect to happen?

Updated values every few minutes (10/15/30) .

### How to reproduce it (minimal and precise)

Just try to pair BLE-YL01 and wait for updated values.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-on in HomeAssistant in VM

lgardelle, 2025-02-21:
+1, same issue on my side with same configuration.

Koenkk, 2025-02-22:
Could you provide the debug log when not getting updates values when starting z2m + 10 minutes?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

closed 2025-05-01 by github-actions[bot]