Koenkk/zigbee2mqtt #26380: Z2M crashes when there is no internet connection

MSL-DA, 2025-02-15:
### What happened?

Z2M reports `Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` and restarts

### How to reproduce it (minimal and precise)

My internet provider was down, but my local network was up.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

SLZB-06M

### Setup

Docker

### Debug log

```
[16:46:15 15-02-2025] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[16:46:15 15-02-2025] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[16:46:15 15-02-2025] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[16:46:15 15-02-2025] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[16:46:15 15-02-2025] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[16:46:15 15-02-2025] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[16:46:15 15-02-2025] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[16:46:15 15-02-2025] error: 	z2m: Adapter disconnected, stopping
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000)
```

MSL-DA, 2025-02-15:
I have had several outages recently as my ISP is doing maintenance on their facility.

Every time there is a outage, Z2M crashes.

ghoz, 2025-02-16:
Could it be that your ISP router downs the local network during those outages ?
It seems z2m is unable to reach the adapter and the watchdog triggers

MSL-DA, 2025-02-16:
Z2M is on 192.168.10.xxx and SLZB-06M is on 192.168.20.xxx

If I create a firewall rule that blocks SLZB-06M from the internet, the same thing happens. I already have a firewall rule that allows connections between Z2M and SLZB-06M (192.168.10.xxx - 192.168.20.xxx)

If I move SLZB-06M to 192.168.10.xxx and pull the network connector to my UDM Pro, or block SLZB-06M from the internet, I don't get the error.

I simply don't understand that.

lux73, 2025-02-17:
I can't second this. My z2m instance plain with USB Coordinator on Debian 12.7 don't crash if my Linux Router goes Offline...

it works and never notice such WAN downtime

srett, 2025-02-17:
> I can't second this. My z2m instance plain with USB Coordinator on Debian 12.7 don't crash if my Linux Router goes Offline...

Obviously not if you're using a USB dongle instead of an ethernet one. :)

> Z2M is on 192.168.10.xxx and SLZB-06M is on 192.168.20.xxx
> If I create a firewall rule that blocks SLZB-06M from the internet, the same thing happens. I already have a firewall rule that allows [...]

Yeah, that doesn't make any sense on first glance. I guess you have no choice but to do some tcpdumping to see what's going on here. On the surface it seems the adapter refuses to process any more requests in that state, but why the difference wether it's on the same subnet or not...?

Nerivec, 2025-02-19:
There are some weird issues in that area... example: https://github.com/Koenkk/zigbee2mqtt/issues/23120#issuecomment-2585908365
Could be a device on your .20 subnet is misbehaving (like that post above) when WAN is down, and affecting the SLZB? When you move the SLZB to .10, that misbehaving device can no longer affect the SLZB. (I'm assuming a few things here, everything depends on your setup, but it's just a rough example anyway).

MSL-DA, 2025-02-19:
> There are some weird issues in that area... example: [#23120 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/23120#issuecomment-2585908365) Could be a device on your .20 subnet is misbehaving (like that post above) when WAN is down, and affecting the SLZB? When you move the SLZB to .10, that misbehaving device can no longer affect the SLZB. (I'm assuming a few things here, everything depends on your setup, but it's just a rough example anyway).

@Nerivec You are absolutely right. It was a mDNS issue. Thanks for the help 👍