Koenkk/zigbee2mqtt #26385: [New device support]: Tongou TOWSMR1  C40 - Smart Circuit Break

stricke2002, 2025-02-16:
### Link

https://www.tongou.com/product/zigbee-smart-circuit-breaker-rcbo-tosmr1/

### Database entry

Starting interview of '0x142d41fffe93e5ff' [2025-02-16 07:44:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff"},"type":"device_joined"}' [2025-02-16 07:44:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff","status":"started"},"type":"device_interview"}' [2025-02-16 07:44:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff"},"type":"device_announce"}' [2025-02-16 07:44:12] info: 	zh:controller: Succesfully interviewed '0x142d41fffe93e5ff' [2025-02-16 07:44:12] info: 	z2m: Successfully interviewed '0x142d41fffe93e5ff', device has successfully been paired [2025-02-16 07:44:12] warning: 	z2m: Device '0x142d41fffe93e5ff' with Zigbee model 'TS0601' and manufacturer name '_TZE284_432zhuwe' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html [2025-02-16 07:44:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff","status":"successful","supported":false},"type":"device_interview"}'

### Zigbee2MQTT version

 2.1.1

### Comments

I thought it was the same device as the Tongou TOWSMR1 C20, but this is the C40 model. This means it has intensity adjustment between 20-40A. The supported device has intensity adjustment between 0-20A.
 I think the settings are the same between the devices, only the limits need to be changed.

[image]
[image]

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_432zhuwe',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I already pasted the external definition, but I don't know if it is correct.

imitbn, 2025-02-20:
External definition based on the TOWSMR C20 from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L11991 works.
It shows over current threshold range: 1A - 40A

stricke2002, 2025-02-21:
Thank you! Problem solved! I wasn't aware of this method to support devices on Zigbee2MQTT until now.

imitbn, 2025-02-21:
May be it's better to leave this issue open until the device gets support?

stricke2002, 2025-02-21:
Yes, you are right!

Danymp1974, 2025-02-21:
It works partially, it does not allow you to set the release current, temperature and other parameters. It is definitely better to leave it open until there is specific support for this device

imitbn, 2025-02-24:
> It works partially, it does not allow you to set the release current, temperature and other parameters. It is definitely better to leave it open until there is specific support for this device

I can change all parameters

wojciechczyz, 2025-04-08:
Got the latest version of the zigbee2mqtt and it is still not defined. Should it be detected with 2.2.1

stricke2002, 2025-04-09:
Seems it is still not supported: 

info 2025-04-09 11:59:50zh:controller: Interview for '0x142d41fffe93e5ff' started
info 2025-04-09 11:59:50z2m: Device '0x142d41fffe93e5ff' joined
info 2025-04-09 11:59:50z2m: Starting interview of '0x142d41fffe93e5ff'
info 2025-04-09 11:59:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff"},"type":"device_joined"}'
info 2025-04-09 11:59:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff","status":"started"},"type":"device_interview"}'
info 2025-04-09 11:59:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x142d41fffe93e5ff","ieee_address":"0x142d41fffe93e5ff"},"type":"device_announce"}'
info 2025-04-09 11:59:52zh:controller: Succesfully interviewed '0x142d41fffe93e5ff'
info 2025-04-09 11:59:52z2m: Successfully interviewed '0x142d41fffe93e5ff', device has successfully been paired
warning 2025-04-09 11:59:52z2m: Device '0x142d41fffe93e5ff' with Zigbee model 'TS0601' and manufacturer name '_TZE284_432zhuwe' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Zigbee2MQTT version
    [2.2.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.2.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

wojciechczyz, 2025-04-09:
I think this is similar request:
https://github.com/Koenkk/zigbee2mqtt/issues/26808

LordRoge, 2025-04-11:
You open this topic first, but I hope this device will receive official support so users don't have to manually add external definitions themselves.

wojciechczyz, 2025-04-22:
[TOWSMR1.js.zip](https://github.com/user-attachments/files/19848829/TOWSMR1.js.zip)
I am attaching the device file I have received from the Tongou aliexpress shop when buying this device, it seems to be working well when I uploaded it to my Home Assistant Zigbee2MQTT external converters folder.