Koenkk/zigbee2mqtt #26390: Given sample Systemd service keeps restarting

sshaikh, 2025-02-16:
### What happened?

Since using `Type=notify` in my service, Z2M appears to restart every 20 mins or so.

### What did you expect to happen?

For Z2M to remain up and stable like previously.

When starting via cmd line it stays up.

### How to reproduce it (minimal and precise)

The main diffs:

```
  [Service]
  Environment=ZIGBEE2MQTT_DATA=/home/homeassistant/atal/homeassistant-config/zigbee2mqtt/data
- ExecStart=/usr/bin/npm start
+ Environment=NODE_ENV=production
+ Type=notify
+ ExecStart=/usr/bin/node index.js
  WorkingDirectory=xxx
  StandardOutput=inherit
  StandardError=inherit
+ WatchdogSec=10s
  Restart=always
+ RestartSec=10s
```

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

sonoff zstack

### Setup

https://www.zigbee2mqtt.io/guide/installation/01_linux.html

### Debug log

Unfortunately the logs are removed each time. I do have one zigbee bulb which remains off during the day, and so see expected `no ack` errors for it. Apart from that nothing seems awry.

sshaikh, 2025-02-17:
I can confirm that removing:

```
+ Type=notify
+ WatchdogSec=10s
+ RestartSec=10s
```
Has removed the issue. Happy to close unless further tests required.

srett, 2025-02-17:
Can you check the logs of the service using `journalctl -u zigbee2mqtt --since today` (or whatever you named the service file)? That way we should be able to confirm if/why systemd is actually killing z2m because of a watchdog timeout. Just check the lines around the point where the service was restarted.

The configured timeout of 10 seconds means that zigbee2mqtt needs to report to systemd every 10 seconds, otherwise it will be killed and restarted. Since you say it takes about 20 minutes we can rule out that the notification mechanism is completely broken, but it seems that occasinally, your z2m instance is so overloaded that it takes longer than 10 seconds to report. You could try setting the WatchdogSec to 20s and see if the issue persists.

While your current solution certainly avoids the bogus restarts, systemd will not kill and restart z2m should it actually ever deadlock and hang.

sshaikh, 2025-02-17:
When I first try starting z2m, I see this:

```
>sudo systemctl restart zigbee2mqtt.service
Job for zigbee2mqtt.service failed because a timeout was exceeded.
See "systemctl status zigbee2mqtt.service" and "journalctl -xeu zigbee2mqtt.service" for details.
```

After which the service is inaccessible for a while (presumably being restarted). The logs for this initial hang shows (I've redacted things I think were important):

```
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] debug:         zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [0,0,0,0] - 87
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] debug:         zh:zstack:znp: <-- SRSP: SYS - nvLength - {"len":0}
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] debug:         zh:zstack:unpi:parser: --- parseNext []
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] debug:         zh:zstack:backup: fetched adapter network security material table (capacity=2, used=2)
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] debug:         zh:zstack:backup: Following devices with link key are missing from new backup but present in old backup and database, adding them back:
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] info:         zh:controller: Wrote coordinator backup to 'coordinator_backup.json'
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] info:         zh:zstack:znp: closing
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] info:         zh:zstack:znp: Port closed
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] info:         z2m: Stopped zigbee-herdsman
Feb 17 18:10:45 home node[80572]: [2025-02-17 18:10:45] info:         z2m: Stopped Zigbee2MQTT
Feb 17 18:10:45 home systemd[1]: zigbee2mqtt.service: Failed with result 'timeout'.
Feb 17 18:10:45 home systemd[1]: Failed to start zigbee2mqtt.service - zigbee2mqtt.
Feb 17 18:10:45 home systemd[1]: zigbee2mqtt.service: Consumed 13.436s CPU time.
Feb 17 18:10:55 home systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Feb 17 18:10:55 home systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Feb 17 18:10:55 home systemd[1]: zigbee2mqtt.service: Consumed 13.436s CPU time.
Feb 17 18:10:55 home systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
Feb 17 18:10:55 home node[80737]: Starting Zigbee2MQTT without watchdog.
Feb 17 18:10:56 home node[80737]: [2025-02-17 18:10:56] debug:         z2m: Removing old log directory 'log/2025-02-16.12-26-55'
Feb 17 18:10:56 home node[80737]: [2025-02-17 18:10:56] info:         z2m: Logging to console, file (filename: log.log)
Feb 17 18:10:56 home node[80737]: [2025-02-17 18:10:56] debug:         z2m: Loaded state from file state.json
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] info:         z2m: Starting Zigbee2MQTT version 2.1.1 (commit #1accb8b6)
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         z2m: sd-notify is not installed
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] info:         z2m: Starting zigbee-herdsman (3.2.5)
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         z2m: Using zigbee-herdsman with settings: '"xxx
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS10"},{"path":"/dev/ttyS11"},{"path":"/dev/ttyS12"},{"path":"/dev/ttyS13"},{"path":"/dev/ttyS14"}>
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/ttyUSB0
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:controller: Starting with options yyy
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] info:         zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] info:         zh:zstack:znp: Serialport opened
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
Feb 17 18:10:57 home node[80737]: [2025-02-17 18:10:57] debug:         zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
```

So it's a timeout. Not sure what actually times out given the timestamps in the log. How does notify work?

I'll follow up with a "runtime" crash if it's different (I don't expect it to be).

sshaikh, 2025-02-17:
So, the timeouts are not different once Z2M is running. However looking at the logs, it seems the restarts are not every 20 mins - it's far more often than that, and I only thought it was 20 mins because that's the window when my external monitoring picks up on the failed service.

In terms of anything interesting, I see the following:

```
Feb 17 18:44:08 home node[81627]: [2025-02-17 18:44:08] debug:         zh:controller:endpoint: Error: ZCL command 0x60a423fffea7e360/1 manuSpecificTuya.mcuSyncTime({"payloadSize" ...
Feb 17 18:44:08 home node[81627]:     at ZStackAdapter.sendZclFrameToEndpointInternal (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:545:23)
Feb 17 18:44:08 home node[81627]:     at ZStackAdapter.sendZclFrameToEndpointInternal (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24)
Feb 17 18:44:08 home node[81627]:     at ZStackAdapter.sendZclFrameToEndpointInternal (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24)
Feb 17 18:44:08 home node[81627]:     at ZStackAdapter.sendZclFrameToEndpointInternal (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24)
Feb 17 18:44:08 home node[81627]:     at ZStackAdapter.sendZclFrameToEndpointInternal (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24)
Feb 17 18:44:08 home node[81627]:     at node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
Feb 17 18:44:08 home node[81627]:     at Queue.execute (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
Feb 17 18:44:08 home node[81627]:     at ZStackAdapter.sendZclFrameToEndpoint (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
Feb 17 18:44:08 home node[81627]:     at Request.send (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
Feb 17 18:44:08 home node[81627]:     at Endpoint.sendRequest (node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
```

I'll try extending the timeout next.

sshaikh, 2025-02-17:
It still fails with up to 60s (although less often of course). I'm not sure any notifications are going out.

srett, 2025-02-17:
Ok so sounds like the notification mechanism isn't working at all. No clue why though 🤷‍♂ 

> How does notify work?

systemd passes a file descriptor to the process and expects it to regularly send a heartbeat over it ([WATCHDOG=1](https://www.freedesktop.org/software/systemd/man/latest/sd_notify.html)), otherwise it considers the process crashed and will restart it.

sshaikh, 2025-02-17:
Ok, so am I supposed to do anything other than copy the service file in the installation instructions? Maybe there's an assumption that something else is set up?

If it's only me then, again, happy to close this.

srett, 2025-02-17:
Out of ideas currently. This should work out of the box, no special setup required. Would be nice to figure out what's going on there, but until someone else has the same issue, I guess just switch to regular service handling by removing Type=notify or changing it to Type=simple.

sshaikh, 2025-02-17:
Sure - happy to loop back once we have more data points. Fwiw as I said I do run my own monitoring and z2m rarely goes down for me anyway ;)

chrootlogin, 2025-02-18:
Same issue with Ubuntu 24.04 running on Raspberry Pi 4B. Service keeps restarting when Type=notify is enabled.

chrootlogin, 2025-02-18:
This issue seems to be a clone of #21463.

For now I could fix it by:
```
apt install libsystemd-dev
rm node_modules
pnpm i --frozen-lockfile
pnpm build
```

So if libsystemd-dev was not available on the install there is no failure but rather something doesn't work then. But in my opinion there should be an error on missed dependencies.

sshaikh, 2025-02-18:
Yes, looks like the same issue and the solution seems to have worked for me. 

The issue in the documentation IMO is that the install of `libsystemd-dev` et al is only mentioned during the (optional) systemd section, which is after the z2m installation section (which asks us to run `pnpm`). So although I had `libsystemd-dev` installed, I installed it too late.

Thanks for surfacing the solution!

srett, 2025-02-18:
Aaaah ok this is dumb, I must've had that lib installed for some reason. I'll see if I can make a PR that switches to [sd-notify-lite](https://www.npmjs.com/package/sd-notify-lite) instead, as it's a pure ts implementation.

closed 2025-04-27 by github-actions[bot]