Koenkk/zigbee2mqtt #26392: Thirdreality 3RTHS24BZ Failed to configure

ghoulsfrank, 2025-02-16:
### What happened?

I'm getting an error with this device in 2.1.1. Work fine in 2.0.2

### What did you expect to happen?

Not having this error 

### How to reproduce it (minimal and precise)

Using this device 

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

EmberZNet 7.4.3

### Adapter

Sonoff E dongle 

### Setup

HA Addon

### Debug log

2025-02-16 10:39:07z2m: Failed to configure 'Temp Sensor Thirdreality', attempt 3 (Error: Bind 0x282c02bfffe3955e/1 msTemperatureMeasurement from '0xe0798dfffed2107a/1' failed (Delivery failed for '11077'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:457:42) at Ezsp.emit (node:events:524:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5055:14) at Ezsp.callbackDispatch (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:736:26) at Ezsp.onAshFrame (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:396:22) at UartAsh.emit (node:events:524:28) at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1171:45) at processImmediate

Nerivec, 2025-02-18:
Try this:
- wake up the device by pushing its button once
- keep the device awake by pushing its button at regular 1-2sec intervals until it's done configuring (below step)
- trigger re-configure (yellow button in frontend) for the device

kosmonot, 2025-03-01:
I've been experiencing exactly the same issue since I migrated to z2m about 10 months ago. I'm running z2m ver 2.11 docker with an slzb-06m with 7.4.5 fw. The device pairs and is recognized, but the binding to the controller always fails. It functions ok and reports temp and hum reliably, but I always get three binding timeout errors every time z2m restarts:

```
[2025-03-01 08:03:48] error: 	z2m: Failed to configure 'bedroom_temp_and_hum', attempt 3 (Error: Bind 0x282c02bfffebb328/1 msTemperatureMeasurement from '0x9035eafffeaf14a7/1' failed ({"target":12382,"apsFrame":{"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":42},"zdoResponseClusterId":32801} timed out after 15000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23)
    at EmberOneWaitress.startWaitingFor (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1832:51
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at EmberAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1760:16)
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:158:13)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:209:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13))
```

I have tried the suggested push button wakeup re-interview trick with no luck:

```
[2025-03-01 08:18:14] error: 	z2m: Failed to configure 'bedroom_temp_and_hum', attempt 5 (Error: ZCL command 0x282c02bfffebb328/1 msTemperatureMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":100,"attribute":"measuredValue"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":12382,"apsFrame":{"profileId":260,"clusterId":1026,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":67},"zclSequence":33,"commandIdentifier":7} timed out after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23)
    at EmberOneWaitress.startWaitingFor (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2023:55
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1963:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:160:17))

```

Nerivec, 2025-03-01:
@kosmonot some devices have more specific quirks, did you check the [zigbee2mqtt docs](https://www.zigbee2mqtt.io/supported-devices/) for the device in question? You can also try to bring the device closer to the coordinator, re-pair it, and see if the configure still fails.
If the device is reporting incorrect power, it could mean the converter is having issues. Check if you find any [previous reports](https://github.com/Koenkk/zigbee2mqtt/issues) for the model in question and if not, you can create an issue for it.

Note: 
- `Interview` (blue `i` button) does a full interview similar to what is done when a device joins the network for the first time (it ignores cached data).
- `Reconfigure` (orange button) only triggers parts of the interview specific to the device (`configure`).

closed 2025-05-09 by github-actions[bot]