Koenkk/zigbee2mqtt #26394: BADRIG failing to join network. Randomly works.

mciantar, 2025-02-16:
### What happened?

I am trying to join some IKEA BADRING water sensor devices, and they fail to join, saying Unsupported with the following error:

error 2025-02-16 22:28:37zh:controller: Interview failed for '0x6cfd22fffe3c8d23 with error 'Error: Interview failed because can not get node descriptor ('0x6cfd22fffe3c8d23')'
error 2025-02-16 22:28:37z2m: Failed to interview '0x6cfd22fffe3c8d23', device has not successfully been paired

### What did you expect to happen?

I expected the devices to pair and join the network like all other devices I attempted to join so far. Network has 13 devices, like lights, temperature sensors and PIR sensors. They all pair without a problem. 

### How to reproduce it (minimal and precise)

Try to Pair a BADRING using a skyconnect with EmberZNet Driver. 

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.5

### Adapter

Skyconnect (EmberZNEt)

### Setup

Container on PI

### Debug log

[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=1)
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=5](frmRx=5) Added to rxQueue
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=1)
[2025-02-16 22:28:33] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=152 Len=22]
[2025-02-16 22:28:33] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=ZIGBEE_DELIVERY_FAILED], [type=DIRECT], [indexOrDestination=54327], [apsFrame={"profileId":0,"clusterId":2,"sourceEndpoint":0,"destinationEndpoint":0,"options":4160,"groupId":0,"sequence":118}], [messageTag=29]
[2025-02-16 22:28:33] debug: 	zh:ember: ~x~> DELIVERY_FAILED [indexOrDestination=54327 apsFrame={"profileId":0,"clusterId":2,"sourceEndpoint":0,"destinationEndpoint":0,"options":4160,"groupId":0,"sequence":118} messageTag=29]
[2025-02-16 22:28:33] debug: 	zh:controller:device: Interview - quirks check for 'undefined'-'undefined'-'Unknown'
[2025-02-16 22:28:33] debug: 	zh:controller:device: Interview - quirks did not match
[2025-02-16 22:28:33] debug: 	zh:controller:device: Interview - node descriptor request failed for '0x6cfd22fffe3c8d23', attempt 5
[2025-02-16 22:28:33] debug: 	zh:ember: ~~~> [ZDO NODE_DESCRIPTOR_REQUEST UNICAST to=0x6cfd22fffe3c8d23:54327 messageTag=30 payload=1e37d4]
[2025-02-16 22:28:33] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=153 Len=24]
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=6](ackRx=1)
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=6](frmRx=6) Added to rxQueue
[2025-02-16 22:28:33] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=2)
[2025-02-16 22:28:33] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=153 Len=7]
[2025-02-16 22:28:33] debug: 	zh:ember: ~~~> [SENT ZDO UNICAST messageTag=30 apsSequence=119 status=OK]

srett, 2025-02-17:
Try enabling join on specific routers or the coordinator only and hold the device close to them when trying. I had a specific brand TRV that would not pair successfully via a certain brand of tuya plugs, but only when using the coordinator or an ikea lightbulb.

gvlekke, 2025-02-25:
maybe this issue is related? https://github.com/Koenkk/zigbee2mqtt/issues/25438

irfancharania, 2025-02-25:
> Try enabling join on specific routers or the coordinator only and hold the device close to them when trying. I had a specific brand TRV that would not pair successfully via a certain brand of tuya plugs, but only when using the coordinator or an ikea lightbulb.

I've got this issue since upgrading to 2.x.
These used to work just fine on 1.4.x. 

I've tried joining specifically to coordinator and router but no luck.

Coordinator is sonoff E dongle and router is ikea tretakt plug.

No issues with other zigbee devices (ikea or other brands)

closed 2025-05-05 by github-actions[bot]