Koenkk/zigbee2mqtt #26397: [New device support]: Vanne de Régulation de Température pour Radiateurs Hydrauliques KOZII by XANLITE

hornet49, 2025-02-17:
### Link

https://www.xanlite.com/fr/systeme-de-regulation-denergie/1779-tete-thermostatique-radio-connectee-pour-radiateur-hydraulique-systeme-thermostats-connectes-3700619438993.html

### Database entry

const definition = {     zigbeeModel: ['TS0601'],     model: 'TS0601',     vendor: '_TZE200_ytryxh0a',     description: 'Automatically generated definition',     extend: [],     meta: {}, };  module.exports = definition;

### Zigbee2MQTT version

2.1.1

### Comments

I think it's the same device as the Powernity PO-THCO-EAU. only the brand changes. I think the settings are the same between the devices.
[https://www.zigbee2mqtt.io/devices/PO-THCO-EAU.html

[image]

](url)

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ytryxh0a',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I already pasted the external definition, but I don't know if it is correct.

DERBESSE, 2025-03-02:
Hi. 
Did you have a look at [https://github.com/Koenkk/zigbee2mqtt/issues/23079](url) ?

DERBESSE, 2025-03-03:
@hornet49 

https://github.com/Koenkk/zigbee2mqtt/issues/23079
Correct now ?

closed 2025-05-10 by github-actions[bot]